SpenceKonde/ATTinyCore #2: AttinyX5 PWM pins

hansibull, 2015-05-25:
Why is only Pin D0 and D1 marked as PWM pins in the pins_arduino.h file?

As far as I can read from the datasheet, these ATtinys have PWM capabilities on 4 pins. There should be PWM on pin D3 and D4 as well.

http://www.pighixxx.com/test/portfolio-items/tiny/?portfolioID=337

SpenceKonde, 2015-05-26:
I modified pins_arduino.h for that so that it now lists the PWM pin in the comments like the other pins. 

PWM on pin 4 should have worked (using timer 1, OC1B); I think all the plumbing for it is in place - are you saying it doesn't work?

PWM on pin 3 is not supported - YES you can get timer output on that pin - but it's the complementary output of OC1B (ie, when OC1B is high, this is low). This is only particularly useful for controlling BLDC motors and other special cases - for which you need to manually fiddle with registers anyway (instead of using analogWrite).

Unfortunately despite the fact that there are two timers, each with two output compare channels, we can only get 3 independent PWM's, because  Timer 1's OC1A output is on pin 1, which is also the OC0B pin :-/ It's a real shame they couldn't have put OC0B on another pin.

hansibull, 2015-05-26:
Thanks for the detailed explanation! Actually, I'm creating an ATtiny development board for Arduino, that supports Attiny13, Attiny25/45/85, Attiny24,44,84 and Attiny2313/4313, and I needed the correct pinout.
I've not tested all the PWM pins yet, but I will when the PCBs arrive :)
The board will look someting like this, but with ratsnest: http://i.imgur.com/eWzodn7.png

SpenceKonde, 2015-05-27:
Cool idea - It's sort of a pity that you can't get the good parts in nice socketable DIP packages (for example, look at an 841 next to an 84, or the 1634) anymore. Let me know if you find problems with this core with those parts. I've only tested this core with the weird ones, and not even very much of that, and I've changed very little from the TCWORLD repo I forked it from. 

One issue you're going to have right off the bat with your board is that you forgot the 0.1uf bypassing capacitors between power and ground right next to the chip. These must be present, otherwise the part will not work, and they must be as close as possible to the power and ground pins - you care even more about that than the traces for the crystal. Just last week I got back a batch of ATTiny85 prototyping boards, and I put the part on and all... fuses burned okay, but I tried to flash sketch? Only got one page written. The traces to the bypass cap were about 1.5" total, but that was too much, and I had to solder a 0.1uf cap across the top of the '85 between power and ground - after doing that, it worked fine.

hansibull, 2015-05-27:
Yeah, too bad 841 and 1634 isn't available in DIP packages. A little weird actually, these attinys would be perfect for breadboard prototyping and development.
Thanks for mentioning the bypass cap! I've never actually had any problems with that before (used several AVRs on breadboard), but I added three bypass caps, as close to each socket as possible :)
With the top (red) and bottom (blue) layer ground plane, the board looks like this:
http://i.imgur.com/MzXY7kS.png

You'll hear from me if this forked core doesn't work properly on these "common" attinys ;)

closed 2015-05-27 by hansibull