SpenceKonde/ATTinyCore #3: Json file for IDE 1.6.4

pierrotm777, 2015-05-31:
Hello,
Can you add a json file for the board manager ?

Thanks

SpenceKonde, 2015-05-31:
No plans to add until board manager stabilizes. Please install the normal way. 

Arduino has made backwards-compatibility-breaking changes of one manner or another in every 1.6.x release so far except 1.6.3 (which was an emergency fix for 1.6.2, so they didn't have time to break anything else). I have serious concerns about design decisions made w/regards to the specs for the json file, and I think changes will be required that would mean I'd have to rewrite the json. Board manager is still quite buggy in 1.6.4 - basically, this is a mess right now. So until things settle down, I'm not planning to support it.

IMO, board manager is a solution looking for a problem. Installation is as simple as downloading a file and extracting it in (documents)/Arduino/hardware - there's very little room for a board manager to make that easier, while the board manager itself adds a level of complexity and another point of failure.

samakar, 2015-06-30:
Hello,
I want to program Attiny88 with IDE 1.6.5. I copied your filed into avr folder. IDE looks fine. I uploaded ISP program to Arduino UNO . When I try to burn bootloader I get this error:

avrdude: invalid byte value ({bootloader.unlock_bits}) specified for immediate mode
avrdude: read from file '{bootloader.unlock_bits}' failed
Error while burning bootloader.

Please advice.

SpenceKonde, 2015-06-30:
Hm, are you using the latest version of the core? Last night I reworked boards.txt, and one of the things I noticed while going through it is that a lot of them didn't have the lock/unlock bits specified (which normally produces that error). So I *think* I fixed that last night...

samakar, 2015-06-30:
Thanks for prompt reply!
I used the one I had downloaded two days ago. But I am not still not quote
sure how to use your zip file. Last time:
1- I copied your avr directory to my existing avr directory and overwrote
all common named files.
2- I deleted you platform file and renamed platform_162 to platform.
3- restarted IDE

Is that right?

SpenceKonde, 2015-06-30:
2 days ago? Yup, that's definitely from before I fixed the 88, then. 

That install process sounds right - you can skip step 2 though. Only 1.6.2 needs platform_162 (it's a workaround for a bug that was fixed in 1.6.3 - though platform_162 is compatible with 1.6.3 and later as well)

samakar, 2015-07-01:
I copied your avr directory over arduino directory and over wrote over all
common files. But I lost all other board definitions even UNO. I cannot
program attiny88!  screeenshot attached. Please advice. Thanks.

SpenceKonde, 2015-07-01:
You have not attached any screenshot. 

Did you copy it over the avr folder inside the arduino installation folder (under program files or wherever it is installed to now)!? That's the only way I can think of that you could have lost board definitions. If you've done that, reinstall the IDE. 

My understanding is that you were asking how to update the core, since you indicated you had already install it - so I assumed that you were talking about the avr folder inside your arduino sketchbook folder (ie, (user folder)/documents/arduino/hardware/ATTinyCore/avr ) where you installed it the first time (which you did correctly). Per the readme file, this core does not require changing anything outside of your documents/arduino/hardware folder.

SpenceKonde, 2015-07-01:
Once you've installed it correctly, you should be able to select ATTiny x8 Series from the boards menu, and select the specific chip (48 or 88), clock speed, and BOD level from the appropriate menus that will appear in the tools menu after selecting ATTiny x8 series as the board.

samakar, 2015-07-01:
Thanks for prompt reply. I  had a copy of contents of the original "program
files/Arduino/hardware/arduino/avr" directory. So I deleted the modified
one and put the original avr directory back. Then I copied your
Attinycore-master directory in hardware directory. IDE looks perfect. But
when I run bootloader I got the error:

avrdude: invalid byte value ({bootloader.unlock_bits}) specified for
immediate mode
avrdude: read from file '{bootloader.unlock_bits}' failed
Error while burning bootloader.

samakar, 2015-07-01:
Last time I chose attiny88 from the boards menu. Now i did it exactly as
you said " ATTiny x8 Series" and got a different error:

Arduino: 1.6.5 (Windows 8.1), Board: "ATtiny x8 series, ATtiny88, 8 mhz
(internal), B.O.D. Disabled"

***failed;

avrdude: verification error, first mismatch at byte 0x0000

         0x3f != 0xff

avrdude: verification error; content mismatch

Error while burning bootloader.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

SpenceKonde, 2015-07-01:
What entry from boards menu did you choose? "ATTiny x8 Series" is the one you want.

I also just fixed a bug that prevented successfully burning bootloader on all the new board defs (forgot the dummy file), so you'll need latest version again. >.>

samakar, 2015-07-01:
I downloaded your last update and chose "ATTiny x8 Series" and error!

Arduino: 1.6.5 (Windows 8.1), Board: "ATtiny x8 series, ATtiny88, 8 mhz
(internal), B.O.D. Disabled"

***failed;

avrdude: verification error, first mismatch at byte 0x0000

         0x3f != 0xff

avrdude: verification error; content mismatch

Error while burning bootloader.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

samakar, 2015-07-01:
I got the same error as the last attempt.

SpenceKonde, 2015-07-01:
Since I can't test for at least 10 days....
Can you enable verbose uploading and send me the output? I think I know what the problem is...

I created a new issue for this https://github.com/SpenceKonde/ATTinyCore/issues/7 - please respond over here, to spare pirrotm77 any more spam, since he presumably is getting updates for this, and this is a distinctly different issue.

SpenceKonde, 2015-07-25:
I'm worried about how to support this. 

The issue is the checksum - the json file requires a checksum. Which means every time I update the repo, I would have to update the json file with a new checksum - or have a separate release cycle, which would involve my designating the code in a "stable" state. With several updates a week coming in to fix newly discovered bugs (as people actually try to use this stuff, and we discover that the core never actually worked).

per1234, 2015-09-19:
>designating the code in a "stable" state

Yes, you would definitely need to start versioning.

Here's something you might find useful in automating the process of updating the json file:
https://github.com/Optiboot/optiboot/blob/master/optiboot/release.sh
https://github.com/Optiboot/optiboot/blob/master/optiboot/package_optiboot_optiboot-additional_index.json.TEMPLATE

It's a bit of work to get the Boards Manager install support set up and maintain it but it offers an easy option for users to install and update the core as well as roll back to previous version if the new version is causing problems. This may reduce the number of support requests you receive.

Other information that might be useful if you decide to do it:
[Arduino IDE 1.6.x package_index.json format specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x---package_index.json-format-specification)
[Guide to Boards Manager Support File Location Options on GitHub](http://playground.arduino.cc/Main/BoardsManagerSupport)

I have experience in implementing this and am happy to help out if you want.

SpenceKonde, 2015-09-19:
I'd love some help setting this up. I've got a server that the scripts can run on and everything (drazzy.com running Amazon Linux), but looking at that system, it just makes my head spin, they've managed to take something so simple "sync github repo to hardware" into something incredibly complicated. With checksums! I just don't even know where to get started in implementing this - I frankly am really uncomfortable working on things like this, and I don't really understand why...

SpenceKonde, 2015-09-19:
I think part of the issue is that board manager is something I would never use myself, for my core or anyone else's - syncing the repo to hardware is just so easy...

SpenceKonde, 2015-09-19:
Yeah, I have no idea how to do the tools section AT ALL. 

And what's this about there not being an AVR folder? So the directory structure is different when you install via board manager?

per1234, 2015-09-19:
>I'd love some help setting this up.. I've got a server that the scripts can run on

First of all I have to confess I am clueless about how to use the optiboot script(but would be willing to see if I can figure it out). I only have experience writing the JSON file and setting up the installation archive(I've done 6 of them now, helped on a few others, and wrote some of the specification). So If you want to go ahead with it I'm happy to write the JSON file and submit a pull request but you will need to do a few things first:
- versioning - what version number do you want to start with
- file locations(see the [guide](http://playground.arduino.cc/Main/BoardsManagerSupport) I wrote for GitHub options):
  - JSON file location - I prefer usinging a gh-pages branch because that makes the boards manager install URL shorter but it's completely up to you.
  - archive file location - I prefer using GitHub releases for this because it's a preexisting mechanism for handling versioning.
- create necessary branches/releases or whatever implementation infrastructure is needed that I can't create via pull request.

>board manager is something I would never use myself...syncing the repo to hardware is just so easy... 

Easy for you but for someone who is already trying to learn many new things at once it's just one more thing to figure out. To me the great thing about Arduino is it attempts to simplify these processes as much as possible so there is less of a barrier for get the hardware doing what they want. Once they have the LED blinking they are hooked and will be willing to spend the time to learn more. Boards manager just removes the need to figure out how to use git or manually install cores.
> I don't know where to get the avr tools - you already have them when you install the IDE.

I don't think there is any need for special tools for this core, the necessary ones are already installed. I'm guessing it will be similar to https://github.com/damellis/attiny/tree/ide-1.6.x-boards-manager which I wrote.
> the directory structure is different when you install via board manager? 

Yes, that's a bit of a hassle. You either have to create the different folder structure in the installation archive file or you can do as damellis/attiny did and make a separate branch with the correct structure and use the automatically generated GitHub archive file. The only problem with that is the checksum value can change which will cause installation to fail. That happened once with attiny but it has been something like 6 months since without it happening again. I don't know if there was a force push or maybe the GitHub server crashed and it had to remake the archive file.

SpenceKonde, 2015-09-19:
Re: Versioning
Aaah - I hadn't seen the Github releases feature until now. I figure I should probably start with a version like, oh, 1.0.0 for the first version that I have json file for? That seems like a good number to start with. Basic functionality is intact.

Likewise on the json file location - I can certainly host it on a real website, and automatically sync that to the repo; this might be nice in that the URL would probably be short and sweet (ie, http://drazzy.com/attiny.json or something), but that does mean there's more work to do for each release. 

I don't understand is how the checksum would be generated if using github for hosting - it sounds like you'd still need a script running on some machine somewhere to dpwnload the archive, generate the checksum, and then update the board manager json file. Maybe I'm missing something obvious here? 

The optiboot script looks like it's dynamically generating some of the files, which is not necessary for anything I'd be doing - but I still don't see how to get away from needing a script to change the directory structure, other than using a branch - but I don't know how to keep the branches in sync if they have different directory arrangements; I don't have much experience with using branches in github, but I don't see any way to have a branch that has all the same files, just with a slightly different directory structure, short of having a computer with the two repos cloned on it, and a script to update the master, copy the files, and commit the changes to the board manager branch...

Re: Tools 
Ah, okay, that page seemed to imply that even I'd still need to include locations to download them from and checksums of the files and all that, whether or not they were included in the IDE

One more thing - you may be aware that I have a second ATTiny core for the 4 "modern style" ATtinies (the ones the PUEx register, and the epic assortments of on-chip peripherals)- the 441/841, 1634, and 828. Unfortunately, since they were forked from separate cores, they're organized very differently, so I can't easily combine them into one (that is a longterm goal a while ago, but I don't think it's something I'll have the time to do in the next couple of months). So in the nearterm, that means everything done for boardmanager here, I need to be able to duplicate for my ATTiny Modern core - I don't think that will be hard though, if you're able to get me sorted out with this core.

per1234, 2015-09-20:
> http://drazzy.com/attiny.json

There is a naming convention in the specification that is intended to avoid duplicate package names: package_YOURNAME_PACKAGENAME_index.json
>more work to do for each release

Yes and I wouldn't be able to make a pull request but I can just post the file contents here and you can copy it over.
>how the checksum would be generated

I have just been using a program(md5deep) to get the checksum. I'm guessing the optiboot script is run on the local repository on his machine and then I think he just manually adds the automatically generated JSON and install archive files to the release though I see releases can be made via the GitHub API so it is possible to automate this process further. I'm also willing to watch this repository(and the other) and then I'll get an email every time you make a release and I'll update the json file and make a pull request. A human script you might say.
>I don't know how to keep the branches in sync if they have different directory arrangements

Apparently git is smart enough that it just works. I just created a test repository, made the avr folder in the master branch, with a file in the avr folder and commited. Then I branched that to a BM branch and moved the file from the avr folder to the root and deleted the avr folder and commited. Then I edited the file in the master branch and commited. Then switched to the BM branch and did `git merge master` and the edit made to the avr/testfile in master was made to /testfile in BM. Pretty cool!
>a second ATTiny core

I think the recommended way to handle boards manager installation for multiple cores maintained by one person is to have a single JSON file for all of them. This is how [adafruit](https://adafruit.github.io/arduino-board-index/package_adafruit_index.json) and [sparkfun](https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json) does theirs but I think it's up to you if you want to have a single JSON file with both cores or a different JSON file for each. I'm happy to help with both, as you say it should be simple to just duplicate the system you decide on for this one to the other.

SpenceKonde, 2015-09-20:
Re: branch sync behavior

That is *awesome* - that takes care of that. I'll get the branches set up tomorrow. 

Re: second core
If a single JSON file can support both cores, that's awesome. I didn't realize that was possible. Makes everyone's life easier - and when I finally combine them into one, it'll be less disruptive to boot. Sounds good. 

Re: Names
Understood on the naming convention - wasn't thinking about the file name, just the hosting location. Even the gh-pages urls are a lot longer than drazzy.com url would be. 

What if I make a new repo for board-manager stuff, including the json file, and, when I write them, the scripts to automate the process. I could then have a trivial script on the server syncing the board-manager-stuff repo every half hour or so, and copying over the new .json file if anything has changed (this script I've got the pieces for already, since I do nightly builds off a github repo for Espruino).

If you were able to watch these repos for releases and update the board-manager json for new releases, at least until I had a chance to write some scripts to automate it, that would be awesome. And I'll probably be in a better position to write such scripts once I've seen the whole system in action for a while. 

So - I need to make the new branches, and repo tomorrow. What information do you need from me to make the json file?

Thanks!

per1234, 2015-09-20:
>What information do you need from me to make the json file?

What do you want the name of the file to be? It looks like the specification recommends package_drazzy.com_index.json

I think I can guess on everything else. I'll submit the JSON file via pull request with a test URL so you can see how it will look in Boards Manager and then I'll make any changes necessary before you merge.

SpenceKonde, 2015-09-20:
package_drazzy.com_index.json  sounds good. 

When there's a spec for the name, that makes a hard decision easy!

ReleaseScripts repo created

SpenceKonde, 2015-09-21:
boardmanager branches created in both repos, and syncing tested

SpenceKonde, 2015-09-21:
And releases have been created. Hopefully that much is good to go now?

per1234, 2015-09-21:
Yes, I think that's everything I need. I'm working on the JSON file now.

per1234, 2015-09-21:
OK, I have a successful Boards Manager install of both cores but I am getting some warnings:
```
Board ATTinyCore:avr:attinyx5 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX5
Board ATTinyCore:avr:attinyx4 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX4
Board ATTinyCore:avr:attinyx61 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX61
Board ATTinyCore:avr:attinyx7 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX7
Board ATTinyCore:avr:attinyx8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX8
Board ATTinyCore:avr:attinyx313 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX313
Warning: platform.txt from core 'ATtiny' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/avr/bin/, automatically converted to compiler.path={runtime.tools.avr-gcc.path}/bin/. Consider upgrading this core.
Warning: platform.txt from core 'ATtiny' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'ATtiny' misses property tools.avrdude.path, automatically set to {runtime.tools.avrdude.path}. Consider upgrading this core.
Board ATTinyCore:avr:attinyx5 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX5
Board ATTinyCore:avr:attinyx4 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX4
Board ATTinyCore:avr:attinyx61 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX61
Board ATTinyCore:avr:attinyx7 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX7
Board ATTinyCore:avr:attinyx8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX8
Board ATTinyCore:avr:attinyx313 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX313
Warning: platform.txt from core 'ATtiny' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/avr/bin/, automatically converted to compiler.path={runtime.tools.avr-gcc.path}/bin/. Consider upgrading this core.
Warning: platform.txt from core 'ATtiny' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'ATtiny' misses property tools.avrdude.path, automatically set to {runtime.tools.avrdude.path}. Consider upgrading this core.
Board arduino-tiny-841:avr:attinyx41 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX41
Board arduino-tiny-841:avr:attiny841opti doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY841OPTI
Board arduino-tiny-841:avr:attiny1634 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY1634
Board arduino-tiny-841:avr:attiny1634opti doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY1634OPTI
Board arduino-tiny-841:avr:attiny828 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY828
Board arduino-tiny-841:avr:attiny828opti doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY828OPTI
Warning: platform.txt from core 'ATtiny Modern' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/avr/bin/, automatically converted to compiler.path={runtime.tools.avr-gcc.path}/bin/. Consider upgrading this core.
Warning: platform.txt from core 'ATtiny Modern' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'ATtiny Modern' misses property tools.avrdude.path, automatically set to {runtime.tools.avrdude.path}. Consider upgrading this core.
Board ATTinyCore:avr:attinyx5 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX5
Board ATTinyCore:avr:attinyx4 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX4
Board ATTinyCore:avr:attinyx61 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX61
Board ATTinyCore:avr:attinyx7 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX7
Board ATTinyCore:avr:attinyx8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX8
Board ATTinyCore:avr:attinyx313 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX313
Warning: platform.txt from core 'ATtiny' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/avr/bin/, automatically converted to compiler.path={runtime.tools.avr-gcc.path}/bin/. Consider upgrading this core.
Warning: platform.txt from core 'ATtiny' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'ATtiny' misses property tools.avrdude.path, automatically set to {runtime.tools.avrdude.path}. Consider upgrading this core.
Board arduino-tiny-841:avr:attinyx41 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX41
Board arduino-tiny-841:avr:attiny841opti doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY841OPTI
Board arduino-tiny-841:avr:attiny1634 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY1634
[... cut ...]
Warning: platform.txt from core 'ATtiny Modern' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'ATtiny Modern' misses property tools.avrdude.path, automatically set to {runtime.tools.avrdude.path}. Consider upgrading this core.
```
I'm sure none of them cause problems but it may freak out some users to see so many warnings every time they open Boards Manager after installing your cores. So let me know if you want to change anything before I make the pull request.

SpenceKonde, 2015-09-21:
Yikes, yeah, I definitely gotta fix those - working on that now!

SpenceKonde, 2015-09-21:
New releases created.

per1234, 2015-09-21:
There's one last warning:
`Board ATTinyCore:avr:attinyx313 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINYX313`
No big deal but just letting you know.

SpenceKonde, 2015-09-21:
Fixed. I always forget about the x313.

SpenceKonde, 2015-09-21:
Script is set up now too, so within half hr from my merging pull request with a json file into ReleaseScripts repo, it'll be at http://drazzy.com/package-drazzy.com-index.json

SpenceKonde, 2015-09-21:
We have a problem... 

I supply my own avrdude.conf, and when installed manually, I had it in the root of the core: https://github.com/SpenceKonde/ATTinyCore/blob/master/avrdude.conf 

And I referred to it with:
tools.avrdude.config.path={runtime.hardware.path}/avrdude.conf

This does not resolve correctly when installed via board manager, and attempting to upload or burn bootloader yields:

avrdude: can't open config file "C:\Users\User\AppData\Roaming\Arduino15\packages\arduino-tiny-841\hardware\avr/avrdude.conf": No such file or directory

because {runtime.hardware.path} doesn't point to where it needs to (ie, C:\Users\User\AppData\Roaming\Arduino15\packages\arduino-tiny-841\hardware\avr\1.0.2 ). Do you know if there's anything like {runtime.hardware.path} that would point to a known place within the core?

per1234, 2015-09-21:
I think it should be {runtime.platform.path}
From https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification:
>{runtime.platform.path}     - the absolute path of the platform folder (i.e. the folder containing boards.txt)

per1234, 2015-09-21:
Ok I tested Burn Bootloader and Upload with an ATtiny85 with {runtime.hardware.path} changed to {runtime.platform.path} and it's working with Arduino IDE 1.6.5r5 and the hourly build.

SpenceKonde, 2015-09-21:
Thanks! Changes made, new releases created.

per1234, 2015-09-21:
Do you want me to make any changes to the core names/maintainer name/boards names/etc. listed on the Boards Manager entry?

SpenceKonde, 2015-09-21:
I think that all looks good... I can see how to change those things if I have to. 

I'll do a test run tonight after work to make sure everything works (ie I can actually program and bootload real boards from both cores).

per1234, 2015-09-21:
Great, I have pushed the update to the pull request and tested it with the ATtiny85. I also have an ATtiny84 I can breadboard up and test if you want. I'm happy to continue updating the JSON file as you create new releases. Once things are finished up you might consider adding your URL to:
https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls
and
http://playground.arduino.cc/Main/ArduinoOnOtherAtmelChips

SpenceKonde, 2015-09-21:
I can cover the testing, you've done plenty for me on this! I've got a set of boards to test with. 

Pending successful testing tonight, I'll definitely add my core to those lists.

ApexMoEng, 2015-09-21:
This core is looking incredibly solid. Good work all around.

SpenceKonde, 2015-09-22:
Good news! The change in platform.txt has also fixed the problem that necessitated the extra entries in programmers.txt!

I'm going to remove the extra entries, and do another release

SpenceKonde, 2015-09-22:
New releases ready. What a wonderful discovery! This should make life easier for everyone using my core.

per1234, 2015-09-22:
See there is an advantage to using Boards Manager ;) Do you want me to replace the 1.0.3 version with 1.0.4 or should I make both versions available in the JSON file?

SpenceKonde, 2015-09-22:
Replace the 1.0.3 - it's not usable due to the path problem. 

The workaround is now gone no matter how you install it, not just for board manager users. Of course,  I wouldn't have discovered it had we not been doing this.

per1234, 2015-09-22:
>not just for board manager users

Yes I know, I was just joking.

The JSON file is now updated and Burn Bootloader and Upload tested sucessfully with ATtiny85

SpenceKonde, 2015-09-22:
Thanks much! I've merged the pull request, so if my script works, the board manager URL should be live in 23 minutes

SpenceKonde, 2015-09-23:
Well, this seems to be working, so I guess It's time to close this!

Thanks!

closed 2015-09-23 by SpenceKonde

SpenceKonde, 2015-09-27:
per1234 - I have encountered a VERY serious issue - I'm wondering if you might have any ideas to help on this, and I forgot to ask for contact info from you (we should probably have eachother's email - mine is spencekonde@gmail ). You seem to know a lot about how the new third party hardware system works, and that's what's at issue here

It seems that the tools.avrdude.config.path defined in avrdude.conf is being ignored when using an ISP programmer, unless that ISP programmer was supplied by a third party in programmers.txt... 
It uses the path specified when bootloading no matter what programmer is selected (and if I supply upload.protocol in boards.txt, it always uses the right conf file, so the attiny-modern boards work if you upload via optiboot). 
Note that it doesn't matter which core's programmers.txt defined the programmer; as long it's from a third party core, it will use the avrdude.conf file specified for the selected board (ie, if I put a copy of usbasp entry into programmers.txt for the attiny modern core and select that programmer, then select a board from attiny classic, it will use the avrdude.conf path specified for attiny classic, not the modern core)... But using an isp programmer supplied with the IDE always uses the default avrdude.conf file... 

Is there a way around this, or do I need to go back to supplying programmers in programmers.txt for the sole purpose of making them use the config file specified in platform.txt?