SpenceKonde/ATTinyCore #4: Compiler error for ATtiny167

ApexMoEng, 2015-06-15:
Arduino: 1.6.0 (Windows 7), Board: "ATtiny167 @ 8 MHz  (internal oscillator; BOD enabled)"

Build options changed, rebuilding all

C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:199:6: error: #error UDR not defined
     #error UDR not defined
      ^
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp: In function 'void __vector_12()':
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:131:24: error: 'rx_buffer' was not declared in this scope
         store_char(c, &rx_buffer);
                        ^
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:135:11: error: 'tx_buffer' was not declared in this scope
       if (tx_buffer.head == tx_buffer.tail) {
           ^
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp: At global scope:
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:180:1: error: expected unqualified-id before '{' token
 {
 ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

----

I get this compiler message when trying to verify with ATtiny167. This does not show up when trying to verify with other boards. I am using Arduino 1.6.0

SpenceKonde, 2015-06-15:
I may have just fixed this (not at home atm, so can't check)

It looks like HardwareSerial had all the support for the combined UART/LIN controller on the 167 in, and all the heavy lifting was done - but they'd left debris in. Under the combined UART/LIN controller, there is just the LIN_TC vector to worry about - and it has to do both tx and rx; on a normal UART, there are separate interrupts. HardwareSerial.cpp didn't have the DRE (data register empty, ie, "hey, I'm transmitting and I need the next byte") ISR #if'ed out for devices with LIN.

There may be other issues, of course.

ApexMoEng, 2015-06-15:
Ok, how long do you expect until you would be able to check?

I also tried FemtoCow's build with Arduino 1.6.0 & 1.6.4 with the same result.

I just tried TCWORLD's buid with Arduino 1.0.5 had other compiler errors.

SpenceKonde, 2015-06-15:
So my fix didn't work? What errors do you get now?

ApexMoEng, 2015-06-15:
Miscommunication. If your fix is up I will download and try it out. I will give you feedback shortly.

ApexMoEng, 2015-06-15:
Arduino: 1.6.0 (Windows 7), Board: "ATtiny167 @ 8 MHz  (internal oscillator; BOD enabled)"

C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp: In function 'void __vector_12()':
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:131:24: error: 'rx_buffer' was not declared in this scope
         store_char(c, &rx_buffer);
                        ^
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:135:11: error: 'tx_buffer' was not declared in this scope
       if (tx_buffer.head == tx_buffer.tail) {
           ^
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp: At global scope:
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:247:3: error: expected identifier before ')' token
   )
   ^
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:226:1: error: prototype for 'HardwareSerial::HardwareSerial(ring_buffer*, ring_buffer*, int)' does not match any in class 'HardwareSerial'
 HardwareSerial::HardwareSerial(ring_buffer *rx_buffer, ring_buffer *tx_buffer,
 ^
In file included from C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\Arduino.h:222:0,
                 from C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:28:
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.h:32:7: error: candidates are: HardwareSerial::HardwareSerial(const HardwareSerial&)
 class HardwareSerial : public Stream
       ^
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.h:48:5: error:                 HardwareSerial::HardwareSerial(ring_buffer*, ring_buffer*)
     HardwareSerial(ring_buffer *rx_buffer, ring_buffer *tx_buffer
     ^
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp: In member function 'virtual size_t HardwareSerial::write(uint8_t)':
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:378:23: error: 'tx_buffer' was not declared in this scope
     unsigned char c = tx_buffer.buffer[tx_buffer.tail];
                       ^
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp: At global scope:
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:399:26: error: 'rx_buffer' was not declared in this scope
   HardwareSerial Serial(&rx_buffer, &tx_buffer);
                          ^
C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:399:38: error: 'tx_buffer' was not declared in this scope
   HardwareSerial Serial(&rx_buffer, &tx_buffer);
                                      ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

----
Fresh download of Arduino 1.6.0
Fresh download of .ZIP file from the main ATTinyCore directory.

SpenceKonde, 2015-06-15:
I think I've fixed the undefined tx and rx buffer issue. 

I'm not certain if I've fixed the paren issue, but I may have. It's complicated, because they used preprocessor #if's in the middle of a function definition, so they can change the number of arguments it takes based on what's defined, and it's easy to get confused by code like that.

ApexMoEng, 2015-06-15:
You're doing good work. I am but a humble test bot.

Seems like the last fix worked. See image. I haven't soldered a 6 pin header to upload any code to a ATtiny yet so I can't tell if everything works from that point forward. But seems to compile now.

[image]

SpenceKonde, 2015-06-15:
Hurray!

Let me know if you have problems when you actually try to use it - it would not surprise me at all, considering the state the rest of this core was in. I think the person who made this core (TCWorld) did so by combining all the available cores for ATTiny's into one core that supported all the parts - but they went light on the testing of the resulting frankencore, probably because he didn't have a menagerie of Tiny's on breakout boards. The 1634 also competes with the 167, and the 1634 looks better on a lot of metrics.

ApexMoEng, 2015-06-15:
I'm getting a board made up that should be done in about two weeks. I'll make sure to comment on the process when that time comes. Currently I only have the Atmel STK500 Development board which comes with a ATMega8515. I might have to go through this whole process again trying to find a 8515 core while I wait for the Tiny to come in.

SpenceKonde, 2015-06-20:
Nope, not fixed, I don't know how you got it to build - I can't make it compile for me, and the fixing broke other x313.

SpenceKonde, 2015-06-20:
Nevermind, sync was fucked up and I was using an old version, and wondering why nothing I did fixed the compile issues

closed 2015-06-20 by SpenceKonde