SpenceKonde/ATTinyCore #5: Compiler errors

Isaac96, 2015-06-19:
I get errors when I try to compile for attiny2313.
Arduino: 1.6.5 (Windows 7), Board: "ATtiny2313 @ 8 MHz"

C:\Users\Isaac\Documents\Arduino 1.6.3\hardware\ATTinyCore-master\avr\cores\tiny\HardwareSerial.cpp:181:1: error: expected unqualified-id before '{' token
 {
 ^
Error compiling.
 Every other board compiles cleanly. Every one.

SpenceKonde, 2015-06-19:
Yeah, I think I messed something up when I fixed the #ifdef hell to get UART working on the 167, which works differently because it's a LIN/UART. The #ifdef's should have put a function declaration before the bracket, but it seems that they're not doing so. 

Actually, wait a darn minute here, isn't the 167 and the x313 the only UART capable chips that this core supports? Am I missing any? We've got defines to support 4 different names for the UART0 interupt vector, plus LIN - but we only support one series of chips...

I can't imagine how it would work on the 4313 but not the 2313 though.

Anyway, I'll try to get it fixed this weekend. 

You might want to look into the ATTiny841 and 1634. They're both ATTinys with 2 uarts, and they've got 8k and 16k of ram respectively, and the 841 has 6 PWM channels (3 timers)

SpenceKonde, 2015-06-20:
Oh, dur. Just a typo.

commit d4ba0cd6 (2015-06-20): Fix #5
closed 2015-06-20 by SpenceKonde