SpenceKonde/ATTinyCore #6: Create a new (and slick) boards.txt for Arduino 1.6

hansibull, 2015-06-25:
Since the launch of Arduino 1.6, a lot of developers have converted their old boards.txt files into new and optimized board files. The new IDE makes it possible to choose what kind of microcontroller you want to use, and what clock frequency from a custom menu. This makes the boards menu much cleaner and easier to use. It takes some time to make this new boards.txt file, but the result is great. I've done it with a boards.txt file for DIP-40 atmegas. You could add this new board file to the reposity as an alternative to the standard one :)

Here is the boards.txt file I made: http://codeviewer.org/view/code:52af

[image]

SpenceKonde, 2015-06-25:
Wow, that looks great, I had no idea that was possible. What version was that functionality added in? Does it work on 1.6.0?

hansibull, 2015-06-25:
I created it for Arduino 1.6.4, but I haven't tried it on lower versions. I believe it should work with 1.6.0.
This guy has done this with another ATtiny core. The great thing about your branch is that your branch supports way more ATtiny's that any other repo out there :)
Maybe it's possible to make a custom menu to select the BOD level, separated with the clock menu?

https://github.com/damellis/attiny

SpenceKonde, 2015-06-25:
I'll give it a try, then. That's really neat.

Yeah, I think I've got all the relevant tinies, either here or in the other repo (for the new-style ones)

hansibull, 2015-06-25:
Great! Looking forward to see the result! :D

SpenceKonde, 2015-06-28:
Any feedback on boards.txt in my attiny841 core? If no problems there, I'll apply the same code to this core's longer list of supported chips. 

This will also allow me to deal with the weird choices of boards in the boards.txt, so there won't be so many random restrictions on choices..... Maybe I'll leave the wacky stuff (which I don't want to deal with, nor be accused of book burning for removing) as separate entries? I dunno. Like, wtf else do I do about the UART-clocked ATwhateveritis? It's not like the core even works right with that option (i don't think the delay functions will work right), but it was in the core I forked....

hansibull, 2015-06-28:
Wow! That's a really neat boards.txt file! Looking forward to use this kind of setup with you ATinyCore! Maybe you could put the "Chip:" (or maybe use the term MCU) selection at the top? If I'd need a microcontroller, I would first pick one, then figure out the suited clock frequency, and then choose BOD.
And remember to use capital M and H in "MHz" :)

Oh, I got idea!
What if a bootloader-option menu was awailable under the tools menu? Then you don't need to have an aditional board called Attiny871 (optiboot)

Dummy pic below:
[image]

SpenceKonde, 2015-06-28:
I can make chip first, I agree it should come first where there's a chip menu

I don't think I can do the bootloader on a menu though, because the bootloader restricts the options possible for clock, and chip.

hansibull, 2015-06-28:
I've been searching around, and I guess you're right. As far as I can see it's not possible to make the menu content depending on other menus. But the new boards.txt file you're creating is so much better than the old one. Thanks for the fast response! I'll check out your other Arduino core as well! :)

SpenceKonde, 2015-06-29:
Actually, I don't think I can control the order of sub-menus. They seem to be alphabetical, no matter what I do.

closed 2015-06-29 by SpenceKonde

hansibull, 2015-06-29:
Actually, you can control the order of sub-menus. you just need to edit the menu name!
menu.bclock=Clock
menu.cbod=B.O.D.
menu.achip=Chip

The order will now be
Chop, Clock and BOD

SpenceKonde, 2015-06-29:
Oh cool. How did I not think of that?

Thanks

commit 1613cb30 (2015-06-29): Add slick new board submenus, fix #6

I left the debris at the bottom, but I think evertthing's okay for the
other options.