SpenceKonde/ATTinyCore #7: Failed Verification on Tinyx8 series

SpenceKonde, 2015-07-01:
https://github.com/SpenceKonde/ATTinyCore/issues/3

samakar, 2015-07-01:
Arduino: 1.6.5 (Windows 8.1), Board: "ATtiny x8 series, ATtiny88, 8 mhz (internal), B.O.D. Disabled"

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -pattiny88 -cstk500v1 -PCOM4 -b19200 -e -Ulock:w:0xFF:m -Uefuse:w:0xFF:m -Uhfuse:w:0xD7:m -Ulfuse:w:0xE2:m 

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM4

         Using Programmer              : stk500v1

         Overriding Baud Rate          : 19200

         AVR Part                      : ATtiny88

         Chip Erase delay              : 9000 us

         PAGEL                         : PD7

         BS2                           : PC2

         RESET disposition             : dedicated

         RETRY pulse                   : SCK

         serial program mode           : yes

         parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         ByteDelay                     : 0

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                  Block Poll               Page                       Polled

           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           eeprom        65    20     4    0 no         64    4      0  3600  3600 0xff 0xff

           flash         65     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff

           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500

         Description     : Atmel STK500 Version 1.x firmware

         Hardware Version: 2

         Firmware Version: 1.18

         Topcard         : Unknown

         Vtarget         : 0.0 V

         Varef           : 0.0 V

         Oscillator      : Off

         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.07s

avrdude: Device signature = 0x1e9311

avrdude: erasing chip

avrdude: reading input file "0xFF"

avrdude: writing lock (1 bytes):

Writing |  ***failed;  

################################################## | 100% 0.16s

avrdude: 1 bytes of lock written

avrdude: verifying lock memory against 0xFF:

avrdude: load data lock data from input file 0xFF:

avrdude: input file 0xFF contains 1 bytes

avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...

avrdude: verification error, first mismatch at byte 0x0000

         0x3f != 0xff

avrdude: verification error; content mismatch

avrdude done.  Thank you.

Error while burning bootloader.

SpenceKonde, 2015-07-01:
Oh FFS!

Problem is in the default avrdude.conf file!

I just updated it with a workaround I hope will work.

samakar, 2015-07-01:
Arduino: 1.6.5 (Windows 8.1), Board: "ATtiny x8 series, ATtiny88, 8 mhz (internal), B.O.D. Disabled"

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -pattiny88 -cstk500v1 -PCOM4 -b19200 -e -Ulock:w:0x3F:m -Uefuse:w:0xFF:m -Uhfuse:w:0xD7:m -Ulfuse:w:0xE2:m 

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM4

         Using Programmer              : stk500v1

         Overriding Baud Rate          : 19200

         AVR Part                      : ATtiny88

         Chip Erase delay              : 9000 us

         PAGEL                         : PD7

         BS2                           : PC2

         RESET disposition             : dedicated

         RETRY pulse                   : SCK

         serial program mode           : yes

         parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         ByteDelay                     : 0

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                  Block Poll               Page                       Polled

           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           eeprom        65    20     4    0 no         64    4      0  3600  3600 0xff 0xff

           flash         65     6    64    0 yes      8192   64    128  4500  4500 0xff 0xff

           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500

         Description     : Atmel STK500 Version 1.x firmware

         Hardware Version: 2

         Firmware Version: 1.18

         Topcard         : Unknown

         Vtarget         : 0.0 V

         Varef           : 0.0 V

         Oscillator      : Off

         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.07s

avrdude: Device signature = 0x1e9311

avrdude: erasing chip

avrdude: reading input file "0x3F"

avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of lock written

avrdude: verifying lock memory against 0x3F:

avrdude: load data lock data from input file 0x3F:

avrdude: input file 0x3F contains 1 bytes

avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...

avrdude: 1 bytes of lock verified

avrdude: reading input file "0xFF"

avrdude: writing efuse (1 bytes):

Writing |  ***failed;  

################################################## | 100% 0.16s

avrdude: 1 bytes of efuse written

avrdude: verifying efuse memory against 0xFF:

avrdude: load data efuse data from input file 0xFF:

avrdude: input file 0xFF contains 1 bytes

avrdude: reading on-chip efuse data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...

avrdude: verification error, first mismatch at byte 0x0000

Error while burning bootloader.

         0x07 != 0xff

avrdude: verification error; content mismatch

avrdude done.  Thank you.

SpenceKonde, 2015-07-01:
Same issue for ext fuse. They just can't seem to decide how to read unused fusebits. Sometimes they decide to mask them out so they read 0, sometimes they leave them so they read 1, and sometimes they split the difference and mask half of them because they fumbled the copy-paste. 

Try now. 

I'll fix it correctly over the holiday weekend.

SpenceKonde, 2015-07-01:
Do let me know if this workaround was successful, or if you ran into some other problem afterwards (so I know that there's something else to do as well).

samakar, 2015-07-01:
Boot loader worked!  Uploaded a blink sketch and now it's blinking. You
rock!

closed 2015-08-17 by SpenceKonde