SpenceKonde/ATTinyCore #8: Default avrdude.conf is a basketcase - don't use it

SpenceKonde, 2015-07-01:
Tiny48 missing.
Tinyx7 missing.

Lock bits are fucked up on the Tiny88

https://github.com/SpenceKonde/ATTinyCore/issues/7

per:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

we can specify a different location and include our own avrdude.conf that works correctly in platform.txt. This needs to be done.

ApexMoEng, 2015-07-17:
So is creating a new .conf file as simple as copying the default one and adding something similar to
https://learn.adafruit.com/introducing-trinket/programming-with-avrdude
to it?

SpenceKonde, 2015-07-17:
Yes - the hard part is getting the entry to add correct!

What I'm going to do is to include an avrdude.conf file based on the one in my 841 repo (only with entries for the x4, x5, x61, x7, x8, x313 series chips, ofc - you need the huge section at the top too), and modify platform.txt to tell avrdude to use that avrdude.conf  instead of the other one.

I've got entries for everything except the x7 series ready. (note - 87 and 167 entries can be easily made from eachother, it's the rest of it that's tricky)

ApexMoEng, 2015-07-17:
So apart from searching for someone else's attempt at a .conf file with the x7 series added to it, is there anything I can lend to the creation of one? I have the 167 operational and talking to AVR Studio. Do you need memory addresses or anything like in that last link?

SpenceKonde, 2015-07-19:
I think the information I need is in the datasheet, but there's always a lot of trial and error once the basics are set up to get everything right to make it upload. You can help by testing it once I've made the first attempt. If you can do that, which chip will you be using (so I know which one to start with - making 167 or 87 from one another is easy, once one of them works)

SpenceKonde, 2015-07-19:
Oooh! I found one! Does adding these to your avrdude.conf work?!
https://github.com/SpenceKonde/ATTinyCore/blob/master/avrdude_conf.txt

ApexMoEng, 2015-07-20:
The default one? I'm not adding it to a new one am I?

SpenceKonde, 2015-07-20:
Add the content of that avrdude_conf.txt to the bottom of your default avrdude.conf. 

(don't use avrdude.conf from this repo, it's not ready yet)

ApexMoEng, 2015-07-20:
Currently getting a:

avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding

Trying to debug and figure out where it is coming from, and get a proper upload atm.

SpenceKonde, 2015-07-20:
Is that when bootloading or when programming? 

Either way, enable verbose logging and post the results here, sometimes they're useful.

ApexMoEng, 2015-07-20:
Arduino: 1.6.0 (Windows Vista), Board: "ATtiny x7 series, ATtiny167, 8 mhz (internal), B.O.D. Disabled"

Build options changed, rebuilding all

Sketch uses 370 bytes (2%) of program storage space. Maximum is 16,384 bytes.

Global variables use 9 bytes (1%) of dynamic memory, leaving 503 bytes for local variables. Maximum is 512 bytes.

C:\Program Files\Arduino/hardware/tools/avr/bin/avrdude -CC:\Program Files\Arduino/hardware/tools/avr/etc/avrdude.conf -v -pattiny167 -cstk500v1 -PCOM1 -Uflash:w:C:\Users\roba\AppData\Local\Temp\build4335163831788404290.tmp/sketch_jul20a.cpp.hex:i 

avrdude: Version 6.0.1, compiled on Apr  3 2014 at 23:12:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Program Files\Arduino/hardware/tools/avr/etc/avrdude.conf"

         Using Port                    : COM1
         Using Programmer              : stk500v1
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x03

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

---------------------------------------
I get the same message when trying to upload or boot loading.

I have successfully programmed through the STK500 board using AVR Studio though.

SpenceKonde, 2015-07-20:
What isp programmer are you using? 

And you've done burn bootloader successfully?

ApexMoEng, 2015-07-20:
STK500 Board: http://www.atmel.com/tools/STK500.aspx

Haven't boot loaded successfully

ApexMoEng, 2015-07-20:
Would doing a teamviewer, and controlling our desktop help at all? I can post pictures of the physical set up as well.

SpenceKonde, 2015-07-20:
See, I'm not at all familiar with the STK500 programmer, and I don't hear about it being used much on the forums, so I'm not familiar with how to debug it. Most people use ArduinoAsISP, USBAsp, or AVRISP II.

Can you program well-behaved parts (like a '328p or something) with the STK500 board through the IDE? If not, that implies a general problem with STK500 board and the IDE. 
Do you have any of the more common (ie, better exercised) ISPs to try?

ApexMoEng, 2015-07-20:
I haven't tried programming other boards with the STK500 yet apart from the ATtiny.

I have an STK500 and an Arduino Uno available.
Is there a good guide for using the Arduino as an ISP available?

ApexMoEng, 2015-07-20:
Followed something like this http://www.instructables.com/id/Turn-Your-Arduino-Into-an-ISP/

Mapping the pins to the 6 pin header on my board.

Then used the Arduino Uno as an ISP.

Still getting.

Arduino: 1.6.0 (Windows Vista), Board: "ATtiny x7 series, ATtiny167, 8 mhz (internal), B.O.D. Disabled"

Sketch uses 370 bytes (2%) of program storage space. Maximum is 16,384 bytes.

Global variables use 9 bytes (1%) of dynamic memory, leaving 503 bytes for local variables. Maximum is 512 bytes.

C:\Program Files\Arduino/hardware/tools/avr/bin/avrdude -CC:\Program Files\Arduino/hardware/tools/avr/etc/avrdude.conf -v -pattiny167 -cstk500v1 -PCOM20 -b19200 -Uflash:w:C:\Users\roba\AppData\Local\Temp\build4335163831788404290.tmp/sketch_jul20a.cpp.hex:i 

avrdude: Version 6.0.1, compiled on Apr  3 2014 at 23:12:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Program Files\Arduino/hardware/tools/avr/etc/avrdude.conf"

         Using Port                    : COM20
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x03

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

avrdude done.  Thank you.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

ApexMoEng, 2015-07-20:
Wait. I may have just gotten it to work. I had not flashed the Uno with the proper ISP program.

I just tried uploading a blank sketch to the ITTiny167 and nothing seems to have failed.

After lunch I will test a sketch with code to see if it really is working.

ApexMoEng, 2015-07-20:
Did a basic blink sketch.

digitalWrite() works.

SpenceKonde, 2015-07-20:
That's good to hear! 

So it sounds like that avrdude.conf works, and the STK500 board does not work with the IDE, or some vital piece of information specific to that programmer is missing from the avrdude.conf entries (I don't know enough about the board to even say if that's plausible). I wouldn't be surprised if the 1.6.x IDE changes may have busted some of the less common programmers, without it being detected; a few people have reported (newish) issues with - I've noticed that most of the people with fancy programmer boards often are the same people who shun IDE upgrades, while the people who use the latest IDE versions buy the $5 and under programmers (ie, nano/pro mini running ArduinoAsISP, or USBAsp).

ApexMoEng, 2015-07-20:
Yeah. And from my findings there could be any number of things that is causing the STK500 not to communicate properly.

Regardless I have been playing with the Arduino IDE functions. I have digitalRead() and digitalWrite() working.

I can't seem to get SPI to work so far. I am not sure if it is hardware or software/code right now.

ApexMoEng, 2015-07-20:
I have a bad feeling the pin mapping is screwy. This is going to take a while to test though.

SpenceKonde, 2015-07-20:
It's supposed to be : 

```

// ATMEL ATTINY167
//
//                   +-\/-+
// RX   (D  0) PA0  1|    |20  PB0 (D  4)
// TX   (D  1) PA1  2|    |19  PB1 (D  5)
//     *(D 12) PA2  3|    |18  PB2 (D  6)
//      (D  3) PA3  4|    |17  PB3 (D  7)*
//            AVCC  5|    |16  GND
//            AGND  6|    |15  VCC
// INT1 (D 11) PA4  7|    |14  PB4 (D  8)
//      (D 13) PA5  8|    |13  PB5 (D  9)
//      (D 10) PA6  9|    |12  PB6 (D  2)* INT0
//      (D 14) PA7 10|    |11  PB7 (D 15)

```

ApexMoEng, 2015-07-21:
Using pinMode()

pinMode(0, X) corresponds to PA0
pinMode(1, X) corresponds to PA1
pinMode(2, X) corresponds to PA2
pinMode(3, X) corresponds to PA3

And so far I can't for the life of me get the other pins to work

ApexMoEng, 2015-07-21:
>=( Wait... Dx are the pins mapped to the code? WHY!? I sat at my desk for 4 hours trying to figure out why PA7/Pin 10 wasn't producing a signal. ARGH.

Ok I will try this out and see if the SPI works this time.

ApexMoEng, 2015-07-22:
SPI.h seems to function. I can produce a clock signal, chip select, MOSI etc. on my oscilloscope, but I am having a hard time driving peripheral devices. The issue is likely hardware, and I would say SPI communication works on the ATTiny167.

ApexMoEng, 2015-07-24:
Tested SPI.h with an FRAM chip. It communicated properly.

Working functions:
digitalWrite();
digitalRead():
SPI.h functions

SpenceKonde, 2015-07-24:
Thanks. Fyi - I just fixed another issue in the conf entry. You'd have found out when you got past 8k ;)

ApexMoEng, 2015-07-24:
Ah, well good catch then.

SpenceKonde, 2015-07-24:
Take a look at this and let me know what you think about it. https://github.com/SpenceKonde/ATTinyCore/issues/12

ApexMoEng, 2015-07-24:
1. Sure. I'll look at it shortly.

2. Tried using a library that came with Adafruits CC3000 Wifi board.
Received this:

Arduino: 1.6.1 (Windows 7), Board: "ATtiny x7 series, ATtiny167, 8 mhz (internal), B.O.D. Disabled"

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\IPAddress.h:24:0,

                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\Client.h:24,

                 from C:\Users\solidworks_laptop\Documents\Arduino\libraries\Adafruit_CC3000_Library-master/Adafruit_CC3000.h:32,

                 from buildtest.ino:30:

C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny/Printable.h:23:17: fatal error: new.h: No such file or directory

 #include <new.h>

SpenceKonde, 2015-07-24:
Yup, new.h and new.cpp are missing. My betting is some idiot at some point before I forked it thought it was an empty file left over from making a 'new' file, and deleted it, saw that basic sketches compiled, and patted themselves on the back for getting rid of unneeded files.

New issue created:
https://github.com/SpenceKonde/ATTinyCore/issues/13

commit 839a9fdf (2015-08-22): Fix #8, Fix #16

Should work the same as my ATTiny Modern core now \o/
closed 2015-08-22 by SpenceKonde