SpenceKonde/ATTinyCore #9: SPI connection on Attiny88

samakar, 2015-07-04:
Attiny88 has a SPI  interface. I uploaded a sketch including SPI.h to communicate with an Arduino. I can program my Attiny by ISP so SPI connection works. However my sketch  does't work and Arduino cannot communicate with Attiny88.
1-My code may  be incorrect. Is there a tested code available?
2-SPI.h does not work with Attiny88. Is there another SPI library for Attiny88?

Please advice.

SpenceKonde, 2015-07-04:
1) I don't have an 88 to test with, and I haven't done much with SPI in general on Arduino - so I'm afraid I can't help you there. The SPI interface on the t88 is identical to that on the 328, though - they should be code compatible, as far as SPI is concerned. 

2) SPI.h *should* work..... Might be missing #defines for MISO, MOSI, SCK pins; I don't know if the library needs them (I don't imagine it would, unless they were boneheaded about it). Could interrupts in general be borked? That would break SPI...

SpenceKonde, 2015-08-17:
No response, assuming problem solved, please reopen if this is not the case.

closed 2015-08-17 by SpenceKonde