SpenceKonde/ATTinyCore #15: [x61] Wrong fuse settings for external clocks and PLL

SukkoPera, 2015-08-20:
Low fuse for x61 for all external clocks (and PLL) is set to 0xFF, which means the chip is always set for an external 8 MHz clock.

Extract from boards.txt:
```
attinyx61.menu.clock.20external=20 mhz (external)
attinyx61.menu.clock.20external.bootloader.low_fuses=0xFF
attinyx61.menu.clock.20external.build.f_cpu=20000000L
attinyx61.menu.clock.20external.bootloader.file=empty/empty_all.hex
attinyx61.menu.clock.16external=16 mhz (external)
attinyx61.menu.clock.16external.bootloader.low_fuses=0xFF
attinyx61.menu.clock.16external.build.f_cpu=16000000L
attinyx61.menu.clock.16external.bootloader.file=empty/empty_all.hex
attinyx61.menu.clock.16pll=16 mhz (PLL)
attinyx61.menu.clock.16pll.bootloader.low_fuses=0xFF
attinyx61.menu.clock.16pll.build.f_cpu=16000000L
attinyx61.menu.clock.16pll.bootloader.file=empty/empty_all.hex
attinyx61.menu.clock.12external=12 mhz (external)
attinyx61.menu.clock.12external.bootloader.low_fuses=0xFF
attinyx61.menu.clock.12external.build.f_cpu=12000000L
attinyx61.menu.clock.12external.bootloader.file=empty/empty_all.hex
attinyx61.menu.clock.8external=8 mhz (external)
attinyx61.menu.clock.8external.bootloader.low_fuses=0xFF
attinyx61.menu.clock.8external.build.f_cpu=8000000L
attinyx61.menu.clock.8external.bootloader.file=empty/empty_all.hex
```

SpenceKonde, 2015-08-20:
Thanks, I'll get this fixed tonight!

SpenceKonde, 2015-08-20:
Or i'll just fix it now. 

It was only wrong for the PLL clock. 
For external crystal @8-20 mhz, the CKSEL settings are the same (0b1111, which means 0xFF for low fuse)

closed 2015-08-20 by SpenceKonde

SukkoPera, 2015-08-20:
Oh right! Of course I needed exactly the PLL clock and it wasn't working :). Thanks!

SpenceKonde, 2015-08-20:
Yeah, the PLL clock is one of the coolest things about the 861 imo. 

I actually had the same problem on the 'x5 too >.<

commit d8e3c05d (2015-08-20): Fix PLL clock fuses, fix #15