SpenceKonde/ATTinyCore #16: Burning bootloader fails on ATtiny85 - read operation not supported for memory "lock"

SukkoPera, 2015-08-21:
```
/opt/arduino-1.6.5/hardware/tools/avr/bin/avrdude -C/opt/arduino-1.6.5/hardware/tools/avr/etc/avrdude.conf -v -pattiny85 -cstk500v1 -P/dev/ttyACM0 -b19200 -e -Ulock:w:0xFF:m -Uefuse:w:0xFF:m -Uhfuse:w:0xD7:m -Ulfuse:w:0xFF:m 

avrdude: Version 6.0.1, compiled on Apr 14 2015 at 19:04:16
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/opt/arduino-1.6.5/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/home/sukko/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATtiny85
         Chip Erase delay              : 400000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    12     4    0 no        512    4      0  4000  4500 0xff 0xff
           flash         65     6    32    0 yes      8192   64    128 30000 30000 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          2    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e930b
avrdude: erasing chip
avrdude: reading input file "0xFF"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0xFF:
avrdude: load data lock data from input file 0xFF:
avrdude: input file 0xFF contains 1 bytes
avrdude: reading on-chip lock data:

Reading | avr_read(): error reading address 0x0000
    read operation not supported for memory "lock"
avrdude: failed to read all of lock memory, rc=-2

avrdude done.  Thank you.

Errore durante la scrittura del bootloader
```

SukkoPera, 2015-08-21:
Well, this is fixed as described here: https://arduinoelectronics.wordpress.com/2012/02/19/programming-theattiny85-or-attiny45-attiny84-attiny25-attiny2313-with-arduino/:

Under the ATtiny85 section of avrdude.conf, add below the line memory “lock”:

```
read            = "0 1 0 1  1 0 0 0  0 0 0 0  0 0 0 0",
                  "0 0 0 0  0 0 0 0  o o o o  o o o o";
```

So maybe this is also linked to #8?

SpenceKonde, 2015-08-21:
Wow, yup. This is just another example of why we need #8. 

That's just terrible. On the upside, I can no longer reproduce the problems I was having before with platform.txt not working with AVRDude.conf, so I should be able to get this fixed pretty soon.

commit 839a9fdf (2015-08-22): Fix #8, Fix #16

Should work the same as my ATTiny Modern core now \o/
closed 2015-08-22 by SpenceKonde