SpenceKonde/ATTinyCore #17: "8MHz internal clock" setting gives only 1/8th speed

hansibull, 2015-08-30:
I guess there has to be a a bug in the ATtiny core somewhere. When I choose the "8 MHz internal clock" selection on the Attiny85, Attiny84 or Attiny2313, I'm only getting 1/8th of the speed. When I running the blink sketch (1000ms on, 1000ms off) the LED toggles every 8th second. I've tried to look at the low fuse, but according to the fuse calculator, it seem fine. When I choose "1 MHz internal clock", the toggle speed is correct.

It also seems like you have mixed some of the fuses for the AttinyX313 (4 MHz and 0.5 MHz), and I get compiler errors for the 4 MHz and 0.5 MHz setting.

SpenceKonde, 2015-08-30:
Did you remember to do "burn bootloader" to set the fuses, after selecting 8mhz internal clock? You must do "burn bootloader" to set the fuses to make it actually run at 8mhz; this is not done as part of the normal upload process. 

To correct the burn bootloader unlock bit error, select a programmer with (ATTiny Classic) after it (as instructed in the readme) - that error is due to a problem with the avrdude.conf supplied with the arduino IDE. Choosing a programmer marked for use with my ATTiny will make it use the avrdude.conf file included with my core, which works correctly (the correct one will not be used for programmers not marked for ATTiny Classic, due to what appears to be another bug in the IDE)

If there are no programmers marked with that, redownload and reinstall the core - I only recently figured out how the IDE was choosing what avrdude.conf to use, and hence how to work around the bug. 

I think I know what the issue is with the ATTiny x313 at the low speeds - I'll look into that one.

SpenceKonde, 2015-08-30:
Actually, the bootloader unlock_bits issue I think is just because you're using old version of the core? 

What board are you choosing? The attiny x4 and x5 series are the ones you should be choosing (for an attiny 24, 44, or 84 / 25, 45, or 85, respectively)

SpenceKonde, 2015-08-30:
I just committed a change to comment out all boards not expected to work in boards.txt - some of those would have produced the error you reported.

SpenceKonde, 2015-08-30:
Compile errors with odd speeds fixed by commit: https://github.com/SpenceKonde/ATTinyCore/commit/2c394cd5554299abd903f5a7c63e48aa66a9a882

I've verified that I can bootload the 84 (x4 series) and 85 (x5 series) without errors with current version of core.

hansibull, 2015-08-31:
Hi! Yes, I forgot to hit "Burn bootloader" before uploading the code. After doing that, the speed seems fine :)
I forgot to mention it, but the bootloader upload error appeared when trying to upload Optiboot.

Last bug: Trying to burn bootloader to the ATtiny2313/4313 gives this error:

avrdude: AVR Part "attiny2313" not found.
Valid parts are:
  t1634    = ATtiny1634
  t167     = ATtiny167
  t24      = ATtiny24
  t25      = ATtiny25
  t261     = ATtiny261
  t44      = ATtiny44
  t441     = ATtiny441
  t45      = ATtiny45
  t461     = ATtiny461
  t828     = ATtiny828
  t84      = ATtiny84
  t841     = ATtiny841
  t85      = ATtiny85
  t861     = ATtiny861
  t87      = ATtiny87
  t88      = ATtiny88

Error while burning bootloader.

SpenceKonde, 2015-08-31:
Okay, yeah, the 84/85 optiboot board defs have not worked since before I forked the project. Some time I might try to resurrect that, but I can't say I really like the idea of a serial bootloader on devices with an internal resonator that needs tuning to make it accurate enough for serial comms and needs a software implementation of serial because it doesn't have a uart... With the 84, it seems to me madness to use such a bootloader, as the 841 is just pennies more per unit, and has two uarts AND an internal oscillator that's accurate enough to use them without special measures. In any event, I removed the non-working board defs. 

I've added the missing entries to avrdude.conf https://github.com/SpenceKonde/ATTinyCore/commit/ae0a7c416afa5060c2a8f1d458e17b0c57d88b92

hansibull, 2015-08-31:
Thanks for fixing the missing 2313/4313 section in the avrdude.conf file. I guess this is like the "Whack a Mole" game. Yoy squeeze one bug, and another one shows up. I deleted the old ATtiny core on my computer, and installed the newest one. Now I can't "burn" bootloader to any of the ATtinys. this is the error message:
avrdude: error at ~/Documents/Arduino/hardware/ATTinyCore-master/avrdude.conf:4493: page size (64) * num_pages (64) = 4096 does not match memory size (2048)
Error while burning bootloader.

I get the same error on X4, X5 and X313.

And could you also include an ATtiny Classic selection for the USBtinyISP? I got a few of these lying around, and they seems to be commented out in your programmers.txt file..

SpenceKonde, 2015-08-31:
Fixed avrdude.conf issue, try now.  Missed a difference between entry for 4313 and 2313 (and not at home, so I couldn't test myself)

hansibull, 2015-08-31:
Yup, that fixed the problem. Everything seems fine now. I'll let you know if I find more bugs ;)

SpenceKonde, 2015-08-31:
Great!

Re: USBtinyISP

Does USBtinyISP work if you uncomment it? I don't have any of those myself to test with. It should, work if you uncomment it; open a new issue if it doesn't. 

The ones that are commented out in programmers.txt *should* work if you uncomment them. I comment out the less common programmers to control clutter. If each core added 6+ programmers to that menu, it could get unwieldy fast (particularly for the majority of computer users, who run 1366 by 768 resolution - the lowest resolution manufacturers can claim to be "HD")

closed 2015-08-31 by SpenceKonde