SpenceKonde/ATTinyCore #18: (enh) Re-implement optiboot support for 84, 85, x7

SpenceKonde, 2015-08-31:
Should be able to do 8/12/16 mhz versions with the same binary, running at 19200, 28800 and 38400 respectively.

x7 version should be able to run at higher baud, because it has a real UART.

hansibull, 2015-08-31:
I'd really appreciate this! I don't have any experience with bootloaders loaded on microcontrollers without hardware UART. Does this mean that I don't have to include the soft serial library if I want to send data to the serial monitor? 

Anyway, if Optiboot support is added, maybe it's a good idea to include a simple sketch to tune the internal oscillator. My ATtinys where completely off scale when I first got them

SpenceKonde, 2015-08-31:
Yes - you need to use softwareserial to get serial on the ATtiny's without a UART- which as you may be aware (from the forums) is a little flaky. 

The readme for the old bootloaders suggests that they can be built with the tuner included (connect @9600 baud after burning bootloader and send a certain character until it says it's done) which you're supposed to use after burning the bootloader, which then writes the new OSCCAL value to flash.

MCUdude, 2016-05-14:
If you're planning on implementing optiboot sometime, have a look at this thread: [https://github.com/MCUdude/MightyCore/issues/24](https://github.com/MCUdude/MightyCore/issues/24)
I think this will spice up your core even more!

SpenceKonde, 2016-05-15:
I'm confused as to why you would put hooks into the core to support calling that to write the flash? How is it better than writing the flash from within the sketch? Or are those chips where only the bootloader can write the flash? The ATTiny chips don't have a special bootloader section that code that writes the flash has to be located in, so I think the callouts to the bootloader may not be helpful here?

How much does it add to the size to permit that sort of thing?

SpenceKonde, 2017-06-05:
This issue now handled in #141

closed 2017-06-05 by SpenceKonde