SpenceKonde/ATTinyCore #19: Attiny2313; Error burning bootloader in Arduino IDE, Avrdude command line works fine

hansibull, 2015-08-31:
I don't know if this is a bug in the core, or something wrong with my (only) Attiny2313. After your last commitment to the core, I get  "content mismatch error" every time I try to burn the bootloader. The microcontroller works fine, and I can upload new sketches to it. This is the error:

 ***failed;  
avrdude: verification error, first mismatch at byte 0x0000
         0x3f != 0xff
avrdude: verification error; content mismatch
Error while burning bootloader.

Whats weird is that I'm able to set fuses and burn bootloader with the avrdude command line on the same microcontroller. Is this a software or hardware problem?
(This is the output of the avrdude command execution:)

avrdude -c usbtiny -p attiny2313 -u -U hfuse:w:0x9f:m -U lfuse:w:0xE4:m -u -U flash:w:/Users/hansi/Documents/Arduino/hardware/ATTinyCore-master/avr/bootloaders/empty/empty2313at8.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e910a
avrdude: NOTE: FLASH memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "0x9f"
avrdude: writing hfuse (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of hfuse written
avrdude: verifying hfuse memory against 0x9f:
avrdude: load data hfuse data from input file 0x9f:
avrdude: input file 0x9f contains 1 bytes
avrdude: reading on-chip hfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of hfuse verified
avrdude: reading input file "0xE4"
avrdude: writing lfuse (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lfuse written
avrdude: verifying lfuse memory against 0xE4:
avrdude: load data lfuse data from input file 0xE4:
avrdude: input file 0xE4 contains 1 bytes
avrdude: reading on-chip lfuse data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lfuse verified
avrdude: reading input file "/Users/hansi/Documents/Arduino/hardware/ATTinyCore-master/avr/bootloaders/empty/empty2313at8.hex"
avrdude: input file /Users/hansi/Documents/Arduino/hardware/ATTinyCore-master/avr/bootloaders/empty/empty2313at8.hex auto detected as Intel Hex
avrdude: writing flash (332 bytes):

Writing | ################################################## | 100% 0.50s

avrdude: 332 bytes of flash written
avrdude: verifying flash memory against /Users/hansi/Documents/Arduino/hardware/ATTinyCore-master/avr/bootloaders/empty/empty2313at8.hex:
avrdude: load data flash data from input file /Users/hansi/Documents/Arduino/hardware/ATTinyCore-master/avr/bootloaders/empty/empty2313at8.hex:
avrdude: input file /Users/hansi/Documents/Arduino/hardware/ATTinyCore-master/avr/bootloaders/empty/empty2313at8.hex auto detected as Intel Hex
avrdude: input file /Users/hansi/Documents/Arduino/hardware/ATTinyCore-master/avr/bootloaders/empty/empty2313at8.hex contains 332 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.30s

avrdude: verifying ...
avrdude: 332 bytes of flash verified

avrdude done.  Thank you.

SpenceKonde, 2015-08-31:
Was problem with boards.txt  vs avrdude.conf - it was trying to write and verify lockbits set to 0xFF, but it only has 6 lockbits, and the two unused high bits are masked off in avrdude.conf, so they read as zero and the verification fails.

closed 2015-08-31 by SpenceKonde

hansibull, 2015-08-31:
I don't know if the issue is solved just yet. I downloaded the latest version again, and tried to burn the bootloader again. I tried a few times, and I always got error at byte 0x0001.
Suddenly the microcontroller stopped responding, and now this is the response from AVRdude:

avrdude -c usbtiny -p attiny2313 -F

avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
avrdude: Expected signature for ATtiny2313 is 1E 91 0A

avrdude done.  Thank you.

I guess I'll have to build a high voltage programmer to fix this "lost child" ;)

SpenceKonde, 2015-08-31:
At what point are you getting the error at byte 0x0001? The messages that
appeared around that are necessary to figure out exactly what happened, as
it's not immediately obvious what memory it's having the problem. Also the
values reported and expected are helpful in this - as is reading the
contents of the flash, if the problem is in flash section of memory.
(I was only able to fix the other problem because from the expected +
reported values it was possible to deduce that it was the lock bits that it
was choking on ).

Do you have a bypass cap in place? (0.1uf between Vcc and Gnd right next to
the chip being programmed). Without that, programming can fail, typically
resulting in verification errors.

SpenceKonde, 2015-09-01:
I found a 4313 and tested it - and it worked. 

I suspect it should work with the 2313 as well, barring something really weird.

You do need that bypass cap though - no chip should be expected to work without it.

hansibull, 2015-09-01:
Since the microcontroller acts like it's dead, I can't get the same error message as I got the first few times I tried to bootload. All I remember was that is was on the next byte. I do got a bypass cap. Its pace a few millimeters from the microcontroller. I'll build a high voltage programmer, and reset the microcontroller. I'll let you know the result

SpenceKonde, 2015-09-01:
I think I got it sorted out - I got the num_pages/page_size change
backwards.

ReDetection, 2015-12-26:
probably I got in the same issue — just made my attiny2313 unresponsive. I burned boorloader with 1mhz (I wanted to get 128khz, but 1mhz was ok for me too), and after burn I'm not able to do anything with it. I had version 1.0.5.

SpenceKonde, 2015-12-27:
What programmer are you using? I've been able to program my 4313 down to 0.5mhz internal and back with an el-cheapo USBAsp from ebay.

ReDetection, 2015-12-27:
I'm using Arduino Nano v3.0 with sketch "Arduino as ISP" from examples. Well, I checked the fuses — should be really correct.
OK, I'll try to find a high-voltage programmer and figure out if I'm able to recover it to try to reproduce the issue again. Will open a new issue in case.

commit 66a31af4 (2015-08-31): Fix #19

Stupid masking...