SpenceKonde/ATTinyCore #23: Provide an oscillator tuning sketch that fit even the smallest microcontrollers

hansibull, 2015-09-13:
I've been searching around for a tuning sketch for the ATtiny2313, but I can't find any thats less than 2048 bytes. The main problem is that every tuning sketch include serial communication, and a lot of long stings. I don't own a AVRISPmkII, so I can't calibrate with avrdude either. I believe that the tuning sketch mostly have to be written in AVR C, but I'm quite new to AVR code, so I'm not able to make one myself. 
A small Attiny tuning sketch that can be used with all microcontrollers in this core would be really great! :smile:

SpenceKonde, 2015-09-13:
I don't know how I'd fit a tuning sketch into 2k either - I don't know how anyone fits anything into the memory on the 2313, at least via Arduino - the basic wiring functions eat up the available flash really really fast.
I'd recommend doing it manually just setting OSCCAL and trying to write serial with it, until the serial works.  A semi-manual sketch like that would probably work. Like, just go through every value of OSCCAL one at a time printing out the osccal value to serial, then look at the serial log and set it to a number in the middle of the non-gibberish part of the log...  (or you could use a crystal - the 2313 has plenty of pins)

In any event, I have no plans to address this - writing a tuning sketch that works and fits in 2048 bytes is not an endevour I think I'd be successful at, and I don't think it would be a worthwhile effort just for the 2k parts, which don't really shine with Arduino because of how inefficient it is with flash on small chips.

closed 2015-09-13 by SpenceKonde

SpenceKonde, 2015-09-17:
I was playing around with tuning based off a dumb idea (as I realized I might soon need tuning, and the other tinytuners looked like a PITA to use) 

Tuning it off blink! 

So you run blink on one board, with a crystal, and this on the other, and it tries to get it within 1%... should be good enough for UART, and this sketch compiles to under 2k. However, I haven't actually been able to test it, as the factory calibration fairy has smiled upon me thus far - mine all seem to be good enough for uart (at least all the ones I had lying around on boards). 

This writes it out to EEPROM location 0 - you can read it out with avrdude/eXtremeBurner AVR, or read it from the eeprom in your sketch. 

https://github.com/SpenceKonde/AzzyProjects/blob/master/AzzyTune/AzzyTune.ino

Let me know if this is of any use :-P