SpenceKonde/ATTinyCore #29: SPI for ATTiny85 not compile

etmatrix, 2015-10-12:
I tried lastest version from github with ATTiny85 and example SPI BarometricPressureSensor.
I have this error:

    In file included from /tmp/arduino-1.6.5-r5/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /tmp/arduino-1.6.5-r5/hardware/tools/avr/avr/include/avr/pgmspace.h:88,
                 from /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/cores/tiny/Arduino.h:8,
                 from /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/libraries/SPI/SPI.h:17,
                 from BarometricPressureSensor.ino:26:
    /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/libraries/SPI/SPI.h: In member function 'void SPISettings::init_AlwaysInline(uint32_t, uint8_t, uint8_t)':
    /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/libraries/SPI/SPI.h:146:16: error: 'SPE' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                ^
    /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/libraries/SPI/SPI.h:146:27: error: 'MSTR' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                           ^
    /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/libraries/SPI/SPI.h:146:65: error: 'DORD' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                                                                 ^
    In file included from BarometricPressureSensor.ino:26:0:
    /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/libraries/SPI/SPI.h: In static member function 'static void SPIClass::beginTransaction(SPISettings)':
    /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/libraries/SPI/SPI.h:203:5: error: 'SPCR' was not declared in this scope
     SPCR = settings.spcr;
     ^
    /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/libraries/SPI/SPI.h:204:5: error: 'SPSR' was not declared in this scope
     SPSR = settings.spsr;
     ^
    /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/libraries/SPI/SPI.h: In static member function 'static uint8_t SPIClass::transfer(uint8_t)':
    /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/libraries/SPI/SPI.h:209:5: error: 'SPDR' was not declared in this scope
     SPDR = data;
     ^
    /tmp/arduino-1.6.5-r5/hardware/ATTinyCore/avr/libraries/SPI/SPI.h:217:14: error: 'SPSR' was not declared in this scope
     while (!(SPSR & _BV(SPIF))) ; // wait

I think library SPI is not ported to ATTiny. On github i have found another project for porting SPI into ATTiny.

SpenceKonde, 2015-10-12:
The tiny 85 doesn't have SPI. It has a USI that can be used as spi or i2c - but you need a different library.

I haven't used it, but I think this will do what you want:
https://github.com/JChristensen/tinySPI

SpenceKonde, 2015-10-21:
Closing, assuming problem solved.

closed 2015-10-21 by SpenceKonde

sreeku85, 2023-02-14:
When I compile the code for the Attiny85+Nrf24l01 remote, I encounter an error.
`In file included from c:\users\sree2\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\io.h:99:0,
                 from c:\users\sree2\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\pgmspace.h:90,
                 from C:\Users\sree2\OneDrive\Documents\Arduino\hardware\ATTinyCore\avr\cores\tiny/Arduino.h:10,
                 from C:\Users\sree2\AppData\Local\Temp\arduino-sketch-DEAE1529AB515C57B96B3D93CEF4201A\sketch\nRF24L01_aattiny85_send.ino.cpp:1:
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h: In member function 'void SPISettings::init_AlwaysInline(uint32_t, uint8_t, uint8_t)':
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:146:16: error: 'SPE' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                ^
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:146:16: note: suggested alternative: 'SPH'
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:146:27: error: 'MSTR' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                           ^
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:146:27: note: suggested alternative: 'PSTR'
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:146:65: error: 'DORD' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                                                                 ^
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:146:65: note: suggested alternative: 'DDRB'
In file included from c:\Users\sree2\OneDrive\Documents\Arduino\libraries\RF24/utility/ATTiny/RF24_arch_config.h:25:0,
                 from c:\Users\sree2\OneDrive\Documents\Arduino\libraries\RF24/RF24_config.h:72,
                 from c:\Users\sree2\OneDrive\Documents\Arduino\libraries\RF24/RF24.h:18,
                 from C:\Users\sree2\Downloads\2d0ea74bfdbc3521385a02dce8cf411e-d8d79e979e23b79030f203cd008be94f1c5e8212\2d0ea74bfdbc3521385a02dce8cf411e-d8d79e979e23b79030f203cd008be94f1c5e8212\nRF24L01_aattiny85_send\nRF24L01_aattiny85_send.ino:7:
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h: In static member function 'static void SPIClass::beginTransaction(SPISettings)':
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:203:5: error: 'SPCR' was not declared in this scope
     SPCR = settings.spcr;
     ^~~~
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:203:5: note: suggested alternative: 'EECR'
     SPCR = settings.spcr;
     ^~~~
     EECR
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:204:5: error: 'SPSR' was not declared in this scope
     SPSR = settings.spsr;
     ^~~~
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:204:5: note: suggested alternative: 'ACSR'
     SPSR = settings.spsr;
     ^~~~
     ACSR
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h: In static member function 'static uint8_t SPIClass::transfer(uint8_t)':
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:209:5: error: 'SPDR' was not declared in this scope
     SPDR = data;
     ^~~~
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:209:5: note: suggested alternative: 'EEDR'
     SPDR = data;
     ^~~~
     EEDR
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:217:14: error: 'SPSR' was not declared in this scope
     while (!(SPSR & _BV(SPIF))) ; // wait
              ^~~~
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:217:14: note: suggested alternative: 'ACSR'
     while (!(SPSR & _BV(SPIF))) ; // wait
              ^~~~
              ACSR
In file included from c:\users\sree2\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\io.h:99:0,
                 from c:\users\sree2\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7\avr\include\avr\pgmspace.h:90,
                 from C:\Users\sree2\OneDrive\Documents\Arduino\hardware\ATTinyCore\avr\cores\tiny/Arduino.h:10,
                 from C:\Users\sree2\AppData\Local\Temp\arduino-sketch-DEAE1529AB515C57B96B3D93CEF4201A\sketch\nRF24L01_aattiny85_send.ino.cpp:1:
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:217:25: error: 'SPIF' was not declared in this scope
     while (!(SPSR & _BV(SPIF))) ; // wait
                         ^
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h:217:25: note: suggested alternative: '_SPI'
In file included from c:\Users\sree2\OneDrive\Documents\Arduino\libraries\RF24/utility/ATTiny/RF24_arch_config.h:25:0,
                 from c:\Users\sree2\OneDrive\Documents\Arduino\libraries\RF24/RF24_config.h:72,
                 from c:\Users\sree2\OneDrive\Documents\Arduino\libraries\RF24/RF24.h:18,
                 from C:\Users\sree2\Downloads\2d0ea74bfdbc3521385a02dce8cf411e-d8d79e979e23b79030f203cd008be94f1c5e8212\2d0ea74bfdbc3521385a02dce8cf411e-d8d79e979e23b79030f203cd008be94f1c5e8212\nRF24L01_aattiny85_send\nRF24L01_aattiny85_send.ino:7:
c:\Users\sree2\OneDrive\Documents\Arduino\libraries\SPI/SPI.h: In static member function 'static uint16_t SPIClass::transfer16(uint16_t)':
[... cut ...]
  radio.setRetries(15,15);
  radio.openReadingPipe(1,pipes[1]);
  radio.startListening();
  radio.printDetails();
  radio.openWritingPipe(pipes[0]);
  radio.openReadingPipe(1,pipes[1]);
  radio.stopListening();
}

void loop(void)
{
  radio.stopListening();
  radio.write( &Command, sizeof(unsigned long) );
  radio.startListening();
  Command++;
  delay(1000);
}
`
[image]

[https://mail.google.com/mail/u/0?ui=2&ik=bd2b52005b&attid=0.1&permmsgid=msg-a:r-6797268873515825230&view=att&disp=safe&realattid=f_le4093fd0](url)

hmeijdam, 2023-02-14:
You may want to read the answer to your question on the Arduino forum first, before asking the same question here.
https://forum.arduino.cc/t/compilation-error-on-atiny-85-nrf24l01/1089593/7