SpenceKonde/ATTinyCore #30: Possible issue with EEPROM.write()

ApexMoEng, 2015-10-22:
I am trying to store some values in EEPROM and getting this error:

-----------------------------------

Arduino: 1.6.1 (Windows 7), Board: "ATtiny x7 series, ATtiny167, 8 mhz (internal), B.O.D. Disabled"

Build options changed, rebuilding all

Using library EEPROM in folder: C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\libraries\EEPROM 

C:\Program Files (x86)\Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=attiny167 -DF_CPU=8000000L -DARDUINO=10601 -DARDUINO_AVR_ATTINYX7 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\cores\tiny -IC:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\variants\tinyX7 -IC:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\libraries\EEPROM C:\Users\SWLAP2\AppData\Local\Temp\build1500510897056469719.tmp\calfDoserMain_Oct22_2015.cpp -o C:\Users\SWLAP2\AppData\Local\Temp\build1500510897056469719.tmp\calfDoserMain_Oct22_2015.cpp.o 

MODIFY_EEPROM.ino: In function 'void writeEEPROM()':

MODIFY_EEPROM.ino:5:42: error: invalid conversion from 'uint8_t* {aka unsigned char*}' to 'uint8_t {aka unsigned char}' [-fpermissive]

In file included from calfDoserMain_Oct22_2015.ino:19:0:

C:\Program Files (x86)\Arduino\hardware\ATTinyCore-master\avr\libraries\EEPROM/EEPROM.h:29:10: error:   initializing argument 2 of 'void EEPROMClass::write(int, uint8_t)' [-fpermissive]

     void write(int, uint8_t);

          ^

MODIFY_EEPROM.ino: In function 'void readEEPROM()':

MODIFY_EEPROM.ino:17:25: error: incompatible types in assignment of 'uint8_t {aka unsigned char}' to 'uint8_t [3] {aka unsigned char [3]}'

Error compiling.

-----------------------------------

I dunno. It seems odd to me that I am getting an error because it can't convert uint8_t to 'uint8_t.
have you seen this before or is this a core issue?

SpenceKonde, 2015-10-22:
What is the code that does that so I can reproduce?

SpenceKonde, 2015-10-23:
My guess, looking at the errors, is that you're passing around pointers to unsigned char arrays, when the functions only take a single unsigned char...

ApexMoEng, 2015-10-23:
Here is the code.
I thought I was pretty explicit with trying to pass a proper uint8_t to it.

```c
#include <EEPROM.h>

unsigned int addr = 0;
uint8_t timerValues[8][3] = {
  {0, 5, 0},
  {0, 5, 0},
  {0, 5, 0},
  {0, 5, 0},
  {0, 5, 0},
  {0, 5, 0},
  {0, 5, 0},
  {0, 5, 0}
};

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

void writeEEPROM() {
  uint8_t temp;
  for (byte i = 0; i < 8; i++) {

    for (byte j = 0; j < 3; j++) {
      temp = timerValues[i, j];
      EEPROM.write(addr, temp);
      addr++;
    }

  }
  addr = 0;
}

void readEEPROM() {
  uint8_t temp;
  for (byte i = 0; i < 8; i++) {

    for (byte j = 0; j < 3; j++) {
      temp = EEPROM.read(addr);
      timerValues[i, j] = temp;
      addr++;
    }

  }
  addr = 0;
}
```

ApexMoEng, 2015-10-23:
Oh, I referenced my array incorrectly...
This is what happens when you jump between languages.

Please close this issue and forget it ever happened.

SpenceKonde, 2015-10-23:
Yup, I read that and was like "wow, can you really address arrays like that in C?" 

Supposedly the comma operator serves a purpose other than making bugs harder to find, but I haven't seen a convincing use case...

closed 2015-10-23 by SpenceKonde

ApexMoEng, 2015-10-23:
I mean doing some research it seems like it is pretty well defined operator. My mistake was not realizing it was a proper operator in C.

From Wiki:
```
(,) is a binary operator that evaluates its first operand and discards the result, and then evaluates the second operand and returns this value (and type)
```

So when I wrote, timerValues[i, j]

i was evaluated, which was evaluated to i (great)
, then j was evaluated, which was evaluated to j and i was discarded, and j was returned.

Leaving me with timerValues[j] which would cause the error I got as I am incorrectly referencing
a 2D array.

The odd part, is that the compiler didn't like timerValues[j] rather than not having proper timerValues[i][j] syntex.

I would have liked an error message that clued me in to the later much better.

SpenceKonde, 2015-10-23:
Well, yes it's a well-defined operator, but it's still bloody useless except for hiding bugs. I don't think I've ever seen it used intentionally!