SpenceKonde/ATTinyCore #31: Error: "No SLEEP mode defined for this device"

luetzel, 2015-11-08:
I'm having trouble to compile a sketch for ATiny84 with the new Arduino-IDE 1.6.6. Something must have changed within the toolchain, because the sketch compiles fine with version 1.6.5. My installed ATTinyCore version is 1.0.5. The compiler stops with the error message:

In file included from /home/luetzel/Programming/Arduino16/TinyTX4/Send_DHT22_Watchdog/Send_DHT22_Watchdog.ino:8:0:
/opt/arduino-1.6.6/hardware/tools/avr/avr/include/avr/sleep.h:634:6: error: #error "No SLEEP mode defined for this device."
     #error "No SLEEP mode defined for this device."

The error has been discussed in the arduino forum (see post #43 by common_ground) as being related  to the core files.

per1234, 2015-11-08:
The forum post they are referring to is http://forum.arduino.cc/index.php?topic=357312.msg2465650#msg2465650 also http://forum.arduino.cc/index.php?topic=357312.msg2465273#msg2465273 may be relevant
ffissore's response: http://forum.arduino.cc/index.php?topic=357312.msg2466479#msg2466479

SpenceKonde, 2015-11-08:
It sounds like this is a regression in the IDE - @per1234  is that your understanding too? 

In that case, I can't do anything about it. 

For the time being, I suggest using 1.6.5 - 1.6.6 seems to have been a stinker of a release. It introduced some regressions in board manager too :-/

luetzel, 2015-11-08:
Thank you for the fast reply. I checked meanwhile the other forks of ATTinyCore from the "Unofficial list" linked above. The issue affects all of them. I'll stay with 1.6.5 and wait for the next release.

per1234, 2015-11-08:
Well if you look at sleep.h there's no sleep mode defined for ATtiny84 so the error given is expected. There is no difference in sleep.h between 1.6.5r5 and 1.6.6. Here's a sketch that causes the error:
```c
#include <avr/sleep.h>
void setup() {}
void loop() {}
```
ffissore mentioned that the build recipes in platform.txt were changed for arduino-builder and that some cores needed to have special conversions written just for them to be compatible. Arduino IDE 1.6.6 gives this warning on every compile when ATTinyCore is installed:
```
Warning: platform.txt from core 'ATtiny Classic' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
```
So it's changing `"{build.path}/{archive_file}"` to `"{archive_file_path}"`

per1234, 2015-11-08:
Ok, I did find the 84 in sleep.h so it seems that the MCU specific define is not working correctly but when I test the define in the sketch it works as expected.

per1234, 2015-11-08:
This compiles fine in 1.6.6:
```c
#define __AVR_ATtiny84__
#include <avr/sleep.h>
void setup() {}
void loop() {}
```
So that's probably the workaround for using it with 1.6.6 but of course you shouldn't have to do that so definitely a bug with arduino-builder.

SpenceKonde, 2015-11-08:
Isn't the __AVR_ATtiny84__ define supposed to be provided by the AVR toolchain? 
Where do those defines normally come from? How could that define not be present?

luetzel, 2015-11-08:
Indeed, it solves the issue when I include the line #define __AVR_ATtiny84__, both in my sketch and library.

SpenceKonde, 2015-11-08:
I'm getting really fed up with Arduino shitting all over third party devs. They haven't gone more than 2 versions without breaking custom cores....

per1234, 2015-11-08:
Here's what I've found so far:
```c
#if defined(__AVR_ATtiny84__)
#error ATtiny84 defined
#else
#endif
void setup() {}
void loop() {}
```
result: `#error ATtiny84 defined`
```c
#if defined(__AVR_ATtiny84__)
#error ATtiny84 defined
#else
#error ATtiny84 not defined
#endif
void setup() {}
void loop() {}
```
result: `#error ATtiny84 not defined`
But if I compile that with damellis attiny with ATtiny84 board selected I get: `#error ATtiny84 defined`
Uno also works as expected.
```c
#define __AVR_ATtiny84__
#if defined(__AVR_ATtiny84__)
#error ATtiny84 defined
#else
#error ATtiny84 not defined
#endif
void setup() {}
void loop() {}
```
result: `#error ATtiny84 defined`

So basically this just demonstrates what's going on in sleep.h and doesn't really tell us anything new except that it has nothing to do with includes or libraries or tools or sleep.h and it is caused by something specific to what you do in your core. The `__AVR_ATtiny84__` define is there for the `#if defined` but not there for the `#else` unless it is defined in the sketch. I'll look at it more later today and see if I can narrow it down more.

Also a note if anyone else is looking into this: `#warning` output is disabled in ATTinyCore, that's something that was confusing the issue for common_ground.

SpenceKonde, 2015-11-08:
How do I fix #warning output so it's not disabled?

per1234, 2015-11-08:
Here's the ATTinyCore platform.txt modified to use the warning level set in **File > Preferences > Compiler warnings:**
```

# Arduino AVR Core and platform.
# ------------------------------

# For more info:
# https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification

name=ATtiny Classic
version=1.0.5

# AVR compile variables
# --------------------- 
compiler.warning_flags=-w
compiler.warning_flags.none=-w
compiler.warning_flags.default=
compiler.warning_flags.more=-Wall
compiler.warning_flags.all=-Wall -Wextra

# Default "compiler.path" is correct, change only if you want to overidde the initial value
compiler.path={runtime.tools.avr-gcc.path}/bin/
compiler.c.cmd=avr-gcc
compiler.c.flags=-c -g -Os {compiler.warning_flags} -ffunction-sections -fdata-sections -MMD
# -w flag added to avoid printing a wrong warning http://gcc.gnu.org/bugzilla/show_bug.cgi?id=59396
# This is fixed in gcc 4.8.3 and will be removed as soon as we update the toolchain
compiler.c.elf.flags={compiler.warning_flags} -Os -Wl,--gc-sections
compiler.c.elf.cmd=avr-gcc
compiler.S.flags=-c -g -x assembler-with-cpp
compiler.cpp.cmd=avr-g++
#compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD
compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD
compiler.ar.cmd=avr-ar
compiler.ar.flags=rcs
compiler.objcopy.cmd=avr-objcopy
compiler.objcopy.eep.flags=-O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0
compiler.elf2hex.flags=-O ihex -R .eeprom
compiler.elf2hex.cmd=avr-objcopy
compiler.ldflags=
compiler.size.cmd=avr-size

# This can be overriden in boards.txt
build.extra_flags=

# These can be overridden in platform.local.txt
compiler.c.extra_flags=
compiler.c.elf.extra_flags=
compiler.S.extra_flags=
compiler.cpp.extra_flags=
compiler.ar.extra_flags=
compiler.objcopy.eep.extra_flags=
compiler.elf2hex.extra_flags=

# AVR compile patterns
# --------------------

## Compile c files
recipe.c.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.c.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.c.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Compile c++ files
recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Compile S files
recipe.S.o.pattern="{compiler.path}{compiler.c.cmd}" {compiler.S.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.S.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"

## Create archives
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}"

## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm
[... 17 lines cut ...]
tools.avrdude.config.path={runtime.platform.path}/avrdude.conf

tools.avrdude.upload.params.verbose=-v
tools.avrdude.upload.params.quiet=-q -q
tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} -p{build.mcu} -c{upload.protocol} -P{serial.port} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"

tools.avrdude.program.params.verbose=-v
tools.avrdude.program.params.quiet=-q -q
tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.hex:i"

tools.avrdude.erase.params.verbose=-v
tools.avrdude.erase.params.quiet=-q -q
tools.avrdude.erase.pattern="{cmd.path}" "-C{config.path}" {erase.verbose} -p{build.mcu} -c{protocol} {program.extra_params} -e -Ulock:w:{bootloader.unlock_bits}:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:{bootloader.high_fuses}:m -Ulfuse:w:{bootloader.low_fuses}:m

tools.avrdude.bootloader.params.verbose=-v
tools.avrdude.bootloader.params.quiet=-q -q
tools.avrdude.bootloader.pattern="{cmd.path}" "-C{config.path}" {bootloader.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{runtime.platform.path}/bootloaders/{bootloader.file}:i" -Ulock:w:{bootloader.lock_bits}:m

# USB Default Flags
# Default blank usb manufacturer will be filled it at compile time
# - from numeric vendor ID, set to Unknown otherwise
build.usb_manufacturer=
build.usb_flags=-DUSB_VID={build.vid} -DUSB_PID={build.pid} '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT={build.usb_product}'
```
You had it set up the way it was a while back in 1.6.0 or something when they didn't have that option yet and they were using the -w flag to disable warnings.

SpenceKonde, 2015-11-08:
Updated platform.txt for warning issue https://github.com/SpenceKonde/ATTinyCore/commit/4aaff3a5ca1c045b46d27fb3db878badaa213ab4

SpenceKonde, 2015-11-08:
Making some progress on this I think.....

SpenceKonde, 2015-11-08:
Can any of you confirm it actually works now with latest platform.txt from github?

It's compiling for me now. 

```

#include <avr/sleep.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
millis();
}

```

SpenceKonde, 2015-11-08:
But now it doesn't compile on 1.6.5, becasue of some other issue with platform.txt that the fix introduced....

SpenceKonde, 2015-11-08:
Okay, that was because I also triued to fix the warnings about the deprecated archive_file_path, which doesn't exist on 1.6.5. 

Can anyone else give this a once-over before I do a new release to pick up this fix?

per1234, 2015-11-09:
I found one archive_file_path you missed in platform.txt and after that was changed back it compiles fine with: hourly build, 1.6.6, 1.6.5r5, 1.6.4, and 1.6.3. 1.6.0 doesn't compile because of {runtime.tools.avr-gcc.path} but that wasn't added in any of the commits related to this issue.

It's annoying that the warnings about the backwards compatible recipe show any time anyone compiles or opens Boards Manager(twice!) or Library Manager. We can't change the recipes but then the users think somthing is wrong when they see the warning.

Sorry if the platform.txt I pasted above was a little messy with the commented out line. I would have taken more care if it was a PR. I see that you kept the `-std=gnu++11` that I had put in there so hopefully that was a happy accident.

Well I'm really impressed that you stuck with this and found a solution. Would you mind giving a short explanation of what the problem was?

SpenceKonde, 2015-11-09:
I had already put the -std=gnu++11 in there (as well as the fix to the defines - which was the preprocessor section). I pulled in almost everytthing from latest arduino platform.txt except things I knew should be different.

SpenceKonde, 2015-11-09:
Which is how I wound up importing the {archive_file_path} issue.

SpenceKonde, 2015-11-09:
Okay, so it looks like the current state of this core is ready for release....

closed 2015-11-09 by SpenceKonde

per1234, 2016-02-12:
The cause of this issue is explained in https://github.com/arduino/arduino-builder/issues/106

SpenceKonde, 2016-02-13:
I'm not sure I understand what you're saying -  are you saying there's
still a bug here on my end? What do I need to fix?

I thought this issue had been fixed months ago as far as my core was
concerned? Or is this still a problem for users of my core on 1.6.6 and
later?

The cause of this issue is explained in arduino/arduino-builder#106
<https://github.com/arduino/arduino-builder/issues/106>

per1234, 2016-02-13:
Sorry for the confusion. I should have explained my comment better. The bug was fixed by https://github.com/SpenceKonde/ATTinyCore/commit/7dd18f4e6ae0af23c9fbc067cee259213e154b88#diff-3fcd89411a125a3edb58066309f7357cR87 but it was never clear(at least to me) which of the changes in https://github.com/SpenceKonde/ATTinyCore/commit/7dd18f4e6ae0af23c9fbc067cee259213e154b88 fixed the problem or why the issue ocurred in the first place until I found the explanation in https://github.com/arduino/arduino-builder/issues/106 so I thought it might be useful to reference that issue here in case it might help out another project that runs into the same thing(https://github.com/digistump/DigistumpArduino/issues/13 for example).

SpenceKonde, 2016-02-13:
Aaah, okay, I get it now. Thanks