SpenceKonde/ATTinyCore #32: Compiling error for ATtiny2313/4313 @ 0.5MHz

MCUdude, 2015-11-09:
Hi!
Just tried to run a simple blink sketch using pure AVR code (tried with the Arduino blink example too), and I get this error. All other clock frequencies works fine. Any idea?

/Users/user/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/wiring.c: In function 'micros':
/Users/user/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/wiring.c:217:60: error: missing binary operator before token "("
 #if (MillisTimer_Prescale_Value % clockCyclesPerMicrosecond() == 0 ) //Can we just do it the naive way? If so great!
                                                            ^
/Users/user/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/wiring.c:220:69: error: missing binary operator before token "("
 #elif (MillisTimer_Prescale_Value == 64 && clockCyclesPerMicrosecond() == 11) // 5.75 vs real value 5.818 (11mhz) 5.78 (11.059)
                                                                     ^
/Users/user/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/wiring.c:223:69: error: missing binary operator before token "("
 #elif (MillisTimer_Prescale_Value == 64 && clockCyclesPerMicrosecond() == 12) // 5.3125 vs real value 5.333
                                                                     ^
/Users/user/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/wiring.c:226:69: error: missing binary operator before token "("
 #elif (MillisTimer_Prescale_Value == 64 && clockCyclesPerMicrosecond() == 20) // 3.187 vs real value 3.2
                                                                     ^
/Users/user/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/wiring.c:235:69: error: missing binary operator before token "("
 #elif (MillisTimer_Prescale_Value == 64 && clockCyclesPerMicrosecond() == 14) //4.5 - actual 4.57 for 14.0mhz, 4.47 for the 14.3 crystals scrappable from everything
                                                                     ^
/Users/user/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/wiring.c:241:69: error: missing binary operator before token "("
 #elif (MillisTimer_Prescale_Value == 64 && clockCyclesPerMicrosecond() == 9) //for 9mhz, this is a little off, but for 9.21, it's very close!
                                                                     ^
Error compiling.

SpenceKonde, 2015-11-09:
Update the core - I fixed that a week or two ago. I'd let a typo sneak in while syncing changes between the two cores.

SpenceKonde, 2015-11-09:
At least, I think that's what it was, I think was this one that fixed it https://github.com/SpenceKonde/ATTinyCore/commit/d711070f9e1e5dfff6b9de853fbd86c0aec429d6

MCUdude, 2015-11-09:
nope, found that bug 15 minutes ago. Removed that bracket, but the code still don't compile at 0.5MHz. 
I also updated the core.

SpenceKonde, 2015-11-09:
oooOOOOhhhh :-( 

I see the problem..... delay() and micros() never worked at under 1mhz did they?

MCUdude, 2015-11-09:
Actually, I can't compile anything right now, after I restarted Arduino with the latest core. got this strange error:

avr-gcc: error: core.a: No such file or directory
Error compiling.

Is this related to this core, or something else I've downloaded?

SpenceKonde, 2015-11-09:
MCUdude, you're using 1.6.5 or earlier and updated the Arduino AVR boards core to 1.6.9 using board manager - See my post here: http://forum.arduino.cc/index.php?topic=358313.0

This core is not effected by that problem (or it shouldn't be) if using the latest version of this core from github. (can you check that other boards in my core do compile for you, as long as they're clocked at 1mhz or more?)

I think I found a workaround for the problem at hand, but I'm gonna give myself 10 or 15 mins to try to get a fix that actually gives correct results (in 1.0.5 version of this core, as you get from board manager)

SpenceKonde, 2015-11-09:
There, this should not only compile, but actually work correctly. Please try it and let me know if it works for you (particularly if you're running it on real hardware - I'd love to know that blink blinked at the right speed now...)

SpenceKonde, 2015-11-09:
(in other words, don't let the fact that this issue is closed dissuade you from commenting) :-)

MCUdude, 2015-11-09:
Thanks! the code compiles, but delay(1000) gives approximately 2 seconds of delay at 0.5MHz. No problems at 1MHz.

So, whats the deal about the core.a error? I'm not interested in upgrading to IDE v1.6.6, as it's been reported as buggy.

MCUdude, 2015-11-09:
Oh, and Why not add a slow version of some of the programmers. a 250kHz ISP programmer won't work at AVR down on 0.5MHz and 128kHz. I'm using this from another core. Haven't tested it at 128kHz though.

usbtinyisp2.name=USBtinyISP SLOW
usbtinyisp2.protocol=usbtiny
usbtinyisp2.program.tool=avrdude
usbtinyisp2.program.speed=8
usbtinyisp2.program.extra_params=-B{program.speed}

SpenceKonde, 2015-11-09:
The 1.6.9 version of the standard AVR boards platform.txt uses the {archive_file_path} substitution pattern to specify the location to output the core.a file to... but the 1.6.5 IDE doesn't have that substitution pattern, so the core.a file instead ends up in a file called {archive_file_path} in the same location as arduino.exe (or gets stuffed into the UAC virtual store under that name if you're on windows and it's on default settings and arduino is installed under program files). Since they later refer to it the old way, they can't find the file at that point in time. 

Creating separate issue for programmers issue. 

Reopening as issue not fixed.

SpenceKonde, 2015-11-09:
Hm.... I can't reproduce that issue with an up-to-date version of the core. Blink is running on my 4313 @ 0.5mhz and it's running at the right speed.

SpenceKonde, 2015-11-09:
https://github.com/SpenceKonde/ATTinyCore/issues/33 < programmer speed issue

MCUdude, 2015-11-09:
Huh,  that's weird. I'll test some more tomorrow. It's 02.18 AM here. ;)

SpenceKonde, 2015-11-09:
Well, then, this is what's gonna be in the next board manager release. Due to the 1.6.6 bullshit, I need to rush out a release today, since you cant use sleep functionality under 1.6.6 with most recent released via board manager.

MCUdude, 2015-11-09:
OK, I tested with both ATtiny2313 and 4313, and the delay is twice as long on both microcontrollers at 0.5MHz. I also tried the AVR _delay_ms() method, and that one is spot on.

SpenceKonde, 2015-11-10:
1.0.6 is now out. Does it fix the problem?

MCUdude, 2015-11-10:
The delay() method is now correct @ 0.5MHz! :) What did the trick?

I'm using the board manager right now, but I prefer to download from Github instead. Will I still get the core.a error if I ditch the boards manager install, and download from Github instead?

MCUdude, 2015-11-10:
(Where should I set the bar for reporting an issue? There are some small typos around (mhz instead of kHz, lower case M and H in MHz etc. Dre these enough to be reported? )

SpenceKonde, 2015-11-10:
they use a macro clockCyclesPerMicrosecond() - but you don't want to introduce floats in the core, because that adds a bunch of crap to the flash to do basic arithmetic with floats, so they set it to 1. I still do that (because wtf else am I to do?) but in micros() which is used for timing delay(), I specifically test for cases where F_CPU < 1 mhz and adjust it there. 

Error is with core.a is fixed in all versions, github or normal. 

Report all issues.

commit b5bd2a59 (2015-11-09): fix #32 (compile failure where F_CPU < 1mhz)
closed 2015-11-10 by SpenceKonde