SpenceKonde/ATTinyCore #37: PWM issue on 2313 and pin 12 + 13

battosai30, 2015-12-04:
Hi,

analogWrite() simply does not work on 2313 and pin 12 and 13. Using this code in arduino it works : 

void setup() {
  // put your setup code here, to run once:

  /* Set Fast PWM mode. */
TCCR1A |= (1<<WGM00) | (1<<WGM01);
/* Clear 0C0A on compare. */
TCCR1A |= (1<<COM1A1);
/* Start timer, no prescaling. */
TCCR1B |= (1<<CS00);
/* Duty cycle 0% */
OCR1A = 0;

//DDRB |= (1<<PB3);
pinMode(12,OUTPUT);
}

void loop() {
  int i;
 for(i=0; i < 255; i++) {
      OCR1A = i;
      delay(10);
  }
 
  for(i=255; i > 0; i--) {
      OCR1A = i;
delay(10);
  }

}

But if I upload fade example on pin 12 or 13 it does not work

SpenceKonde, 2015-12-04:
It looks like INITIALIZE_SECONDARY_TIMERS is set to 0 for the x313. I didn't do it (it was like that when I forked the repo) - per comments it was done to reduce flash use (since if you don't use PWM on that pin, the initialization of those timers is pure overhead)... 

I'll see how much flash it actually saves and assess what the right way to resolve this is. The 2313 is extremely tight on flash, as you've no doubt noticed, so I can see why they might have done this.

In the meantime, you can use analogWrite() on those pins normally if you set the TCCR's manually in setup().

battosai30, 2015-12-04:
Thanks !! I tried to change it before, but in wiring.c, I forgot to look in pin_energia.h. 

And you were right, changing it to 1 allowed PWM on 12 and 13 !!

Regards

SpenceKonde, 2016-03-14:
I wonder if this could be done with a menu option...

commit 528a6912 (2016-03-17): Merge pull request #45 from per1234/INITIALIZE_SECONDARY_TIMERS-menu

x313 Initialize Secondary Timers menu. I may fiddle with the wording, but that's good enough for me. Fix #37.
closed 2016-03-17 by SpenceKonde