SpenceKonde/ATTinyCore #39: Update EEPROM library

SukkoPera, 2016-01-18:
The EEPROM library has lagged behind Arduino's. More and more sketches are using the newer methods (update(), put(), get()), and updating the library would make them more compatible with ATtiny's.

At a quick glance, the library should't require many changes, it might even compile as-is.

SpenceKonde, 2016-01-18:
Dang, I didn't even know that stuff had been added...

SpenceKonde, 2016-03-14:
Need to benchmark compile size for simple operations and determine whether they can just be swapped out. The reason not to would if the new library has a larger overhead associated with it for the legacy calls. In that case, I'll pull it in, but call it EEPROM2

SukkoPera, 2016-03-14:
Yeah, or you can just implement a few wrappers for the new functions, just to be compatible with the updated interface.

SpenceKonde, 2016-03-14:
As I said, I haven't benchmarked compiled code size. If there is significant overhead, though, I expect that the overhead will be inherent to the new functionality, and not separable from it.

commit 272e4f2a (2016-03-26): Fix #39

Turns out the new library is *more* space efficient. Awesome!
closed 2016-03-26 by SpenceKonde