SpenceKonde/ATTinyCore #40: Hardware Serial

lnxbil, 2016-02-08:
Hi,

Why are the hardware serial pins hardcoded? Do they use USI?

Best,
Andreas

SpenceKonde, 2016-02-08:
Chips with actual hardware serial (in this core, x313, x7) have pins fixed by the hardware.

The software serial on chips that don't  uses the analog comparator interrupt (on analog comparator pins) instead of a pcint, so they don't conflict with other libraries that use pcint. You can still use a library like SoftwareSerial to get software serial on other ports.

lnxbil, 2016-02-08:
Thank your for the fast answer. I didn't specify which Tiny I meant, so you're right. I'm using the library on the ATtinyX5 chips and I wanted to use the build in ```Serial```, which is IMHO not remapable without changing the sourcecode. It's every convenient to have the "real" thing also on the Tinies.

In the end, I used SoftwareSerial to get serial working, because I wanted to use TinyWireM to simulate a I2C Slave, which uses USI on the SDA/SCL (PB0/PB2) lines.

closed 2016-03-13 by SpenceKonde