SpenceKonde/ATTinyCore #50: Highlight the pinout even more

MCUdude, 2016-03-24:
Merging your two ATtiny cores is a great idea! But now that you got support for SO many ATTinys, you (or maybe me) should unify all the pinout pics. Have a look at how I did it with the [MightyCore](https://camo.githubusercontent.com/daa487cc1378d8a61882e622086c9f957b81ad5e/687474703a2f2f692e696d6775722e636f6d2f3963706654476c2e706e67) and the [MegaCore](https://camo.githubusercontent.com/73c1422b0a2f0e9c2d36d85452bdbdd4e444e26b/687474703a2f2f692e696d6775722e636f6d2f3554684542386f2e6a7067). If you want I can make similar pinouts for this core, I just need a little time :wink:

sleemanj, 2016-03-24:
I made (less pretty) diagrams for the x4 and x5 series in my ATTinyCore fork a while back, and for the t13 also (but havn't stored it in that fork for some reason I forget).

[x4 with reverse numbering (what @SpenceKonde ATTinyCore uses)](https://github.com/sleemanj/ATTinyCore/blob/master/avr/variants/tinyX4/Pin_Mapping_ATtiny24_44_84.svg)

[ x4 with forward numbering (what @damellis attiny uses)](https://github.com/sleemanj/ATTinyCore/blob/master/avr/variants/tinyX4/Pin_Mapping_ATtiny24_44_84.svg)

[x5](https://github.com/sleemanj/ATTinyCore/blob/master/avr/variants/tinyX5/Pin_Mapping_ATtiny25_45_85.svg)

[t13](https://github.com/sleemanj/optiboot/blob/master/dists/diy_attiny/avr/Pin_Mapping_ATtiny13.svg)

SpenceKonde, 2016-03-25:
I fully agree with you on the need for decent pinout images. ( #25 ) - the main reason I haven't done it is that I'm not talented with visual design, like, at all. It would be **awesome** if you'd be willing to make diagrams like that for me - I'd be happy to send you a few freebies from my tindie store as thanks. There's sort of a lot of diagrams to make now...
1634
x4- apparently we need the numbers going in both directions now? 
x41 - same as x4, only with more peripherals. I think this needs to go in both directions too, if the @damellis core going the other direction :-(  
828 (it's not the same as 88 or mega x8 - like, except for the power pins)
x5
x61
x7
x8
x313

@sleemanj Aw fuck, is the @damellis core using the other pin mapping? I guess I need a menu option for that... I don't want people to have to change their sketch when switching from his lousy core to my awesome one...

MCUdude, 2016-03-25:
I don't think you should follow damellis' footprints. His core doesn't got a readme (like yours got) and no pinout diagrams either. I don't see why people should use his core instead of yours. Multiple pinout options = more mess and more to keep track of for the end user. Stick with the best (and most logical) pinout and instead document the heck out of it :smile:

SpenceKonde, 2016-03-25:
Well, damellis is one of the founding arduino people, which gives his core some gravitas. Otherwise I'd be like "screw him"...  The fact is that one can make every bit as strong of an argument (maybe stronger) for David's pin mapping. I picked mine solely for backwards compatibility with the old tiny core - by which token it's very hard to justify not offering a way to use that pinout. If I had known he was going to use the other pin mapping when I started this core, I would have copied him. But now I'm locked into offering that to avoid breaking people's sketches. 

Anyway - this doesn't matter for the purposes of pinmapping images - I'm capable enough with paint that I could take an x4 diagram and make an x4-damellis-pinout diagram from it - that's not the hard part. It's getting pinout images in the first place - ideally ones that match eachother.

sleemanj, 2016-03-25:
@SpenceKonde @damellis - Yes the attiny "core" by damellis uses the [reverse digital pin assignments to ATTinyCore](https://github.com/damellis/attiny/blob/master/variants/tiny14/pins_arduino.h#L35) for the 14 pin variants. 

In my fork you can see I just added an additional variant "tinyX4_Arduino_Numbering"
https://github.com/sleemanj/ATTinyCore/tree/master/avr/variants/tinyX4_Arduino_Numbering
(I never got around to adding it into the boards.txt though I think)

---
PS: I prefer damellis ordering, because with SpenceKonde ordering if you are using an external xtal, the available digital pins start from 2, instead of 0 (because 0 and 1 are taken by XTAL).  With damellis the unavailable pins are at the top end (9 = XTAL2, 10 = XTAL1 and 11 = RESET),

In my "diy_attiny" setup which uses my fork I use the tinyX4_Arduino_Numbering only
https://github.com/sleemanj/optiboot/tree/master/dists#attiny 
firstly because I prefer it, and secondly because a lot of people will start out with the HighLowTech stuff and want to move to something based on ATTinyCore to get more features (or better optimisation).

SpenceKonde, 2016-03-26:
@sleemanj  - yeah, that's sort of my thinking on the pinouts - I didn't know the other pinout was being widely used, like I said, else I'd have used that. My prime directive is compatibility - programers' time and thought is the most valuable resource for most people and organizations I think, so I want to treat the users' time with respect, and not make them go through their code fixing the pin numbers (Yes, they should have used #defines at the top of their sketch, but people do all sorts of dumb shit. And I can imagine cases where people, desperately cramming code into the limited space, would do ugly things like for loops with pin numbers and other awful stuff like that).

SpenceKonde, 2016-03-26:
Added the option for x4 - Any thoughts on what to call the two options?

MCUdude, 2016-03-26:
I'll figure out a nice way to display both pinouts (24/44/84 & 241/441/841) in the same pinout diagram. What about pinout names such as "logical", "clockwise" or/and "counter clockwise"? Personally I'm a fan of pinouts that matches the port pins as close as possible.

Here's a first draft for a pinout diagram for the ATtiny25/45/85. What do you think @SpenceKonde?
[image]

SpenceKonde, 2016-03-26:
I like it. 

The bar over the inverted timer should be clearer - it vanishes if you zoom out a bit - it's important, since it means no PWM on that pin unless you've got them set up in some bizarro mode for driving motors (maybe a footnote mentioning that the inverted output isn't normally usable?) I wonder if the Tone pins should be marked (ie, the pins that will use the output compare for tone, and which won't work for PWM if you're using Tone) - that's OC1B on the x5, and OC1x on everything else, unless I did something weird somewhere.

MCUdude, 2016-03-26:
I've made the bars a little bit thicker, and I should probably add a footnote about the inverted timers. 
About the tone library - I personally don't think it should be marked. I prefer a pinout diagram to be as clean and neutral as possible. The library should inform the user what pins to use. How about adding some `#if defined(__MCU__)` in the examples instead?

MCUdude, 2016-03-26:
Like this?
[image]

jassing, 2016-03-26:
Why not ask permission to re-use pighi's?
http://pighixxx.com/attinypdfv3_0.pdf

MCUdude, 2016-03-26:
He just got pinouts for the ATtinyX5 and ATtinyX4. There's a lot more to create, and I prefer to use the same style for all of them :smile:

SpenceKonde, 2016-03-27:
mmmm... Let's say the damellis version is default - this will represent a change from the current version. But I think his pinout does make more sense as the default. I need to aggressively document that change.

SpenceKonde, 2016-03-28:
Oh! On chips that do *not* have serial ports (ie, everything except the x313, x41, x7, 828, 1634), there is a built-in software serial implementation available on the analog comparator pins. Not sure if you've seen this in readme:

```

The ATtiny x4, x5, x61, and x8 chips do not have hardware serial. For these parts, a software serial is included. This uses the analog comparator pins (to take advantage of the interrupt, since very few sketches/libraries use it, while lots of sketches/libraries use PCINTs); the serial is named Serial, to maximize code-compatibility. TX is AIN0, RX is AIN1. This is a software implementation - as such, you cannot receive and send at the same time. If you try, you'll get gibberish, just like using SoftwareSerial.

```

MCUdude, 2016-03-28:
Thanks! I'll add this as well :smile:

SpenceKonde, 2016-04-11:
How's this been doing? I'm planning to do the 1.1.0 release this week, so if you've got any ready, I'd love to put them into the release. Thanks

MCUdude, 2016-04-12:
I'm sorry, but I've been really busy with school after the easter break. I'll continue to work on it when I got some spare time, but I don't think i'll be able to finish it this week :disappointed:

SpenceKonde, 2016-04-12:
Okay, thanks for the update, I've marked this as not being part of 1.1.0 - 1.1.0 will go out without pinout charts, and they'll be picked up in 1.1.1.

MCUdude, 2016-04-13:
I've finished pinout for ATtiny25/45/85, ATtiny24/44/84, ATtiny241/441/841 and ATtiny2313/4313. I'm now working with the ATtiny1634, but I can't find a pinout diagram for it anywhere. Do have one I can use?

SpenceKonde, 2016-04-13:
Awesome - glad to hear it! Can't wait to see them

The closest I have to a pinout diagram is the one I've been working off of - which of course looks like ass. What sort of image are you looking for? 

http://drazzy.com/e/img/Tiny1634.jpg

MCUdude, 2016-04-13:
Oh, sorry! I meant the ATtinyX61 series

SpenceKonde, 2016-04-13:
Uhh.... err.... 

```

// ATMEL ATTINY861
//
//                   +-\/-+
//      (D  9) PB0  1|    |20  PA0 (D  0)
//     *(D  8) PB1  2|    |19  PA1 (D  1)
//      (D  7) PB2  3|    |18  PA2 (D  2) INT1
//     *(D  6) PB3  4|    |17  PA3 (D 14)
//             VCC  5|    |16  AGND
//             GND  6|    |15  AVCC
//      (D  5) PB4  7|    |14  PA4 (D 10)
//     *(D  4) PB5  8|    |13  PA5 (D 11)
// INT0 (D  3) PB6  9|    |12  PA6 (D 12)
//      (D 15) PB7 10|    |11  PA7 (D 13)
//                   +----+
//

```

Old pinout:
```
// ATMEL ATTINY167
//
//                   +-\/-+
// RX   (D  0) PA0  1|    |20  PB0 (D  4)
// TX   (D  1) PA1  2|    |19  PB1 (D  5)
//     *(D 12) PA2  3|    |18  PB2 (D  6)
//      (D  3) PA3  4|    |17  PB3 (D  7)*
//            AVCC  5|    |16  GND
//            AGND  6|    |15  VCC
// INT1 (D 11) PA4  7|    |14  PB4 (D  8)
//      (D 13) PA5  8|    |13  PB5 (D  9)
//      (D 10) PA6  9|    |12  PB6 (D  2)* INT0
//      (D 14) PA7 10|    |11  PB7 (D 15)
//                   +----+
//
// * indicates PWM pin.

```

New pinout:

```

// ATMEL ATTINY167
//
//                   +-\/-+
// RX   (D  0) PA0  1|    |20  PB0 (D  8)
// TX   (D  1) PA1  2|    |19  PB1 (D  9)
//     *(D  2) PA2  3|    |18  PB2 (D 10)
//      (D  3) PA3  4|    |17  PB3 (D 11)*
//            AVCC  5|    |16  GND
//            AGND  6|    |15  VCC
// INT1 (D  4) PA4  7|    |14  PB4 (D 12)
//      (D  5) PA5  8|    |13  PB5 (D 13)
//      (D  6) PA6  9|    |12  PB6 (D 14)* INT0
//      (D  7) PA7 10|    |11  PB7 (D 15)
//                   +----+
//
// * indicates PWM pin.

```

SpenceKonde, 2016-04-13:
I'm on Windows. Not sure if PDF or full-res PNG is better for this. Once I have a nice full-res version of it, I can resize it to fit smaller areas if appropriate.

MCUdude, 2016-04-13:
What pinout version do you want to stick with? Personally I prefer the "new" pinout :smile: 
Is the plan to have one pinout for the X61 series, and two for the X7 series, or do you want to use the "new" pinout for both series?

Oh, here's a sneak peek:
[image]

SpenceKonde, 2016-04-13:
ATtiny 861 has one pinout (shown above)

ATtiny 167 has two pin mappings. Compatibility (old) and new (down the sides, the much better one), because someone released a core with a different, much stupider pin mapping. I think the installed base is very small here, so I may actually separate them into two images, and bury the compatibility one deeper in the documentation (as I noted below, hiding one set of pinouts is something I can do myself at the image level). 

I'm still not sure whether I'll expose both pinouts on the 841/441 - I need to keep the old one default because that's what's printed on the boards I'm selling. But I can handle hiding one of the pinouts from the image if need be (that's MS-paint stuff, which is about the limits of my graphics skills).

SpenceKonde, 2016-04-13:
Those ones for the Tiny85/84/841 look *beautiful* - I love it.

MCUdude, 2016-04-13:
Good to hear! :smiley: About the X61, What should I do about the i2c bus? I i've understood correctly it actually got two busses (no swap) but can't run in master and slave mode simultaneously. which one is the default one?
[image]

SpenceKonde, 2016-04-13:
The 861 can use either set of pins for the USI (including i2c or spi /
three wire mode) but default is PB0~PB2.

There's a register to change it, like how the 841 can move spi and pwm
around

Good to hear! [image: :smiley:] About the X61, What should I do about the
i2c bus? I i've understood correctly it actually got two busses (no swap)
but can't run in master and slave mode simultaneously. which one is the
default one?
[image: skjermbilde 2016-04-14 00 40 53]
<https://cloud.githubusercontent.com/assets/3238730/14511695/a93b3870-01d9-11e6-9fe2-3e403d902573.png>

MCUdude, 2016-04-14:
It's an easy fix for me to remove the alternative X4 / X61 pinouts. Just tell me which one you want to keep

MCUdude, 2016-04-14:
One more question; What PWM pins on the ATtiny87/167 are supported i this core? I've never seen names like OC1AW and OC1BV before. Are these just regular timer outputs?

SpenceKonde, 2016-04-14:
Re: the x61 pinout you posted above - the pin numbers you have don't match the pin mapping

```

// ATMEL ATTINY861
//
//                   +-\/-+
//      (D  9) PB0  1|    |20  PA0 (D  0)
//     *(D  8) PB1  2|    |19  PA1 (D  1)
//      (D  7) PB2  3|    |18  PA2 (D  2) INT1
//     *(D  6) PB3  4|    |17  PA3 (D 14)
//             VCC  5|    |16  AGND
//             GND  6|    |15  AVCC
//      (D  5) PB4  7|    |14  PA4 (D 10)
//     *(D  4) PB5  8|    |13  PA5 (D 11)
// INT0 (D  3) PB6  9|    |12  PA6 (D 12)
//      (D 15) PB7 10|    |11  PA7 (D 13)
//                   +----+
//

```

The pinout is weird, but it's not as wierd as it would be if it were what you posted. Didn't notice that yesterday, since I was on my phone (girlfriend date night). 

Regarding the pin remap options for the x61 USI, I'd mark them with an asterisk like was done on the x41, and make sure that it was noted that that indicates a remap option. Unless you can think of a better way to mark the remap options? In any event, the default one on x61 is on PB0:2

Pins 2, 11, and 14 (by new pinout) are supported for PWM on the x7 series (see the asterisks next to the pin numbers in the ugly ascii diagrams)

SpenceKonde, 2016-04-14:
(the other OC1Ax pins on the 167 are essentially remap options)

MCUdude, 2016-04-14:
I hadn't finished the Arduino pins on the last screenshot. It looks like this now:
[image]
Is this correct? Are the i2c pin the only one that's swappable. I have no experience with these modern ATtinys :wink:

SpenceKonde, 2016-04-14:
Miso/mosi/sck are too (its the whole usi that gets moved, so spi and i2c
are effected)

MCUdude, 2016-04-14:
Ok, so DI = alternative MISO, DO = alternative MOSI, but what about USCK - alternative SCK. Is it PA5 or/and PB2?

SpenceKonde, 2016-04-14:
USCK (which serves as SCK when the USI is in SPI mode and SCL when USI is in I2C mode) is on PB2 by default, with option to remap to PA2

MCUdude, 2016-04-14:
I think I'll need to do some reading about USI.. It makes the pinout a whole lot more complex, and all those different names, oh my.. :disappointed:  The only problem by adding all these swappable pins and this USI stuff is that it makes the whole pinout diagram much bigger, and even overwhelming for some. I've already skipped functions such as ICP, AIN, CLKO. Is all these swappable USI pins really necessary to know about to use this ATtiny? Just a thought :wink:

SpenceKonde, 2016-04-14:
Hmm, yeah - It might make more sense to just not list the remap options at all, since they're not something most people will be using anyway. In that case, they should come off the 841 as well, in the interest of keeping the diagram readable

The idea with the USI is that there's one internal interface for I2C or SPI, and you can set it to be one or the other (it's a cost-cutting measure).

MCUdude, 2016-04-14:
I'd much rather write a bottom text under the microcontroller saying "Pin swapping possible more info in the datasheet" or something. It makes all these diagrams easier for me to create while still containing all the information an average user would need

SpenceKonde, 2016-04-14:
That sounds like a good solution - "Some pin functionality can be remapped to other pins, see the datasheet for more information" or something like it. (Atmel calls it remapping )

SpenceKonde, 2016-04-15:
Was re-reading the datasheet for the ATTiny167. Here's what's weird - The 167 *also* has a USI, and has pins marked MISO/MOSI/SCK, and the USI can be remapped to use two sets of pins, one of which is the same as the MISO/MOSI/SCK pins - However, on the 167, the USI on  is *in addition to* the SPI interface, and defaults to using the other set of pins, not the ones used by SPI - so you could use the USI and the SPI to have two separate SPI busses on the same chip, or an SPI and an I2C. Weirdo chip.

MCUdude, 2016-04-24:
Sorry I've been busy this week, but I try to work on this when I got some time left. About the PWM output pins om the ATtinyX7, what are the default output pins? I've added PA2, but aren't there any others?

SpenceKonde, 2016-04-24:
You can output PWM on those other pins, but the PWMs are not independent - There is one OC1A, and one OC1B - you can just pipe each one to any of 4 pins. So, we only *really* have three PWM pins. I think there's some obscure application they had in mind for moving the PWM around between pins (you can individually set and unset each output channel), but I'm not sure what it is.
The x7 series is weird.

MCUdude, 2016-04-24:
Thanks! I've not done all the pinouts yet, but what do you think so far?
[ATTiny_pinout.pdf](https://github.com/SpenceKonde/ATTinyCore/files/233627/ATTiny_pinout.pdf)

SpenceKonde, 2016-04-25:
That looks great! So it's just the 88 and 828 left?

On the x5, ADC1 is marked ADC0.

MCUdude, 2016-04-25:
Yes, 88, 828 and 1634 left 😉 
I've corrected the AYtinyX5 error

SpenceKonde, 2016-05-03:
How's this coming along?

If you don't see yourself having time to get the remaining ones wrapped up soon, can you send me current version of the ones you have so I can put them up and do a release this week? 

Thanks

MCUdude, 2016-05-03:
Sorry, I've been really busy the last couple of weeks. I just got the ATtiny88 and the ATtiny828 left, but I don't know exactly  when I can manage to finish them up. Here are the current version of the pinouts!
[ATTiny_pinout.zip](https://github.com/SpenceKonde/ATTinyCore/files/247891/ATTiny_pinout.zip)

per1234, 2016-05-03:
I'd like to make a suggestion that the original files for these images be added to the repository also, maybe as a separate branch to keep the core download file size small. The reason is this will make it easier for anyone to make changes to the images. If the originals and jpegs were both stored in the images branch and then links to those jpegs added to the master branch's README.md then the edits could be submitted as a single pull request.

SpenceKonde, 2016-05-06:
Hmm, I'll look into that. In the short term, I may just throw them onto drazzy.com - they're going to be referenced from quite a few places. Does github let you hotlink to images from all over hell? I somehow feel like they probably wouldn't like that.

sleemanj, 2016-05-06:
See RawGit 

 https://rawgit.com/ 

(I'm not affiliated, just a user thereof). 

 

Links:
------
[1]
https://github.com/SpenceKonde/ATTinyCore/issues/50#issuecomment-217576099

SpenceKonde, 2016-05-09:
Awesome - got them up. 

The 167 has an alternate pin mapping for backwards compatibility with other cors. 

For now, I just threw them onto my website in the interest of time - I'll look into how to keep them in github - but part of the complicating factor is that the files have to be converted into multiple different formats and sizes, and I'm not sure how I'd do that with a github storage solution, and don't have time to figure it out now.

SpenceKonde, 2016-05-09:
For the 85, 84, 861 and 88, the AIN0 and AIN1 pins need to be listed, because those are TX and RX pins (respectively) for the built-in software serial.

MCUdude, 2016-05-10:
That's fine! I also did some other minor changes (like indicating the alternative pinout on X4 and X41)
[ATTiny_pinout.zip](https://github.com/SpenceKonde/ATTinyCore/files/258267/ATTiny_pinout.zip)

SpenceKonde, 2016-05-11:
Thanks much for your work on this so far, I updated the 4/5/61 images in the readme and on my website. 

Pushing remaining improvements (alternate layout for x7-series, 88 and 828) to version 1.1.2

MCUdude, 2016-05-13:
Here's the pinout pic for the ATtiny48/88 😃 
[ATtiny48_88.zip](https://github.com/SpenceKonde/ATTinyCore/files/264165/ATtiny48_88.zip)

SpenceKonde, 2016-05-22:
Thanks! 

ATTiny841 has a few problems - there are 12 ADC channels, only and there's PWM on PA3 and PA4 (OC2B, OC2A respectively)

MCUdude, 2016-06-19:
Oh, I forgot all about this! I'll fix it soon 😃

MCUdude, 2016-06-19:
Here is the "new" ATtiny441/841 pinout 😉 
[ATTiny_pinout.zip](https://github.com/SpenceKonde/ATTinyCore/files/322225/ATTiny_pinout.zip)

[17 more replies]

closed 2016-08-13 by SpenceKonde