SpenceKonde/ATTinyCore #51: Delay on x8 way off

kontakt, 2016-03-26:
I'm using an Attiny88 at 8MHz internal, and I started off with the standard blink sketch. The blink is way off, something on the order of 10-20 times faster than anticipated.

SpenceKonde, 2016-03-26:
Hm, I cannot reproduce this. I just uploaded blink to my ATTiny88 @ 8mhz, and it's blinking at the right speed. I also can't think of a way that blink could be *faster* than expected.... If you haven't done burn bootloader (hence it's still running at 1mhz), it would blink at the wrong speed - but slower, not faster.

kontakt, 2016-03-26:
Yeah it's strange. I ran an exponential time loop with the delay increasing each pass and it's definitely off. I'll set up another arduino to measure the timing and see what I get.
[Here's a video](https://youtu.be/URP0D2XdYTI)
I'm using the standard setHigh/delay/setLow/delay blink sketch with delay(1000) in this video.

SpenceKonde, 2016-03-26:
I'm looking at that board in the video, and I can't see the decoupling capacitors - do you have them installed between power and ground right next to the chip? If not, that is probably the problem - it's resetting when it tries to turn the LED on. Often without caps, they won't even program.

kontakt, 2016-03-26:
Well there we go. Put them on my PCB design but didn't think about it during breadboarding. I really appreciate your help!

closed 2016-03-26 by kontakt