SpenceKonde/ATTinyCore #53: Consider "improved" versions of standard libraries (I2C, SPI)

SpenceKonde, 2016-03-28:
What if there were a Wire library that looked like the normal Wire library to the software, but would automatically (using ifdefs ) use the right library (Wire for 48/88, TinyWireM (USI-based) for ones with a USI, and SoftI2CMaster for ones without it). Likewise for SPI library - it could use the USI if applicable.

jassing, 2016-03-28:
that sounds like a good plan to me...

sleemanj, 2016-03-28:
I think it is better that a variant explicitly states which version of SPI it wants, auto-detecting is usually somewhat crusty, especially since you have variant files right there.

There already exists "USE_SOFTWARE_SERIAL" for example - I took that further in my fork by making the variant define `USE_SERIAL_TYPE` to [specify the type of serial it supports](https://github.com/sleemanj/ATTinyCore/blob/master/avr/variants/tiny13/pins_arduino.h#L71)

    // #define USE_SERIAL_TYPE           SERIAL_TYPE_HARDWARE
    // #define USE_SERIAL_TYPE           SERIAL_TYPE_SOFTWARE
    #define USE_SERIAL_TYPE              SERIAL_TYPE_HALF_DUPLEX
    // #define HALF_DUPLEX_SERIAL_DISABLE_WRITE
    // #define HALF_DUPLEX_SERIAL_DISABLE_READ

the core files just check USE_SERIAL_TYPE to determine what to do (SERIAL_TYPE_xxxxx are just unique constants defined in Arduino.h)

SukkoPera, 2016-05-18:
I second this: it would make code more portable across different chips.

If somebody doesn't like it, he/she can still user the specific library, so I really see no cons.

I had tried to make a similar library for SPI, but then suspended work because I needed SPI slave functionality, which didn't look implemented in the official Arduino library.

SpenceKonde, 2016-05-18:
I would only target SPI/I2C master configurations for this functionality. 

The reason being that for slave configurations, you typically do not have libraries that #include the SPI/I2C library - so using a different library only requires making changes to your sketch. For SPI/I2C master, any library that supports an SPI or I2C device has to #include Wire or SPI library - meaning that using the library with a different SPI or I2C library requires modifying the library, which complicates development. 
SPI/I2C slave is also much less popular than SPI/I2C master within Arduino land.

SpenceKonde, 2016-09-14:
I'm going to move this to 1.1.3 - I think I'm going to do a release sooner than I'll be able to get this sorted out.

cardre, 2017-02-09:
As this issue has been open for a while and I need SPI USI capability for an ATTiny1634 project, does anyone else have something working yet?

Otherwise, I'll try and hack something together, but it may not be suitable for a pull request. I've found the following http://playground.arduino.cc/Code/USI-SPI and got it to compile ok after adjusting it for the ATtiny1634 pins I was using, but I haven't confirmed if it is actually working yet as I have a different SPI device I need to talk to. I actually want to use two different SPI busses on the ATtiny 1634 so may need to do some extra work in reconfiguring the USI when switching between the two busses.

cardre, 2017-02-11:
As there's been no comment/reply (so far) I'm guessing I'll need to get started on this as I need this USI based SPI master functionality on a ATTiny1634.

In looking at other code/libraries, it appears there's just a whole series of #ifdef statements for each of the ATTinys to try and define functionality that does/doesn't exist per chip. Is there a better way to check for existence of USI vs. native SPI, or should I just use the same approach, i.e. in my case an #ifdef for the ATTiny1634?

My goal here is to modify the SPI library that's included in this repository to make it work for the USI devices supported in ATTinyCore and if good enough, issue a pull request for it. But I'd be keen for feedback to know if this is the right approach as opposed to an external library such as TinyWireM?

SpenceKonde, 2017-02-11:
Sorry, I typed out a reply. I don't know why it's not here, will update this comment shortly - I have a lot of the work (possibly all of it) done for SPI already, but never tested.

SpenceKonde, 2017-02-11:
Okay, I found the code, and with some wrangling, got it to compile for both normal ATTiny's and those lucky few that have real SPI. 

This is in the Unified-SPI branch - give it a whirl. You'll surely find issues, since I haven't tested it, but I think it's most of the way there. 

And yeah, you need a bunch of #ifdefs - for the big picture ones, you can check whether register names are defined (see how I check SPDR and USICR to check for SPI and USI respectively. Atmel is kind enough to give us consistent peripherals across their entire AVR line, so if you check for SPDR and find it, you can be confident that you've got something that'll act like a normal SPI peripheral. And so on. So it's really not that bad... SPI and USI are friendlier in this way than the timers, where it seems like they put different spellings of the register names into a hat, and picked one for each new chip they released

cardre, 2017-02-11:
OK, thanks for providing this. I'll review in the next couple of days and see if I can get it running. I'm trying to support the LowerPowerLab RFM69 radio library on this chip and will use this as a test case for getting the USI based SPI to work.

SpenceKonde, 2017-02-13:
Awesome! I keep hearing about those thingamabobs - personally I avoid leaving WiFi range so I haven't done any of the long range stuff myself. I'd love to hear about how you get on with it.

cardre, 2017-03-08:
OK, just a quick note to let you know that I know have updated the code to compile and work with the USI device to communicate with SPI on an ATtiny1634. As I don't have any other ATtiny devices, I can't confirm whether it would work, but I think it should be just a matter of defining the ports and pins used for USI in an #elif set of defines for each ATtiny supported by this library. I'll need to work out how to do a pull request for the Unified-SPI branch and then get this merged back into main. FYI, it is being used to control (obviously via SPI) an RFM69 radio module.

SpenceKonde, 2017-03-08:
This is excellent news! Thank you! I'd love to see that PR - especially as we've got other people using the core asking for this.

SpenceKonde, 2017-03-17:
@cardre - Any update? I'd love to see what was needed to get SPI working. If you're being held up by difficulties with sorting out how to submit a PR against Unified-SPI branch, feel free to like, zip up and email the relevant files or something (spencekonde@gmail.com).  Thanks!!

SpenceKonde, 2017-04-15:
@cardre - *nudge* Any way I could get that code from you?

MCUdude, 2017-04-17:
Ooh, I'm definitely watching this! 😉 Let's hope @cardre will answer soon...

cardre, 2017-04-19:
OK, I've now created a pull request (#118) for you to merge with the Unified-SPI branch. It's certainly not complete, but is working for me with the ATtiny1634. I've made it clear where other #defines will need to be set for other ATtiny chips to be supported. So consider it a first cut. If you'd like me to test with other chips and SPI hardware, you'll need to send them to me. That said, it shouldn't be hard for others to take this now and extend it with other #defines. Also note that SPI bus clock speeds are ignored in this first cut, but comments have been inserted where I believe it all needs to be added.
Lastly, would be good to get this merged into the master branch to take advantage of its ongoing updates.

SpenceKonde, 2017-04-19:
Thanks! I'll take a look at what needs to be added for the other chips, add in what I think is correct, do a few sanity checks, and then merge it back into master.

cardre, 2017-04-20:
Let me know when its merged back into master and I can reverify it again with my environment.

SpenceKonde, 2017-04-28:
Okay - to make this work with other chips, I'd really like to move the chip-specific defines (for the pins) out into the variants file; an attempt was already made at that, but it's not quiiite what is needed. That way the library will stay independent of the chip - as long as the defines are there, we'll be good. I hope to do that this weekend, and will update when I've done so.

bperrybap, 2017-05-13:
I think your approach sounds good. Sometimes depending on the conditionals or defines it can get tricky since Arduino selects things by board which then points off to variant. i.e. things get set by board type not specifically by chip type or core.

I just spent the past two days pulling my hair out with the TinyWireM library walking through it and finding and fixing a few bugs.
Overall, I don't think doing proper wrappers is that difficult, at least not for something the TinyWireM code. 
The key to really making it all "just work" is to not only get the wrapper in place, but also to fix some of the bugs, and incorrect status returns in the code. (At least that is what is needed for TinyWireM)
If you are about to go down this path, I could help as I'm quite fresh on the code have just been through it.
After reading some of the code I just wanted to throw rocks at the authors for some of the stuff being done.
Like have great big large comments describing functions and then returning something other than what the comments/documentation says. And then being inconsistent with their return values like sometimes 0 means pass and sometimes 0 means fail. And some functions return a pass/fail or an actual status code or error code and the return code points overlap.
That kind of stuff just makes my blood boil. It is the sign of an immature hacker vs a s/w developer.

bperrybap, 2017-05-13:
I just did a quick test. USIWire compiles and works on attiny85 using my hd44780 library.
You do have to change <Wire.h> to <USIWire.h> but other than that, it "just worked".
It is a little bit bigger (400 bytes) than TinyWireM but it looks promising since it sounds more complete.
It was also just a little bit faster than TinyWireM. ~10%
Would be trivial to create a Wire.h header file that simply include USIWire.h  to include it in the core as Wire.

Definitely worth a look.

bperrybap, 2017-05-13:
Just a follow up on what I used for the test of USIWire and a issue I ran into.
I used my hd44780 library with USIWire with the Mellis' attiny core. (that worked)
When I try the same thing with ATTinyCore it gets a linker error.
This does not appear to be anything to do with USIWire but it there is an issue with the ATTinyCore board package core that won't allow it to be used with my hd44780 library because of some of the hd44780 virtual functions.
I created another issue for this with full details: https://github.com/SpenceKonde/ATTinyCore/issues/129

SpenceKonde, 2017-05-14:
I committed some changes to get Unified SPI working. 

It builds, at least. If anyone has anything they can test it on (I would have to wire something up, and tomorrow I will likely be busy all day) 

Also - there's a new branch UnifiedWire which has a first attempt at unifying Wire and USIWire! No support for things that need to use software master, hardware slave yet. That'll be harder. 

Same deal with that - it builds on everything it's expected to, but nothing has actually been tested.

bperrybap, 2017-05-14:
For a i2c software master that is Wire API compatible, the library SoftwareWire is a nice library.
There is currently an issue with the ATTinyCore board package that keeps if from running.
It accidentally works "as is" on the other Attiny cores because they are using the arduino:arduino core.
It is easily fixed and I just bugged for them to fix: https://github.com/Testato/SoftwareWire/issues/8

Once SoftwareWire is fixed for this core, it can work with the ATTinyCore board package with existing code that expects a Wire library, after you tweak the existing code to include the correct header and name your i2c object Wire.

I tested it on the Attiny85 with the ATTinyCore with my hd44780 library.
(I've also tested SoftwareWire and used it on other cores)
SoftwareWire is available in the IDE library manager.

I haven't looked at the i2c signals with an analyzer yet but USIWire doesn't to get faster with my lcd testing when  you bump the i2c clock rate to 400khz, whereas SoftwareMaster does. 
At 8mhz talking to an LCD is twice as fast using USIwire vs SoftwareMaster @ 100khz
At 8mhz talking to an LCD is  but when running at 16Mhz and @ 400khz clock SoftwareMaster is only 5% slower.

bperrybap, 2017-05-14:
I'll go off and try the SPI stuff with my hd44780 library.

SpenceKonde, 2017-05-15:
Made a few improvements to the Unified Wire branch. I would love it if others could give it a spin - it would be great to get that merged in for 1.1.3, which I'd really like to release ASAP; It's been way too long since I did a release.

bperrybap, 2017-05-15:
I'm about ready to do some testing.
I have a question as to how you set up your environment for testing your board package.
With libraries for doing development and testing, I simply clone the git repo directly into my {sketchbook}/libraries directory.
This allows doing pulls/pushes directly from the repository and even using branches.

Unfortunately, it looks a bit different for board packages as the IDE uses a screwy naming convention/directory structure that does not allow a simple clone of the board package repo (like your ATTinyCore repo) directly into the {arduino15}/package directory.
I might be able to fudge up something, but I'm curious what strategy you are using to get your cloned repo into your packages area.

SpenceKonde, 2017-05-15:
I clone into {sketchbook}/hardware

The only time I ever install anything with board or library manager it to make sure it works before a release :-P I just clone things into {sketchbook}/libraries and {sketchbook}/hardware

bperrybap, 2017-05-15:
Cloning into {sketchbook}/hardware worked. I don't know why I didn't think of that alternative. Thanks.
I did a quick test of the SPI library with my hd44780 library on an Attiny85 and it didn't work.
It probably works ok for other simpler environments but my library is trying to work on many different platforms and tries to autodetect what resources are available like h/w spi.
The LCD i'm using is a write only device so it only needs MOSI and doesn't need any input from MISO.
The hd44780 library and test code did compile and run but it didn't end up using the attiny SPI library code because it reverted back to using shiftout() bit banging.
My library code looks for the symbol SPI_HAS_TRANSACTION to auto detect if the user included SPI.h to use h/w spi.
This doesn't get defined in SPI.h, so my library code assumes that SPI.h was not included and/or that h/w spi is not available and reverts back to using shiftout() rather than h/w spi.
I noticed that if the SPI.h code doesn't see SPDR (which apparently doesn't exist for the attiny85) so  it doesn't turn on SPI_HAS_TRANSACTION and the TinySPI class being used does not have the beginTransaction() or endTransaction() methods.

I haven't looked at the code but surely there is a way to either add  beginTransaction() and endTransaction() methods on TinySPI for fake them out if they can't be made to support all the functionality so that code that expects them to exist and uses them can work when TinySPI is used.
Worst case, maybe even something as ugly as dummy functions that ignore all the parameters.

In my case I need to use beginTransaction() and SPIsettings() to limit the clock rate so that it doesn't run too fast on other platforms like the chipkit or the ESP8266 which could clock too fast for the LCD.

SpenceKonde, 2017-05-15:
It already has beginTransaction() and endTransaction() and SPIsettings() - I just didn't have the SPI_HAS_TRANSACTION define. I just committed a fix for that. 

I don't have usingInterrupt() and notUsingInterupt() though, and I need to study what exactly their purpose is before I could take a stab at implementing them

bperrybap, 2017-05-15:
I already tried doing that. There is something else needed. Now the code won't link due to the missing methods.
```
/home/bill/Arduino/libraries/hd44780/examples/hd44780examples/LCDiSpeed/LCDiSpeed.ino:367: undefined reference to `tinySPI::beginTransaction(SPISettings)'
/home/bill/Arduino/libraries/hd44780/examples/hd44780examples/LCDiSpeed/LCDiSpeed.ino:367: undefined reference to `tinySPI::endTransaction()'
```

bperrybap, 2017-05-15:
The methods are declared in SPI.h for the tinySPI class but the methods
are never defined in SPI.cpp.
There are inline versions in the SPIClass class but only declarations in tinySPI class.

bperrybap, 2017-05-15:
To test for compilation you can compile one of my hd44780 library examples
[File]->Examples->hd44780->ioClass->hd44780_NTCU165ECB->HelloWorld

puuu, 2017-05-15:
@bperrybap thanks for giving USIWire a try.

My main aim of USIWire was the slave mode. So I did not any fine tuning of the master mode and `setClock()` is not yet implemented. You may can play with `T2_TWI` and `T4_TWI` in `src/USI_TWI_Master/USI_TWI_Master.h` to increase the timing.

SpenceKonde, 2017-05-15:
I took a stab at getting the SPI transaction support in there. I won't be able to even attempt to compile it until tonight. 

I'm not sure I understand what the interrupt stuff is achieving. 

Clock speed configuration is not yet implemented in my version either.

bperrybap, 2017-05-15:
You might want to consider moving this kind of experimental development onto its own branch vs using master. Surely those with the ability to help test it know how to use git branches.

SpenceKonde, 2017-05-15:
That's a fair point; will do that in the future.

SpenceKonde, 2017-05-17:
Huh, I thought I updated this on monday. I tested that it does indeed compile your example with SPI now. I can't check if it actually functions though, since I don't have that hardware.

bperrybap, 2017-05-17:
Apparently you did update it on Monday - more than once. I tried it after the first update but before the other updates.
Now the hd44780 code compiles but does not work. I'll have to dig deeper to see what is going on.
I'll look through the SPI code and if I don't see anything obvious, I'll hook up  a logic analyzer and see what is happening.

bperrybap, 2017-05-18:
OK, so I know what is going on. Just not sure how to fix it as I'm not fully understanding the USI registers yet.
The issue is the Attin85 SPI library is sending its SPI output data on the MISO pin instead of MOSI.
On the other AVRs, MOSI is always MOSI and MISO is always MISO regardless of whether the AVR is a master or slave. i.e. if I program the m328 AVR the MOSI pin is driven by the ISP programmer. 
If the m328 is driving the ISP signals, it drives the data to the slave on the same MOSI pin.

If I hook the MISO pin to the lcd instead of MOSI the lcd works as expected.

I'm assuming this is due to an "oops" somewhere in the SPI library code as it needs set up the direction of the pins based on knowing if it is in slave or master mode.

SpenceKonde, 2017-05-18:
That's correct and expected. The MISO/MOSI markings only apply for programming the chip via ISP. 

A USI does not have MISO or MOSI. It has DI and DO. When data is going out, it goes on the DO pin - regardless of whether it's master or slave. We updated the pinout diagrams to account for this - on boards that have a USI, the MOSI and MISO pin labels are highlighted as "programming pins" while SCK/DI/DO are highlighted as "serial interface" pins (like we do for spi/I2C/uart pins). I need to update the documentation to remind people of this, though.

bperrybap, 2017-05-18:
It kind of sounded like it was possible to set the direction of the two DI and DO pins (the datasheet mentions being able to set their direction)
This would allow them to be flipped around so they technically were not locked to "in" or "out" and could work like MOSI and MISO do on the other cores.
At least it sounded that way from my initial read of the Attiny85 datasheet.

If they truly are locked, then this will create some issues.
In particular there are issues with some pin constants in the pins_arduino.h header file.
It has constants for MISO and MOSI pins and if DO and DI pins are locked in a fixed direction, then this is not compatible with other cores.
Which means that sketches can no longer depend on MISO and MOSI constants working correctly.
As they are now, they are wrong for master mode when using the SPI library as the data is clocked out on the pins_ardino.h MISO pin.

Here is an example of the issue it creates for my lcd library.
My example creates an lcd object for SPI mode by specifying the SS, SCK, and MOSI pins.
This allows a single example to be built that should work on all cores.
It uses the constants from the pins_arduino header file to specify the SPI pins.
The library detects if the SPI.h header was included by the sketch and if so checks to see if the pins specified match SS, SCK, and MOSI. If so it uses the hardware SPI library. if not, it bit bangs out the MOSI pin.
(It auto detects if h/w SPI is possible, if so it uses it otherwise drops down to s/w bit banging)
The problem is that, currently, the pin that the hardware SPI uses is the one that pins_arduino defines as MISO.
This means that h/w SPI and s/w bit banging SPI end up using different pins.
No other core has this issue.

SpenceKonde, 2017-05-18:
I just reread the USI chapter of the datasheet several times. The only mentions of direction refer to the data direction register; there is no way to swap the DI and DO pins. DO is marked "output only", and I can't find any indication that the DI/DO pins can be swapped :-/

I adjusted the defines to match master mode, since using SPI slave mode is rare indeed in the Arduino - this was my intent all along, but I forgot about it when I added the MISO/MOSI defines. There also isn't a hardware defined SS pin

As an aside, why check if SS pin matches? Toggling of SS is done outside of the SPI library, so it shouldn't matter which pin they use for chip select...

bperrybap, 2017-05-18:
Actually, I went back and looked and I don't check for a match on SS. Only MOSI and SCK.
I need to go read the datasheet again.
I still don't understand how the data direction register affects the DI and DO pins.

MOSI and MISO on these parts is likely to be confusing and so it is probably worth a comment in the pins_arduino header file that the MISO and MOSI pins don't work like the other AVRs the constants are for master mode and will not agree with the MISO and MOSI pins used for slave mode and ISP programming or the labels on a Atmel ISP header.

i.e. if you have an Atmel ISP header on your ATTiny device you cannot use the MOSI and MISO labels when using the SPI library in master mode - they will be reversed.

closed 2017-06-05 by SpenceKonde