SpenceKonde/ATTinyCore #55: Optiboot for ATtiny167

SpenceKonde, 2016-04-11:
We need Optiboot for ATtiny167 - it's got hardware serial, this should not be hard.

SpenceKonde, 2016-04-12:
This may or may not be put into 1.1.0 - if I can, I'd like to, so I can start selling bootloaded boards on Tindie. The real question is whether it's just a matter of the LIN register names, or deeper changes are needed to the UART code to deal with the slightly different serial system here.

sleemanj, 2016-04-12:
Attached zip file contains **blindly configured and totally untested** optiboot hex's for ATTiny167 using SOFT uart (on the same pins as the hardware LIN/UART though, so... it should work).  

A brief look at the datasheet I don't think it's just a simple translation of register names to get it working with hardware uart, I don't see any particular disadvantage in using software uart particularly as there is no reserved boot area on the 167 so the flash consumption can be minimized anyway (I have reserved 580 bytes, including the 2 byte version header (err, footer).

[soft-uart.zip](https://github.com/SpenceKonde/ATTinyCore/files/216121/soft-uart.zip)

I don't have a 167 to test these with, they compiled that's all I can say :-)  Note there are different hex files for each baud and frequency combination.

This config was added to my optiboot fork...
https://github.com/sleemanj/optiboot/commit/82f248f303e51c242e4e674e0f0fc94fbe4893f7

sleemanj, 2016-04-12:
I should mention also that those bootloaders are (save bugs) **not** set to flash an led on boot, unlike standard optiboot does.

SpenceKonde, 2016-04-13:
Thanks - I'm hoping to get it to work with a real hardware serial - I think I'm pretty close working from the codebase in this repo (which also has properly working virtual boot, as used in 1634 and 841). The LIN/UART is not that different with all the weird shit turned off - in fact, the hardware serial code for the two is very similar), but I'm definitely doing something wrong with LIN that's keeping the communication from working right.

closed 2016-04-16 by SpenceKonde

SpenceKonde, 2016-04-16:
There! Full speed optiboot on the hardware UART. There is definitely much room for improvement, and 12mhz is busted (anything other than 57600@8 or 115200@16 doesn't work)

That was considerably more work than I expected - the LTXOK and UDRE flags are not quite interchangable, and it took me an incredibly long time to realize this.

commit c98a2017 (2016-04-16): Implement bootloader for x7 series (fix #55)

Booyah. It's a bit ugly, but it works fine now, and can be cleaned up
later.

Uses the hardware LIN/UART, cause software serial is for chumps.