SpenceKonde/ATTinyCore #58: Examples

SpenceKonde, 2016-04-26:
Isn't it possible to include examples with a core? 

If so, need:
* Blink that blinks the same LED as optiboot on all boards, and autodetects the board. 
* Examples of using I2C, SPI, particularly on systems with USI. See #53 
* ???

per1234, 2016-04-26:
This can be accomplished with a dummy library, something like:
```
|_hardware
  |_ ATTinyCore
    |_avr
      |_libraries
        |_ATTinyCoreExamples
          |_ATTinyCoreExamples.h
          |_examples
            |_Blink
              |_Blink.ino
```
The examples will appear under File > Examples when an ATTinyCore board is selected. The file ATTinyCoreExamples.h is required for it to be considered a valid library but can be an empty file. In other projects that I have done this with I added a comment to explain the purpose of the file. Optionally you could add a library.properties file to provide more information of the purpose of the "library" in its Library Manager entry.

SpenceKonde, 2016-04-26:
Aaaha, thanks - that makes sense. 

The USI ones can go in the improved SPI/I2C libraries, but yeah, that still leaves the blink demo in an examples library.

SpenceKonde, 2020-04-08:
No longer relevant. Unified Wire and SPI libraries eliminate need for USI-related examples. Stock Blink now blinks the correct led. There's a few examples in there of ATTinyCore specific tricks too now anyway. Should have been closed ages ago.

closed 2020-04-08 by SpenceKonde