SpenceKonde/ATTinyCore #62: board.txt for import arduino project to atmel studio 7

SuperJerry69, 2016-05-10:
It's bad, but for import tiny-arduino sketch in Atmel Studio 7 ( debugWire) my board.txt ( tiny85 1-8mhz intenal only ):

menu.cpu=Processor

attinyx5.name=ATtiny x5 series
attinyx5.upload.maximum_size=8192
attinyx5.upload.maximum_data_size=512
attinyx5.upload.tool=avrdude
attinyx5.bootloader.tool=avrdude
attinyx5.bootloader.unlock_bits=0xFF
attinyx5.bootloader.lock_bits=0xFF
attinyx5.build.core=tiny
attinyx5.build.board=AVR_ATTINYX5
attinyx5.bootloader.extended_fuses=0xFF
attinyx5.bootloader.file=empty/empty_all.hex
# BOD disable :
attinyx5.bootloader.high_fuses=0xD7

attinyx5.upload.maximum_size=8192
attinyx5.upload.maximum_data_size=512
attinyx5.build.variant=tinyX5

attinyx5.menu.cpu.85_8=ATtiny85 (8mhz internal)
attinyx5.menu.cpu.85_8.build.f_cpu=8000000L
attinyx5.menu.cpu.85_8.bootloader.low_fuses=0xE2
attinyx5.menu.cpu.85_8.build.mcu=attiny85

attinyx5.menu.cpu.85_1=ATtiny85 (1mhz internal)
attinyx5.menu.cpu.85_1.bootloader.low_fuses=0x62
attinyx5.menu.cpu.85_1.build.f_cpu=1000000L
attinyx5.menu.cpu.85_1.build.mcu=attiny85

SpenceKonde, 2016-05-11:
Can you explain what the issue you have is?

Is there a problem with the boards.txt in this core? If so, what?

SuperJerry69, 2016-05-11:
Sorry, i'm not clear ... :  When i want to import my tiny-Arduino sketch in Atmel Studio , it crash because of the board.txt of your (wonderful) core.
It's seam that Arduino import of AS support only 2 levels of menu ( board/mcu) in board.txt.
It's better ?

SpenceKonde, 2016-05-11:
It's unfortunate that Atmel Studio does not support multiple submenus. Can you provide any more detailed documentation on what the Atmel Studio import option requires of boards.txt, and how I might do that without trashing the user experience for people using the Arduino IDE? 

The clocking option menu is particularly critical - there are 9 different series' of ATTiny supported by this core - if each one has an average of 6 supported clock options (some have only 2, but others have like 10+), that means we'd have to convert that to 50 different boards.txt entries. The board menu would end up being three or more screens long, and people would hate it. 

Absent a way to make Atmel Studio accept additional submenus, I do not see a way to resolve this.

SuperJerry69, 2016-05-11:
Maybe if someone want import the sketch ( for me it's very useful ) it's easy to change the boards.txt for AS and it works nice !
I'm agree with you , It's better to keep all sub-menu in Arduino Ide , to let the Tinycore "universal"

SpenceKonde, 2016-05-11:
So the solution to this is to provide a "second" boards.txt for importing into Atmel Studio, which hardcodes the BOD setting, and wraps speed and MCU into one menu, so each boards.txt entry has craploads of options on the mcu menu?

SuperJerry69, 2016-05-11:
But if AS crash after 5 items in the menu ... i'll test it. 
In fact AS crash with the  pinout sub menu ... strange ...

xoriath, 2016-06-09:
Hi, I'm one of the Atmel Studio devs. Just wanted to let you know that I have registered this bug in our internal bug tracker as well. Internal bug reference AVRSV-7501.

I can't unfortunately promise any timeline on our side.

SpenceKonde, 2016-06-10:
@xoriath Thanks a lot for letting us know that! Looking forward to a fix in a future version of Atmel Studio.

SpenceKonde, 2016-08-12:
Closed because is an Atmel Studio issue.

closed 2016-08-12 by SpenceKonde