SpenceKonde/ATTinyCore #63: Missing tools for avrisp

KingJimmy27, 2016-05-11:
When I use Arduino 1.6.9 and select Tools - Programmer- Avrisp MKII (ATiny) I get a message:
Error while uploading: missing 'program.tool' configuration parameter

Any help please.

KingJimmy27, 2016-05-11:
> Closed #63 <https://github.com/SpenceKonde/ATTinyCore/issues/63> via
> 6837779

OK Spence,  thanks for the response but I am still new to this so am not
sure what to do.
I re-downloaded all the files and unzipped them to:
C:\Users\Jim\Documents\Arduino\hardware\ATTinyCore-master\avr
but nothing changed.
Interestingly if I use the Boards Manager to install in Arduino I cant find
those files in C:\Users\Jim\Documents\Arduino\hardware\

SpenceKonde, 2016-05-11:
See this for instructions for manual installation:
https://github.com/SpenceKonde/ATTinyCore/blob/master/Installation.md

It is likely that you must uninstall any board manager releases prior to
manual installation.

The board manager releases currently do not have this fix in, but I just
did a release this morning, so once Per does his thing and creates the new
json file for me within the next 24 hours, the board manager release with
this fix will be available from board manager.

KingJimmy27, 2016-05-11:
OK what I did was un-installed the Boards Manager versions then from
https://github.com/SpenceKonde/ATTinyCore I hit  download Zip and un-zipped
the files as per your manual installation instructions.
Then selecting avrisp MKII(ATtiny) I get:

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is
"C:\Users\Jim\Documents\Arduino\hardware\ATTinyCore-master\avr/avrdude.conf"

         Using Port                    : com1
         Using Programmer              : stk500v2
         Setting bit clk period        : 5.0
avrdude: ser_open(): can't open device "\\.\com1": The system cannot find
the file specified.

avrdude done.  Thank you.

Problem uploading to board.  See
http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

SpenceKonde, 2016-05-11:
Can you try it again now? I think I fixed it - I don't have an AVRISPmkII to test with so I don't know if that did the trick.

SpenceKonde, 2016-05-11:
Re-opening issue to help track this. 

I'm putting the release of 1.1.1 on hold, assuming you can verify this today (otherwise I'll have to go with what I had, I don't want to hold up the release any longer).

KingJimmy27, 2016-05-11:
Sorry for the delay in responding, its late evening here but that seems to
have done the trick.
I downloaded and un-zipped the latest and selected avrisp MKII(ATtiny)  and
then it uploaded OK.
Thanks for your effort, I hope this is of use to others as well.

closed 2016-05-12 by SpenceKonde

KingJimmy27, 2016-05-13:
As a matter of interest, in Arduino 1.6.9 I can use Programmer- Avrisp MKII (ATiny) or just the Avrisp MKII. Not sure what the object of your Avrisp MKII (ATiny) is.

SpenceKonde, 2016-05-13:
That only works for a some chips (ones with an entry in the default avrdude.conf file).

The point of the special one is because the Arduino IDE will ignore the path to avrdude.conf given in platform.txt except for programmers defined in a custom board package's programmers.txt - otherwise, it will use the default one which does not support all of the attiny chips supported by this core.

KingJimmy27, 2016-05-14:
Hi Spence, something strange.
In Arduino 1.6.9 everything appears to be OK but I usually use Visual Micro
http://www.visualmicro.com for all my Arduino stuff.
Now in VM when I try and build anything, even non ATTiny I get errors.
I started a new blank Arduino project in VM and selected Arduinio/Genuino
Uno. I also made sure no ATTinys were installed from Board manager and I
made the \Documents\Arduino\hardware folder ZZZZhardware to stop its use
but when I build I get:
(this is a only part of the Verbose )
Include Path 'C:\Program
Files\Arduino\hardware\arduino\avr\variants\standard'
"C:\Users\Jim\AppData\Local\arduino15\packages\ATTinyCore\tools\avr-gcc\4.8.1-arduino5\bin\avr-g++"
-c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections
-fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328p
-DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR
-I"C:\Program Files\Arduino\hardware\arduino\avr\cores\arduino"
-I"C:\Program Files\Arduino\hardware\arduino\avr\variants\standard"
"C:\Users\Jim\AppData\Local\Temp\VMicroBuilds\jimstest1\uno\jimstest1.cpp"
-o "nul"
The system cannot find the file specified

Obviously the reference to packages\ATTinyCore shouldnt be there.

Any suggestions.

SpenceKonde, 2016-05-14:
It sure looks like you have installed a tiny core through board manager? And also, it looks like it's trying to use the default core *and* the attiny core. 

I think this is a visualmicro bug - the Arduino team keeps fucking with the spec for third party hardware packages, so it's entirely possible that they updated it to work with an early 1.6.x version - but the spec has changed in mutually incompatible ways a few times since.. so it wouldn't be surprising if visualmicro was incompatible with it.

I'm not familiar with visualmicro - but if you want to make a new issue to track progress trying to figure out how to make it work with visualmicro, but I can't really put time into getting these sorted out.

KingJimmy27, 2016-05-14:
OK Spence, I will let you know if I resolve this.

KingJimmy27, 2016-05-18:
see this thread:
http://www.visualmicro.com/forums/YaBB.pl?num=1463298707/0#0

SpenceKonde, 2016-05-18:
Hmm - that suggests that it's a configuration issue, and that the problem is the interaction of "uninstalled" board manager packages with visualmicro? Anyway - let's track this in #65

commit 68377793 (2016-05-11): Fix #63

Thanks for catching this. I don't have those exotic programmers, so I hadn't tested them.
commit 616b5682 (2016-05-11): actually fix #63