SpenceKonde/ATTinyCore #64: Where in the repo should part-specific documentation go?

SpenceKonde, 2016-05-11:
For now am putting in ChipDocs in the root - but is there a better place to put that sort of thing?

@per1234, I'm looking at you ;-)

per1234, 2016-05-11:
The problem with putting it in the root is that it is seen as a different architecture by the IDE so after doing a manual installation when you open/close Boards Manager you get the warning:
```
Could not find boards.txt in E:\arduino\hardware\ATTinyCore-master\ChipInfo. Is it pre-1.5?
```
This doesn't actually break anything but it might cause some confusion for the users so I think it would be better to put the ChipDocs folder under the avr folder. The libraries specification recommends putting things like this in an extras folder(so it would be avr/extras/ChipDocs), that doesn't apply to cores but might be something to consider if there are eventually going to be multiple folders like ChipDocs just to keep them in one place.

SpenceKonde, 2016-05-11:
Moved them to extras so it doesn't cause problems for manual installs.

/avr/extras/docs or having the files loose in /avr/extras/ could both work.

SpenceKonde, 2016-08-12:
Okay, extras it is, should have closed this months ago..

closed 2016-08-12 by SpenceKonde