SpenceKonde/ATTinyCore #66: Consider use of -fshort-enums

SukkoPera, 2016-05-18:
On platforms with lesser RAM/flash, the core could make use of [GCC's -fshort-enums flag](https://gcc.gnu.org/onlinedocs/gcc-4.6.2/gcc/Code-Gen-Options.html):

Allocate to an enum type only as many bytes as it needs for the declared range of possible values. Specifically, the enum type will be equivalent to the smallest integer type which has enough room.

This also saves flash, since less code is needed to access a single byte in the AVR architecture (I *think*).

There is actually a caveat, as *the -fshort-enums switch causes GCC to generate code that is not binary compatible with code generated without that switch*, but this shouldn't be much of an issue on the Arduino platform. Perhaps it could even be enabled for all platforms, but I'm not that much of a tech guy :).

SpenceKonde, 2016-05-18:
Hm - I would love to hear commentary from someone who knows the compiler better; I don't know whether there would be any issue with this. For example, why would this not be used all the time? What is the disadvantage? 

The binary compatibility shouldn't be an issue, should it?

SukkoPera, 2016-05-20:
Yes, I agree that advice from compiler experts is essential.

Anyway, this can currently be achieved in the source on a per-enum basis using `__attribute__((packed))`.

SpenceKonde, 2016-08-12:
Which commands does this flag get put on? I'll try it in the optimization (lto) branch - the more I think about it, the less harmful that flag sounds.

Are there any other optimization flags we should be using but aren't?

SpenceKonde, 2017-05-15:
My study of this has led me to the conclusion that this would yield negligible benefit under typical scenarios - and it can be done on a per enum basis in code. I think it's best to leave it as in in the absence of a compelling case for the change and confirmation that it isn't likely to break anything.

closed 2017-05-15 by SpenceKonde