SpenceKonde/ATTinyCore #71: Incompatible with sudar/Arduino-Makefile ?

celem, 2016-07-27:
I realize that your ATTinyCore is designed for use with the Arduino IDE but many folks, including myself, like to use Makefiles instead of the Arduino IDE. Please consider future versions of ATTinyCore that could be compatible with both the Arduino IDE and the sudar/Arduino-Makefile. Consequentially, this issue is more of a suggestion than a bug, per se.

See: [https://github.com/sudar/Arduino-Makefile](url)

I tried to utilize ATTinyCore with the very popular sudar/Arduino-Makefile but gave up after encountering more than one incompatibility.

Sudar's boards.txt expects the same format used by the Arduino IDE but ATTinyCore uses "...menu.chip.85..." instead of the Arduino IDE's "...menu.cpu.85...". I edited the boards text to fix that problem but then I encountered other problems.

attinyx5.build.variant=tinyx5 needed to be attinyx5.build.variant=tiny in order to match the directory structure.

The f_cpu is in an unexpected format. The Arduino IDE and the sudar/Arduino-Makefile use these two formats:
attinyx5.menu.cpu.85.build.f_cpu=8000000L (compatible with ATTinyCore menu structure)
or
attinyx5-8.build.f_cpu=8000000L (not compatible with ATTinyCore menu structure)
However, ATTinyCore's boards.txt uses this structure:
attinyx5.menu.clock.8internal.build.f_cpu=8000000L (not compatible with sudar/Arduino-Makefile)

At this point, I gave up - simply much too incompatible.

SpenceKonde, 2016-07-27:
I would opine that this is an issue with their makefile not supporting a valid boards.txt file due to an incomplete implementation of support for the user defined menus that are available in 1.6.x versions - you should raise the issue with them.

My brief impression looking at your issue is that the makefile does not support a single board definition having a submenu to select the clock speed - is that accurate? If so, I cannot resolve this issue without gutting the core - notice how my core supports multiple submenus for all chips, allowing configuration of not only the clock speed (we could never have as many clock options if we didn't have a submenu, as the tools->boards menu would be unusably long) but the BOD settings, and in the case of the Tiny85 and Tiny861, the clock source for Timer1 (ie, system clock, PLL, or PLL/2).

celem, 2016-07-27:
sudar/Arduino-Makefile does support sub-menus, although its use isn't well documented. For example, I was mostly able to parse your boards.txt sub-menus with the Makefile define **shown below**. It can also improve parsing by setting  "BOARD_SUB". I used BOARD_SUB = 85 with a boards.txt with the format: "attinyx5.menu.cpu.85.build.mcu=ATtiny85"

# https://github.com/SpenceKonde/ATTinyCore
ALTERNATE_CORE = ATTinyCore
ISP_PORT = /dev/ttyS0
AVRDUDE_ARD_PROGRAMMER = usbasp
# F_CPU defined here due to incompatible syntax in boards.txt
F_CPU = 8000000L
BOARDS_TXT = /home/ecomer/Arduino/hardware/ATTinyCore/avr/boards.txt
ARDUINO_VAR_PATH  = /home/ecomer/Arduino/hardware/ATTinyCore/avr/cores
ARDUINO_CORE_PATH = /home/ecomer/Arduino/hardware/ATTinyCore/avr/cores/tiny
BOARD_TAG         = attinyx5

I am currently using "http://code.google.com/p/arduino-tiny/downloads/detail?name=arduino-tiny-0150-0020.zip" because I have it working 100% with sudar/Arduino-Makefile

SpenceKonde, 2016-08-16:
I looked over this again - some of the things you mentioned in the first message (for example, changing variant to tiny - the variant must be tinyx5, thats how it knows to use this pins_arduino.h which contains all the pin mapping for the tiny85 https://github.com/SpenceKonde/ATTinyCore/tree/master/avr/variants/tinyX5 ) indicate that there are far deeper problems here. 

As far as I'm concerned, this is a bug in the Arduino Makefile tool - I suggest you take it up with them - I have my hands full keeping my core working with the official IDE. Of course, if you are able to come up with a fix for boards.txt that doesn't remove functionality, I'd be happy to merge it (after testing it, of course)!

A work-around would be to modify your boards.txt to remove all the submenus (ie, hard-code it to the settings you want) so it won't confuse the makefile. 

Anyway, closing as won't-fix. Reopen or create new issue if situation changes.

celem, 2016-08-17:
Thank you for looking at it again. For now, I have taken the path of least resistance by using the Google solution at:https://code.google.com/archive/p/arduino-tiny/downloads

surr, 2016-09-03:
To use this core with arduino-makefile it is sufficient to provide a custom `boards.txt` file.
Probably, this boards file could be generated by makefile on the fly based on selected menu options.

This is an example of config files for attiny2313 with custom boards file.

`Makefile`
```
ALTERNATE_CORE = tiny
ARCHITECTURE = avr
BOARD_TAG    = attiny2313at8
BOARDS_TXT = boards.txt
ARDUINO_SKETCHBOOK = .
include /usr/share/arduino/Arduino.mk
```

`boards.txt`
```
attiny2313at8.name=ATtiny2313 @ 8 MHz
attiny2313at8.build.mcu=attiny2313
attiny2313at8.build.f_cpu=8000000L
attiny2313at8.build.core=tiny
attiny2313at8.build.variant=tinyX313
```

celem, 2016-09-03:
surr - interesting idea. Yes, maybe I'll write a helper bash script that does what you suggest.
Thank you for the suggestion!

slavkoja, 2016-09-04:
Please, leave it open (perhaps with wontfix tag) to one can easy find mentioned workaround.

SpenceKonde, 2016-09-04:
Done, will monitor this. If you make such a script please let me know here, and we can talk about how best to let people know about it

slavkoja, 2016-09-04:
One can append to `Makefile` (not tested yet):
```Makefile
boardsinit: $(BOARDS_TXT)
	cat $(BOARDS_TXT)

$(BOARDS_TXT): Makefile
	$(file >$@,$(BOARD_TAG).name=ATtiny2313 @ 8 MHz)
	$(file >>$@,$(BOARD_TAG).build.mcu=attiny2313)
	$(file >>$@,$(BOARD_TAG).build.f_cpu=8000000L)
	$(file >>$@,$(BOARD_TAG).build.core=$(ALTERNATE_CORE))
	$(file >>$@,$(BOARD_TAG).build.variant=tinyX313)
```
(change spaces to TAB before any `$(file ...)` and `cat ...` lines)

Then use it (to create `boards.txt`) as this:
```
make boardsinit
```
On any change in `Makefile` the `borads.txt` will be created from fresh. To remove it, simple use `rm boards.txt`.

SpenceKonde, 2016-09-14:
How's this been coming? Did this work out for you guys?

If any of you have instructions on how to use it with the makefile, I'd love to hear it so I can add that to the docs.

celem, 2016-09-14:
I'm out of the country and cannot work on it until October.

JelmerT, 2017-01-22:
Any progress on this?
I'm interested in using this core with a proper makefile.

celem, 2017-01-23:
I could not get slavkoja's idea to work and I am out of time to work on it. I see:
ecomer@bergamont ~/Arduino/BlinkATTiny8520su $ make boardsinit
...
-------------------------
grep: boards.txt: No such file or directory
grep: boards.txt: No such file or directory
cat boards.txt
cat: boards.txt: No such file or directory
make: *** [boardsinit] Error 1
ecomer@bergamont ~/Arduino/BlinkATTiny8520su $

SpenceKonde, 2017-01-25:
Thanks for the update - I wish I were able to be more useful here. If there's anything I can do to help (in light of the fact that I'm not good at working with makefiles), let me know. It would be cool if we could add support for this, since it looks like there are a few people interested. 

Is the issue just that the submenus in boards.txt are confusing it or what?

celem, 2017-01-30:
I discovered one problem. Only Make version 4+ supports the "file" function used by slavkoja and my is uses make version 3.8. This will require invoking an external bash script. I'm looking into just such a script that will parse the tinycore boards.TXT to create a local boards.TXT for use by the Makefie.

celem, 2017-01-30:
**MAKEFILE WORKING WITH ATTINYCORE**
OK, I have the Makefile working. I switched to the Makefile at [https://travis-ci.org/ladislas/Bare-Arduino-Project](url)
Following the project directions I made the following changes to get it working for attiny85. Obviously, your file locations may differ:

PROJECT_DIR       = /home/ecomer/MyArduinoProject
ARDMK_DIR         = $(PROJECT_DIR)/Arduino-Makefile
ARDUINO_DIR       = /home/ecomer/arduino-1.6.11
BOARD_TAG         = attinyx5
BOARD_SUB         = 85
ALTERNATE_CORE    = ATTinyCore
F_CPU             = 8000000L

SpenceKonde, 2017-02-03:
Thanks - did you end up writing a script to generate a special boards.txt or what? Ie, how did you deal with the issue of all the submenus, which as I understood it was the main issue? 

If anyone wants to write up a step-by-step procedure (ideally as a .md in a pull request, so all I have to do is add some links to it in the readme), I'd be happy to send you a few freebies from my Tindie store ( https://www.tindie.com/stores/DrAzzy/ ) as thanks.

celem, 2017-02-03:
Re: "_did you end up writing a script to generate a special boards.txt_" - no. I got the current version Sudar's Makefile working unmodified. It just took a lot of reverse engineering time to figure out the defines (in my previous post) especially the BOARD_SUB. Sudar quietly changed the latest version of the Makefile in a way that accommodates your ATTinyCore boards.txt. In summary, the sudar makefile package is _off the shelf_ ready to work with ATTinyCore **IF** the defines are correctly set in the user's Makefile.

celem, 2017-02-05:
Re   "...write up a step-by-step procedure (ideally as a .md in a pull request, so all I have to do is add some links to it in the readme)” -   My bandwidth is currently maxed out but when I can get a breather I'll see what I can do for documentation for using a a Make file for ATTinyCore. However, anything that I write is likely to  be Linux centric as I haven't used Windows for many years.

celem, 2017-02-12:
I see that your .MD files are html rather than true markdown syntax. Which is preferred?

SpenceKonde, 2017-02-12:
They are?

My .md files should be normal github markdown. Please create a new issue to bring my attention to any cases where they are not.

celem, 2017-02-12:
I simply downloaded a couple of them and then opened with a plain text editor - they are HTML. I looked at the installation and readme MD files at attinycore.

SpenceKonde, 2017-02-12:
You're looking what's sent to the browser when viewing the files - github processes the files on the server side to generate pretty formatted HTML, but the files themselves are straight .md - click the "raw" view, or edit the files (or clone the repo to local machine) to see the actual file contents.

celem, 2017-02-12:
Agreed - I just did a git clone of your github repository and in the cloned directory the md files are in markdown syntax.The little thinks will trip you up!

SpenceKonde, 2017-02-13:
Tell me about it. Took me hours the first time I encountered it...

commit 05bb2fcc (2017-02-16): Merge pull request #109 from celem/makefile : fix #71

Add makefile.md file
closed 2017-02-16 by SpenceKonde