SpenceKonde/ATTinyCore #72: ATTiny1634 AREF

Smrtokvitek, 2016-08-15:
Does anyone have tried external voltage reference on ATTiny1634? I am using LM4041 voltage reference (1.225 V) connected to AREF but I can read only 1023 (overflow?) and nothing else. When I use internal 1.1 V reference or 5 V (Vcc) reference it works. Input voltage I measure is lower than 1.225 V.

SpenceKonde, 2016-08-15:
Huh. Interesting. I've had some odd (but different) issues related to AREF with internal reference. Have not tried with external reference.

Smrtokvitek, 2016-08-15:
I will try to set the prescaler differently tomorrow. I am now using default way of reading from analog input via analogRead() and analogReference(EXTERNAL) as the ADC setup/init. I will let you know.

Smrtokvitek, 2016-08-16:
so it turns out that I am an idiot. The minimum voltage that can be connected to AREF is 2 V. Sorry.

SpenceKonde, 2016-08-16:
Well that would do it. 

That's a super lame limitation - where did you find that mentioned, by the way? I don't see it in the datasheet...

Smrtokvitek, 2016-08-16:
table 24-9 External Voltage Reference minimum 2.0 V

SpenceKonde, 2016-08-16:
Wow, what a great example of crystal clear documentation!

closed 2016-08-16 by SpenceKonde