SpenceKonde/ATTinyCore #73: (sketchname)_bootloader.hex exports don't work for 841, 1634, and 167.

SpenceKonde, 2016-08-18:
On these platforms, we have to use Virtual Boot in order to get a bootloader, since they don't have BOOTRST fuse. This means that the bootloader has to rewrite two interrupt vectors during the write process - these changes are obviously not made when we export .hex files, so the images run the sketch but not the bootloader. 

In order to generate a working (sketchname)_bootloader.hex I would need to be able to run some script after generating the export to rewrite the vectors. 

Alternately, a solution to simply suppress the generation of the image with the included bootloader - while not ideal - would be a partial solution.

MCUdude, 2016-08-30:
Is this the reason why the bootloader gets overwritten when I'm uploading a sketch to an ATtiny841 that runs Optiboot? After uploading the bootloader is gone.. IIRC the exported HEX file is the same as the one that gets uploaded.

SpenceKonde, 2016-08-30:
No. 

Upload using ISP programmer always erases the bootloader on Atmel 8-bit AVR chips. The only command available to erase any part of flash via ISP is chip erase (erases whole chip plus lock bits). 

The point of the hex file with bootloader is so that you can manually upload the hex file containing the bootloader along with the sketch, thus creating a chip with the bootloader and sketch present on. These hex files (which do not work for the chips listed above) are not actually used by the IDE.

MCUdude, 2016-08-30:
OK, got it. May I ask you; then how can Optiboot get replaced (Replace itself?) when uploading using a USB to serial adapter? I have no experience with the virtual boot section implementation

SpenceKonde, 2016-08-30:
That should not happen, are you saying that it does? 

Upload via serial should never break the bootloader. Is the autoreset working correctly? If the board isn't resetting when you open the serial connection, the bootloader will never run. Do you get the triple blinks when you press reset?

MCUdude, 2016-08-30:
I guess I just hadn't done my homework; the RST_EN header wasn't connected! I'll solder a header to the board, so I don't forget it. I'll usually prefer to have reset enabled, but sometimes it sucks that the microcontroller get reset every time i open the serial monitor. 

One more thing: you should probably replace the resistor for the LED to one with a larger value, It's destroying my night vision 😄 I got some green LEDs on the boards I sell, and they were so bright compared to the red ones I had to use 15k resistors for the green ones, and 390R for the red and yellow ones to make them the same brightness!

(Anyway where do you buy those LEDs? The are amazing!)

SpenceKonde, 2016-08-30:
Aaah, good call, I should add a note about that on product page. 

Heh, thanks for the feedback on the LEDs - was I using 1k or 680 for those? Should have been 1k, might have accidentally used 680's. But it sounds like maybe I can back it off more - that's always better, since it means less current and less interferance with whatever the user might want with that pin. 

Those are nothing-special LEDs - they just come from china via ebay and aliexpress. If they're the weird bluish-white, they're the "ice blue" or "water blue" ones.

MCUdude, 2016-08-30:
on my boards you're used 100 ohms (101)! No wonder it where so darn bright. You've used 820R on the ATmega328pb board and 470R on the Attiny1634 board.

SpenceKonde, 2016-08-30:
Shit! Those were supposed to be 1k! 

I'll go through my inventory and make sure there aren't anymore!

SpenceKonde, 2017-01-25:
Cleaned up this issue (there was an off topic discussion here before) - This issue is still here. I don't know how to fix it.

per1234, 2017-01-26:
>Alternately, a solution to simply suppress the generation of the image with the included bootloader - while not ideal - would be a partial solution.

I have managed to accomplish this, see https://github.com/per1234/ATTinyCore/commit/e812e1d868fe109d0620796c29ef6d38d632fb73
I only suppressed the export of the merged output of the ATtiny841/1634/167 with optiboot boards but it can easily be changed to suppress this for any board.

It's a bit tricky because there's no way to determine the sketch folder so I had to delete the merged output file from the build folder before the export using the `presavehex` hook.

The shell script is untested, I only tested the Windows batch file.

I don't know how to do the vector rewrite part but what I've done might be adapted to provide a full solution to this issue. If you can give me some direction in how to do the vector rewrite I'm happy to work on this more.

SpenceKonde, 2017-02-03:
Woah, how did I miss this? 

This is very exciting. I'll try your solution sunday (hopefully) and check it on my Linux box as well. At that time I'll also get some more information together on the vector rewrite bit, it's been a while since I implemented it. We basically have to copy what *was* the reset vector to the wdt vector (note - for the '841 this requires changing the value, since it's an rjmp not a jmp - for the 16k chips, you just copy it) change the reset vector to point to the bootloader.

per1234, 2017-02-03:
I'm glad if it can be useful. As for the current scripts, it struck me later that maybe it would be safer to rename the "with_bootloader" hex file (to something like {build.project_name}.with_booloader.hex.disabled) rather than deleting files but that's moot if we can do the vector rewrite thing.

SpenceKonde, 2017-05-14:
Everything except tiny828 with optiboot has been set to not export merged output. 

Will close after testing on Linux.

per1234, 2017-05-14:
I have a Raspberry Pi now that I can try it on if you need a Linux tester.

SpenceKonde, 2017-05-14:
It DOES NOT WORK on Linux out of the box. 

The script to delete the files ends up not having execute permission. I don't know enough about how linux and git interact to know how to make it come out with the correct permissions. Need help here - @per1234  - do you have any idea?

SpenceKonde, 2017-05-14:
Can the presavehex hook have multiple commands? Ie, could it just chmod +x the script before invoking it? No matter what method is used for install, the user should have the privs to do that, right?

SpenceKonde, 2017-05-15:
Works for me on Linux. 

Should probably be tested on more systems to make sure we're not putting in something that will error out for some people.

per1234, 2017-05-15:
Glad you were able to find a solution! I just tested it on Raspbian and it worked fine.

SpenceKonde, 2017-05-15:
Thanks! 

Anyone got a mac to check it on?

closed 2017-06-05 by SpenceKonde