SpenceKonde/ATTinyCore #75: Clean

MCUdude, 2016-08-20:
Hey!
I was reading through my
have a

MCUdude, 2016-08-20:
Sorry, I accidentally hit enter. I'll finish this in a minute 😉

MCUdude, 2016-08-20:
Have a look at [how I named my bootloader HEX files](https://github.com/MCUdude/MightyCore/blob/master/avr/boards.txt#L46). I think this is the simplest way to do this. If so, we'll have to edit the Optiboot make file too

MCUdude, 2016-08-20:
Here's the first attempt! I don't have any "bootloader compatible" chips yet (will arrive soon 😃), but the IDE sends the correct fuse values to avrdude (enable the self program fuse when the bootloader option is selected). AVRdude returns this error:

`Bootloader file specified but missing: /Users/Hans/Documents/Arduino/hardware/ATTinyCore/avr/bootloaders/optiboot/optiboot_attiny167_57600_8000000L.hex`
which indicates that this should work!

```
attinyx7.name=ATtiny87/167
attinyx7.upload.maximum_data_size=512
attinyx7.upload.tool=avrdude
attinyx7.build.board=AVR_ATTINYX7
attinyx7.bootloader.tool=avrdude
attinyx7.bootloader.unlock_bits=0xFF
attinyx7.bootloader.lock_bits=0xFF
attinyx7.build.core=tiny

attinyx7.menu.chip.167=ATtiny167
attinyx7.menu.chip.167.build.mcu=attiny167
attinyx7.menu.chip.167.upload.maximum_size=16384
attinyx7.bootloader.extended_fuses=0xFF
attinyx7.bootloader.file=empty/empty_all.hex

attinyx7.menu.chip.167optiboot=ATtiny167 (Optiboot)
attinyx7.menu.chip.167optiboot.build.mcu=attiny167
attinyx7.menu.chip.167optiboot.upload.maximum_size=15744
attinyx7.bootloader.extended_fuses=0xFE
attinyx7.bootloader.file=optiboot/optiboot_{build.mcu}_{upload.speed}_{build.f_cpu}.hex

attinyx7.menu.chip.87=ATtiny87
attinyx7.menu.chip.87.build.mcu=attiny87
attinyx7.menu.chip.87.upload.maximum_size=8196
attinyx7.bootloader.extended_fuses=0xFF
attinyx7.bootloader.file=empty/empty_all.hex

attinyx7.menu.chip.87optiboot=ATtiny87 (Optiboot)
attinyx7.menu.chip.87optiboot.build.mcu=attiny87
attinyx7.menu.chip.87optiboot.upload.maximum_size=15744
attinyx7.bootloader.extended_fuses=0xFE
attinyx7.bootloader.file=optiboot/optiboot_{build.mcu}_{upload.speed}_{build.f_cpu}.hex

attinyx7.menu.clock.8internal=8 MHz (internal)
attinyx7.menu.clock.8internal.bootloader.low_fuses=0xE2
attinyx7.menu.clock.8internal.upload.speed=57600
attinyx7.menu.clock.8internal.build.f_cpu=8000000L

attinyx7.menu.clock.20external=20 MHz (external)
attinyx7.menu.clock.20external.bootloader.low_fuses=0xFF
attinyx7.menu.clock.20external.upload.speed=115200
attinyx7.menu.clock.20external.build.f_cpu=20000000L

attinyx7.menu.clock.16external=16 MHz (external)
attinyx7.menu.clock.16external.bootloader.low_fuses=0xFF
attinyx7.menu.clock.16external.upload.speed=115200
attinyx7.menu.clock.16external.build.f_cpu=16000000L

attinyx7.menu.clock.12external=12 MHz (external)
attinyx7.menu.clock.12external.bootloader.low_fuses=0xFF
attinyx7.menu.clock.12external.upload.speed=115200
attinyx7.menu.clock.12external.build.f_cpu=12000000L

attinyx7.menu.clock.8external=8 MHz (external)
attinyx7.menu.clock.8external.bootloader.low_fuses=0xFF
attinyx7.menu.clock.8external.upload.speed=57600
attinyx7.menu.clock.8external.build.f_cpu=8000000L

attinyx7.menu.clock.6external=6 MHz (external)
attinyx7.menu.clock.6external.bootloader.low_fuses=0xFD
attinyx7.menu.clock.6external.upload.speed=57600
attinyx7.menu.clock.6external.build.f_cpu=6000000L

attinyx7.menu.clock.4external=4 MHz (external)
attinyx7.menu.clock.4external.bootloader.low_fuses=0xFD
attinyx7.menu.clock.4external.upload.speed=38400
attinyx7.menu.clock.4external.build.f_cpu=4000000L

attinyx7.menu.clock.1internal=1 MHz (internal)
attinyx7.menu.clock.1internal.bootloader.low_fuses=0x62
attinyx7.menu.clock.1internal.upload.speed=9600
attinyx7.menu.clock.1internal.build.f_cpu=1000000L

attinyx7.menu.bod.disable=B.O.D. Disabled
attinyx7.menu.bod.disable.bootloader.high_fuses=0xD7
attinyx7.menu.bod.1v8=B.O.D. Enabled (1.8v)
attinyx7.menu.bod.1v8.bootloader.high_fuses=0xD6
attinyx7.menu.bod.2v7=B.O.D. Enabled (2.7v)
attinyx7.menu.bod.2v7.bootloader.high_fuses=0xD5
attinyx7.menu.bod.4v3=B.O.D. Enabled (4.3v)
attinyx7.menu.bod.4v3.bootloader.high_fuses=0xD4
attinyx7.menu.pinmapping.new=New style (down each side)
attinyx7.menu.pinmapping.new.build.variant=tinyX7_New
attinyx7.menu.pinmapping.old=Old style (SCL on 13)
attinyx7.menu.pinmapping.old.build.variant=tinyX7
```

SpenceKonde, 2016-08-22:
This means we would need to build bootloaders for every supported clock speed, then? 

Also, some of the boards (the x41, 1634 and 828) need 2 entries for internal, but only for bootloader ones (because we need to use different bootloader files if they want to use the board at 5v vs 3.3v because of the voltage dependence of the internal oscillator - serial communication doesn't work at 5v if you build for 8mhz)

MCUdude, 2016-08-22:
Yes,  I suppose there would have to be a bootloader for every clock frequency, but I'll have to double check. Dealing with 5v/3.3v should be an issue. Just create a "voltage select" menu,  and use {some.argument} in the bootloader file name

SpenceKonde, 2016-08-22:
But you can't have a voltage menu for things that use a crystal, otherwise
I'll be forever dealing with people asking what the difference is....

MCUdude, 2016-08-23:
Aah, I didn't thought about that. Anyways, what do you think about this setup? Here they are ordered by name:

[image]

SpenceKonde, 2016-08-24:
Isn't that what I have now? 

I wonder if we could combine the 87 and 167 Optiboot options into one - I think we can. 

Also - I think there is some hope for combining bootloader and non-bootloader board defs...

MCUdude, 2016-08-24:
> Isn't that what I have now?

This is what we have now:
[image]

Combining the 87 and 167 Optiboot options should be possible. Here's the slightly modified boards.txt file if you think the new order is slightly more logical

[boards.txt](https://github.com/SpenceKonde/ATTinyCore/files/435694/boards.txt)

SpenceKonde, 2016-08-31:
Well I've got a pull request in a branch to test this.

However there's a merge conflict. I do not know how to resolve merge conflicts :-( 

https://github.com/SpenceKonde/ATTinyCore/pull/78

So assuming it works in the branch, I still won't be able to implement this change unless someone can tell me how to make the merge work. I hate github sometimes. It's just like "merge conflict!  go use these hideously complicated tools you've never used before and have miserable documentation to fix a problem which we have not identified to you"

MCUdude, 2016-08-31:
I'm completely lost when git won't cooperate.. @per1234 has been really helpful before. Maybe he knows how to fix this?

SpenceKonde, 2016-08-31:
In any event, it's academic until the changes are tested
but after that, we'll be dead in the water.

per1234, 2016-09-01:
I've actually never merged a pull request but I frequently deal with merge conflicts because I tend to get a bit perfectionist about commit history in my local development branches. I do a lot of rebasing to squash/fixup previous commits or reorder commits. It's always nerve-wracking watching git shuffle through a long list of commits while praying there won't be a conflict.

When there is a conflict git puts both copies of the conflicting lines in the file(s), indicated by things like:
```
=======
>>>>>>> refs/remotes/origin/New-boards.txt
```
So you have to manually edit the file. In the easiest case you only need to delete the version of the lines that you don't want but often there are parts of each you want to keep so you have to merge them together.

In this case the problem is that https://github.com/SpenceKonde/ATTinyCore/commit/6d793ef829bcbb4d3ddfefd7da2b17569ca07935 was made after the New-boards.txt branch was created. Since that commit edited the same lines of boards.txt that were edited in the New-boards.txt branch git can't automatically figure out how to merge. Overall the version of boards.txt in the New-boards.txt branch is what you want to keep but I'm guessing you don't want to lose the support for internal oscillator at 5V that was added in https://github.com/SpenceKonde/ATTinyCore/commit/6d793ef829bcbb4d3ddfefd7da2b17569ca07935.

So my advice is to rebase New-boards.txt on master and do a force push before you do the testing so that you will be testing the final version of boards.txt instead of making last minute changes while resolving the conflict during the merge.

I am happy to do this but I think I would have to submit a new pull request.

I agree about git being a bit overly complicated and the documentation not so friendly. At first I was like "no I'm not going to learn a new programming language just to do version control" but using GitHub, then GitHub desktop, then resorting to the command line when I had no other choice has eased me into it. I still prefer using a GUI for most things but GitHub desktop is pretty limited and I wasn't impressed by any of the other options I tried. I usually end up writing down some instructions for myself when I learn how to do something new in git rather than having to dig through the documentation/tutorials all over again in a few months after I've forgotten how to do it.

SpenceKonde, 2016-09-01:
Thanks @per1234 - but you were too late. I got help from a guy I know irl, and got it sorted out

commit 1dcff21d (2016-09-01): merged (fix #75)
closed 2016-09-01 by SpenceKonde