SpenceKonde/ATTinyCore #76: Find way to get rid of compiler warnings

SpenceKonde, 2016-08-24:
The 1634, 828, and x41 chips give warnings about int-to-pointer cast. I do not know how to make these warnings go away without changing the progmem byte arrays to ints, and thus wasting flash space.

MCUdude, 2016-08-24:
There are some other (and minor) warnings as well. When compiling _any_ code for _any_ microcontroller I get these warnings:

```
/Users/Hans/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/WInterrupts.c:38:1: warning: 'static' is not at beginning of declaration [-Wold-style-declaration]
 volatile static voidFuncPtr intFunc[NUMBER_EXTERNAL_INTERRUPTS];
 ^

/Users/Hans/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/TinySoftwareSPI.cpp:80:6: warning: unused parameter '_data' [-Wunused-parameter]
 byte SoftSPIClass::noTransfer(byte _data){
      ^

/Users/Hans/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/TinySoftwareSPI.cpp:211:6: warning: unused parameter 'rate' [-Wunused-parameter]
 void SoftSPIClass::setClockDivider(uint8_t rate)
      ^
/Users/Hans/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/WString.cpp: In member function 'int String::lastIndexOf(char, unsigned int) const':
/Users/Hans/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/WString.cpp:551:38: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
  if (fromIndex >= len || fromIndex < 0) return -1;
                                      ^
/Users/Hans/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/WString.cpp: In member function 'int String::lastIndexOf(const String&, unsigned int) const':
/Users/Hans/Documents/Arduino/hardware/ATTinyCore-master/avr/cores/tiny/WString.cpp:567:63: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
    if (s2.len == 0 || len == 0 || s2.len > len || fromIndex < 0) return -1;
                                                               ^
```

When it comes to all the warnings for the 1634, 441/841 and 828 I have no idea.. The Arduino team chose the easiest way: 
``` c++
const uint16_t PROGMEM port_to_mode_PGM[] = {
	NOT_A_PORT,
	NOT_A_PORT,
	(uint16_t) &DDRB,
	(uint16_t) &DDRC,
	(uint16_t) &DDRD,
};
```

SpenceKonde, 2016-08-24:
Huh! That's a new warning, then.... I did not see that warning before from static not being first before. That's a fucking dumbass thing to warn about, too!

Interesting about the String warnings, I'll need to look into what the right way to fix that is. I had not seen them - maybe they're only from non-modern tiny's? I don't build for those very often (though they used to build clean, I thought). 

The tinysoftspi things are debris that needs to be excised from the core and replaced with #53 - I've almost finished an SPI version of that, and I think a master only wire one will be straightforward. 

Yeah - but the way they did it wastes flash space. I refuse to make my core worse in a functional sense just to get rid of warnings.

bperrybap, 2016-10-01:
with a cast like this:
```
  (uint8_t) (uint16_t)(&DDRA),
```
or this:
```
  (uint8_t) (int)(&DDRA),
```

In the larger picture of flash space, when the pin count is small, a ternary is more efficient than having a table  lookup from flash on the AVR because the AVR doesn't support direct access to flash which makes flash lookups expensive.

i.e. if the digitalPinToXXX() and portXXXRegister() macros were in the pins_arduino.h header file down in the variant file then the variant file could decide whether to use a table in flash or a simple ternary for the operation depending on which was more efficient.

On small pin numbered devices the ternary would be less code and faster lookups which would increase the speed of things like digitalWrite(), particularly when the pin numbers are constants since the ternary will be completely optimized away at compile time and replaced with the final result.

sleemanj, 2016-10-01:
>  if the digitalPinToXXX() and portXXXRegister() macros were in the pins_arduino.h header file dow

This is what I did in my fork for 4/5/9/10/13/x4 and x5...
https://github.com/sleemanj/ATTinyCore/blob/master/avr/variants/tinyX5/pins_arduino.h#L216

> increase the speed of things like digitalWrite()

I also optimise pinMode, digitalWrite for compile-time-known situations...
https://github.com/sleemanj/ATTinyCore/blob/master/avr/cores/tiny/Arduino.h#L174

commit f8474c70 (2016-10-04): Fix #76

With casts. Thanks @bperrypap
closed 2016-10-04 by SpenceKonde