SpenceKonde/ATTinyCore #80: Attyny828 with external clock generator

battosai30, 2016-09-12:
Hi,

Would it be difficult to implement external clock generator for attiny828 ?

I have a specific application with RS-485 and I would like to don't have to use the internal clock ...

Regards

SpenceKonde, 2016-09-12:
Making the 828 use the external clock source is trivial. 
 Burn bootloader normally, then just change the low fuse to E0 or EC (depending on how the tool you're using to burn bootloader handles unused bits - some treat as 0's, some treat as 1's) - or modify boards.txt to set low fuse to EC instead of EE and do burn bootloader. If it's not running at 8mhz, you'll also need to set the CPU speed in boards.txt to the speed of the external clock. If using Optiboot (instead of non-bootloader version), the included version of Optiboot is assuming it's 8mhz - ofc, it will work at 16mhz with double the baud rate. 

The problem with putting this into the mainline core is that the external clock could be at any speed, so I'd need a whole bunch of extra entries in the menu - and I worry that that would cause pain for users who misunderstood the difference between clock and crystal, and accidentally bricked their part.

battosai30, 2016-09-13:
Thanks for your quick answer.

I understand your point of view, even if I think that Attinys in Arduino are not very "mainstream", so generally noobs don't play with them. Moreover, there are no pin to connect a crystal (It's how I discovered this "subtility") so you quicly see that there is a problem. One remark about that : the github mainpage does not clearly show this particularity : 

External crystal (all except x8 series):

    20 MHz
    16 MHz
    12 MHz
    8 MHz
    6 MHz
    4 MHz

As Attiny828 cannot use any crystal, It should be mentionned "(all except x8 series and Attiny828)".

Anyway I will do my tests based on your advices and I will write my return here ;)

SpenceKonde, 2016-09-14:
Added that correction to readme.md - In a project like this there are just so many places where you have to remember to mention things, and update documentation when something changes. I must have forgotten to update that when I merged the cores.

battosai30, 2016-09-14:
It's OK, this extension is just awesome soooo :)

SpenceKonde, 2016-10-03:
Feel free to contact me if you've got other questions on this.

closed 2016-10-03 by SpenceKonde