SpenceKonde/ATTinyCore #85: ATTinyx61 interrupt driven servo library (8 bit)

AtomicCactus, 2016-10-12:
I've put together a really basic interrupt driven servo library that works fairly okay on ATTinyx61 series. It uses Timer1 to control up to 8 servos. Perhaps this could be added to the AVR libraries since there currently is only a library for ATTiny1634? Or even better if it could be merged to create a more universal library. This may, with minor modifications, work on similar AVR 8-bit controllers.

https://github.com/AtomicCactus/Servo_x61

SpenceKonde, 2016-10-12:
I think the commit comment is confusing you. Originally I didn't have a servo library included at all - "just use the normal one" - but that library lacked #ifdefs to know to use TIFR not TIFR1, so I added a check for that and put it in. 

It's a good question which chips actually work and don't work with that version of Servo. I don't think anyone has even done a "sanity check" trying to compile an example with it. 

What I do feel pretty confident in saying though, is that neither x5 nor x61 will work with it, which is a total fucking gloom cloud. 

I love the idea of merging that library into the core. Does it implement the same API as the normal one?

AtomicCactus, 2016-10-12:
You're right, the commit might have been a bit confusing. I did try just importing `Servo.h` at first but it did not compile. Mind you, this was before I spent ~2 weeks learning about timers and such things. I come from a Java server / iOS mobile dev background 😂  

The "Servo_x61" library has a public interface in line with the standard Servo lib, except it lacks the `writeMicroseconds()`. Which is funny because under the hood it uses microseconds. I'll update it. Would be interesting to see what other controllers this would work with. Thanks for giving it a chance! 🤓

PS: I played around with an oscilloscope and this could use some minor adjustments because the total frame length isn't exactly 20 ms (it moves with the pulse width, but it's very close) and other things are a bit whacky. It does keep the servo pointed in the right direction and doesn't produce any jerky movements so I'm using it for my project because there's no love for the x61 out there 😭

SpenceKonde, 2016-10-12:
Not producing jitter is probably the important part.

The work you did is by no means for naught, as the Servo.h I have only has any code for 16-bit timers, of which the x61 (and x5) have none.

SpenceKonde, 2018-10-09:
Last deferral, I swear - 1.2.1 will come fairly soon, since we're sure to end up with bugs relating to the new bootloader stuff too.

SpenceKonde, 2018-11-27:
I've gotten a tiny85 servo implementation that seems to work, will be checked in shortly. Remember that you can't use software serial (including the builtin software serial "Serial" this core provides on chips without a hardware one) while controlling a servo. Anything that disables interrupts for any length of time will produce terrible jitter.  

Will still need to do it for the tinyx61 and tiny43 (which each have slightly different timers available).

AtomicCactus, 2018-11-27:
I have an implementation now that allows for software serial use, but need to package it appropriately. It's not checked in anywhere, but I'm using it in my product here (felik.net). Traveling this week, but will try to submit a pull request once I'm back. My only concerns are:

- requires use of special pins on x61 chips
- doesn't play well with metal-gear servos, only the SG90 kind, despite emitting the correct pulse when measured with an oscilloscope

The code I presented a while back works fine, but like you said we must allow software serial use or else it's not very usable..

SpenceKonde, 2018-11-27:
I consider the "no software serial" limitation to be more acceptable than requiring the servos to be on specific pins. As I understand this limitation exists on the official Arduino core as well, doesn't it? 

I'm about to check in code that works on the tinyx61.

closed 2018-11-27 by SpenceKonde

AtomicCactus, 2018-11-27:
Up to you! Thank you for supporting the ATTinies 👍

SpenceKonde, 2018-11-27:
That said, if the library is in a packagable state, it might be good to put it up on your github so I can add a note pointing to it to the docs for the applicable parts, next to where I remind people not to use software serial with servo.