SpenceKonde/ATTinyCore #86: Generate assembler listing when compiling

per1234, 2016-10-13:
From http://forum.arduino.cc/index.php?topic=428999.msg2957177#msg2957177:
>I would *love* to know how to get the Arduino IDE to output the assembler listing (.lst) with a modified platform.txt - it should be possible... 

I had a go at this and thought I should share my findings. First of all I should admit that I don't know exactly what is meant by "assembler listing" but I'm assuming it is some sort of avr-objdump output from the .elf file generated when the IDE compiles the sketch, there may need to be some specific options added to the command or maybe I'm even wrong about using avr-objdump but hopefully some of this will still be of use.

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#pre-and-post-build-hooks-since-ide-165 specifies some post build hooks. These are only available in IDE 1.6.5 and later but this doesn't break backwards compatibility, it only means that the command is never run on IDE 1.6.4 and older so the .lst file is not generated when using those IDE versions.

My first attempt was adding this command to ATTinyCore's platform.txt:
```
recipe.hooks.objcopy.postobjcopy.1.pattern="{compiler.path}/avr-objdump" -D "{build.path}/{build.project_name}.elf" > "{build.path}/{build.project_name}.lst"
```
Unfortunately the IDE(at least on Windows 7, I haven't tested other OSs) doesn't seem to be able to pipe the avr-objdump output to the .lst file in this way, it treats it as three separate commands:
```
recipe.hooks.objcopy.postobjcopy.1.pattern="{compiler.path}/avr-objdump" -D "{build.path}/{build.project_name}.elf"
```
and:
```
>
```
and:
```
"{build.path}/{build.project_name}.elf"
```
The first prints the avr-objdump output to the console window, the second two are not valid commands.

The solution I found was to pass the command string to the Windows command interpreter, cmd:
```
recipe.hooks.objcopy.postobjcopy.1.pattern=cmd /C "{compiler.path}avr-objdump" -D "{build.path}/{build.project_name}.elf" > "{build.path}/{build.project_name}.lst"
```
which works fine. Unfortunately I'm guessing this only works on Windows. It is possible to specify OS specific versions of a recipe:
```
recipe.hooks.objcopy.postobjcopy.1.pattern.windows=cmd /C "{compiler.path}avr-objdump" -D "{build.path}/{build.project_name}.elf" > "{build.path}/{build.project_name}.lst"
```
So this would work if similar recipes were added for the other OS options, I think you can even set a default without the OS specifier if the same recipe will work for Linux and macosx. Ideally a recipe could be found that works on all operating systems. There's probably some easy solution to my issue that I just don't know. My understanding is that using `>` to pipe to a file is universal.

P.S. I really like the `-S` option with avr-objdump, which intermixes source code with the disassembly. Unfortunately it doesn't work(no source code is shown but the disassembly is otherwise fine) with 1.6.5(you can get it to work if you specify the sketch folder with `-I` but you can't get the location of the sketch folder from platform.txt) or with LTO enabled(I have no idea why).

SpenceKonde, 2016-10-17:
Fascinating, thank you!

What a pity about -s not working with LTO . 

I need to study this - I've got several more fires to iron more urgently

per1234, 2016-10-17:
There's also the option of using a batch file/script instead of the cmd thing to make the pipe work but I thought it was better this way because it avoids adding more files to the repository and you can access the command right there in platform.txt instead of having to figure out that they're in separate files. bperrybap shared a script for Linux(and Mac also?) here:
http://forum.arduino.cc/index.php?topic=428999.msg2958716#msg2958716

I feel like there should be some way to get the intermixed source working with LTO, I've done a bunch of trial and error and searching for a solution without success but I really don't know what I'm doing. It may be that  LTO changes things around so much that the assembly isn't able to be matched up with the source anymore.

per1234, 2017-09-13:
I'm not crazy about adding so much extra output to the console because it makes for much more scrolling but it certainly makes the disassembly immediately accessible. Note that this output will only be shown in the console if you have **File > Preferences > Show verbose output during: > compilation** checked. If it's decided to go with this, I'd recommend adding a custom Tools menu to turn the disassembly output to the console on or off. One consideration is I believe there is a maximum console length setting of 500 by default (`console.length` in preferences.txt) so that could be a limitation for large programs. I tried a quick test of this and the IDE permanently locked up when I dumped a lot of stuff to the console.

Regarding the the intermixed source in disassembly with LTO enabled issue I mentioned above, the fix was found and @celem's comment reminded me I had been meaning to submit a PR for it to this repository, which I have now done: https://github.com/SpenceKonde/ATTinyCore/pull/170.

SpenceKonde, 2018-12-02:
Above it seemed like we couldn't do this in a platform independent way. Why can't we do it the same way that we do the deleting of merged hex files with a different script for different OS's?

If it can't be done in a platform independent way currently, I'm going to shelve this one.

per1234, 2018-12-02:
>Why can't we do it the same way that we do the deleting of merged hex files with a different script for different OS's?

I'm not sure I would consider that "platform independent", since a different recipe was required for each OS, but it's certainly doable.

I was hoping someone might come along and suggest a solution that permitted a single recipe to work for every OS. After 2 years, it doesn't seem like that's going to happen and I don't think it's really such a big deal to have separate recipes so I went ahead and submitted a PR: https://github.com/SpenceKonde/ATTinyCore/pull/269

SpenceKonde, 2018-12-02:
Thanks! 

Is it possible to make this export the file to the sketch folder, instead of the build folder? That way we don't have to dig for it? Maybe as part of the "Export compiled binary" process would be best?

SpenceKonde, 2018-12-02:
Regarding your questions in the PR
* I will get someone to test on a mac. Worst case my roommate's mac will be pressed into service. 
* I don't know enough about dissembler listing options to know what the right ones to use are, honestly.
* I think lst is fine. Is there a reason it wouldn't be? 

 When I used it, I got a bunch of entries mentioning a file that doesn't look right (i mean, this is windows, but this is clearly a linux path - and who is jenkins?): /home/jenkins-mingw32/workspace/avr-gcc-staging/label/Ubuntu14.04x64-mingw32/gcc-build/avr/avr25/libgcc/../../../../gcc/libgcc/config/avr/lib1funcs.S:2490
Any idea from whence that comes?

SpenceKonde, 2018-12-02:
I think .lst is fine, and since that extension seems to be used almost universally, it would probably cause more confusion to not do it. 

Okay, cool. I will document what you said about those precompiled files so if someone asks I can just point them to the docs

So that just leaves the location of the output files.

per1234, 2018-12-02:
>Is it possible to make this export the file to the sketch folder, instead of the build folder?

Unfortunately, I've never found a way to programmatically get the sketch folder location from the Arduino IDE. I looked into this quite a bit for https://github.com/SpenceKonde/ATTinyCore/issues/73 and in the end that project was only possible because I could delete the file from the build folder before it was saved to the sketch folder.

SpenceKonde, 2018-12-02:
Bummer. What a weird thing to be lacking.  

Okay, so it looks good. Closing this - please do let me know if you discover a way to find the path to the sketch; having to dig for the build folder kinda sucks.

closed 2018-12-02 by SpenceKonde

SpenceKonde, 2018-12-03:
It now works on mac (but didn't as submitted). the postfix for macos is macosx not macos.