SpenceKonde/ATTinyCore #87: Bootloader issue in v1.1.1

jdslabs, 2016-10-19:
Using ATTinyCore v1.1.1, Arduino generates the following error when attempting to burn the bootloader to an ATTiny88 via AVRISP mkII (ATtiny):

avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xff instead of 0x3f (double check with your datasheet first).
avrdude: WARNING: invalid value for unused bits in fuse "efuse", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xff instead of 0x07 (double check with your datasheet first).
avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xff instead of 0x3f (double check with your datasheet first)

jdslabs, 2016-10-20:
FYI - This bug occurs under OS X, yet not under Windows.

I'm actually debugging analogRead(). After updating Arduino and loading the latest ATTinyCore v1.1.1, analogRead() fails to read. 

analogRead() is okay on another board running the same code, but burned last week with older ATTinyCore and Arduino 1.6.8. I manually read flash from this older board and then wrote the .hex file to a newer board; this restored functionality of analogRead(). I noted avrdude reads the following fuses from the older boards:

Lfuse: E2
HFuse: D7
EFuse: 7

So, perhaps the deactivated ADC is related to this bootloader bug?

SpenceKonde, 2016-10-20:
It should happen on windows... I hope it does, else I can't fix it without borrowing hardware. 

You may need to update arduino official avr boards package to trigger it - it only happens with the latest couple of versions of the AVR boards package installed (which brings the new compiler and new avrdude)

ADC issue is unrelated - these warnings from avrdude do not impact functionality, they are just warnings that need to be fixed in the avrdude.conf and boards.txt in order to ensure future compatibility. 

Are you using 1.1.1 or the github install? I was playing with analogRead() the other day to make it work with the differential ADC channels, and I don't think I broke anything, but maybe I did. I also may have fixed some things that are broken in the 1.1.1 release (I feel like I've done something to the ADC code not that long ago)

These changes are in github, but not the release - so if you're seeing bad behavior with the ADC, my recommendation would be to:
Remove the board package (if installed by board manager), and install from github (clone the repo into sketchbook/hardware folder). See if the problem still reproduces. If it does, open a new issue for the ADC problem, with test code if possible.

jdslabs, 2016-10-20:
I've relied on the Arduino Boards Manager for all testing so far, which shows v1.1.1 Installed. Will try your suggestions today. Thanks!

jdslabs, 2016-10-20:
Findings: The Bootloader error is consistent with latest Arduino v1.6.12, using Arduino AVR Boards v1.6.14, and your latest ATTiny commit cdf0955. The bug was absent while I was running an outdated version of AVR Boards.

Moving ADC comments to separate bug.

closed 2016-10-23 by SpenceKonde