SpenceKonde/ATTinyCore #89: LED_BUILTIN not defines

MCUdude, 2016-10-22:
Since Arduino IDE 1.6.12 the blink sketch example have been slightly changed to fit different HW configurations. The problem now is that the LED_BUILTIN isn't defined in your core. I chose PB0 (digital pin 0) as the "LED pin" on the ATtiny13. Maybe you should choose the same pin on the ATtinyX5, since they're pin compatible? The definition is supposed to be located in the pins_arduino file 😉 

``` c++
// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

SpenceKonde, 2016-10-22:
Fucking christso they changed the example so we have to change our cores?

MCUdude, 2016-10-22:
```
modified 2 Sep 2016
by Arturo Guadalupe
```

Go and get him!!

MCUdude, 2016-10-22:
It seems like damellis' core doesn't have the definition either.

sleemanj, 2016-10-22:
My opinion on this is... make specific examples for the Tiny series, as I did in my fork for the Tiny13 so far

  https://github.com/sleemanj/ATTinyCore/tree/master/avr/libraries/ATTinyCore

of course it's a bit sucky to have to stick them in a dummy library like this, but it works well enough.

Since ATTinyCore is not targeted to any specific board, but to primarily DIY, having a specific "LED PIN" defined seems silly.

SpenceKonde, 2016-10-23:
Well, the Optiboot bootloaders assume the LED is on a specific pin (and that's the one I use in the boards I sell). That is the pin I should use for LED_BUILTIN ;-) 

It's his star power. His entire approach is wrongheaded, but totally to be expected considering his position as an arduino developer - he has faith in the official core, and the ability to get changes he needs stuffed into it. I have neither.

commit 89e4f67e (2016-10-23): Fix #89 LED_BUILTIN missing

Also added PIN_xx macros to the classic tinys.
closed 2016-10-23 by SpenceKonde