SpenceKonde/ATTinyCore #91: yield() promotes bad code, but should support it for compatibility

SpenceKonde, 2016-10-25:
Recent arduino releases have introduced a weakly defined (and empty) yield() function that gets called from delay(). 

This abomination should be supported in this core, much as it promotes foolish abuse of delay.

SpenceKonde, 2017-01-24:
Was it really this easy?! Try it out and let me know if it works as expected. 

There's an issue with the length of the yield() function (in terms of execution time). The original implementation of delay()/yield() totally ignored this and started screwing up when yield() took over a millisecond to run. On the official core it was fixed the Right Way, but this takes an extra 40 bytes, and the people using chips with 2k of flash would bust out the pitchforks. I halfassed with a while loop, but 16-bit variables - as long as the number of microseconds yield takes fits in a uint_16, it's fine. Otherwise, delay()s will be longer - will document this in core documentation.

closed 2017-01-24 by SpenceKonde

sleemanj, 2017-01-24:
ELI5 ... if yield is empty, why are you bothering to call it from delay() at all?

Just define it (emptily) for the sake of compatibility with 3rd party stuff (FastLED) but you don't need to use it yourself and risk messing up timing and stuff.

SpenceKonde, 2017-01-24:
You think that's better? I guess I could define it normally so they couldn't override it and it would barf if they tried to use it, but then they'd request working yield() support in a future version! I think this will cover most use of yield without problems. If the user doesn't write their own yield() function it won't mess up the timing. I think they've got like 65 milliseconds with my hackjob - only if they write a longer yield() function will it cause timing issues.

sleemanj, 2017-01-24:
No you're right, I misunderstood/remembered what yield was added for.

commit 3981e0b8 (2017-01-24): fix #91?

Compile tested. Behavior not verified. Let me know how it works out.