SpenceKonde/ATTinyCore #94: 4313 refuses to compile on 1.6.11

Isaac96, 2016-11-10:
When I try to compile "BareMinimum" for the 4313, I get this error:
`Arduino: 1.6.11 (Windows 7), Board: "ATtiny2313/4313, Disabled, ATtiny4313, 8 MHz (internal), B.O.D. Disabled, no"

C:\Users\Isaac\Desktop\Chrome Downloads\arduino-1.6.11-windows\arduino-1.6.11\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\Isaac\Desktop\Chrome Downloads\arduino-1.6.11-windows\arduino-1.6.11\hardware -hardware C:\Users\Isaac\AppData\Local\Arduino15\packages -hardware C:\Users\Isaac\Documents\Arduino 1.6.3\hardware -tools C:\Users\Isaac\Desktop\Chrome Downloads\arduino-1.6.11-windows\arduino-1.6.11\tools-builder -tools C:\Users\Isaac\Desktop\Chrome Downloads\arduino-1.6.11-windows\arduino-1.6.11\hardware\tools\avr -tools C:\Users\Isaac\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\Isaac\Desktop\Chrome Downloads\arduino-1.6.11-windows\arduino-1.6.11\libraries -libraries C:\Users\Isaac\Documents\Arduino 1.6.3\libraries -fqbn=ATTinyCore:avr:attinyx313:LTO=disable,chip=4313,clock=8internal,bod=disable,INITIALIZE_SECONDARY_TIMERS=0 -ide-version=10611 -build-path C:\Users\Isaac\AppData\Local\Temp\build58fb4bd34f79bd74c90b0f9decda248a.tmp -warnings=none -prefs=build.warn_data_percentage=75 -verbose C:\Users\Isaac\Desktop\Chrome Downloads\arduino-1.6.11-windows\arduino-1.6.11\examples\01.Basics\BareMinimum\BareMinimum.ino

Board attinyx313 (platform avr, package ATTinyCore) is unknown

Error compiling for board ATtiny2313/4313.`

SpenceKonde, 2016-11-10:
That's very strange - it looks like the core is somehow installed wrong. I would try removing and reinstalling it through board manager. 

I can't reproduce it on my system.

per1234, 2016-11-10:
`Board attinyx313 (platform avr, package ATTinyCore) is unknown`
Typically this is caused by the IDE not removing the folder of the previously installed version when you do an update of the core in Boards Manager. I can reproduce by creating an empty folder named 1.1.1 in the avr folder of the core.

Isaac96, 2016-11-12:
Thanks per-it works now :smile:

closed 2016-11-12 by Isaac96