SpenceKonde/ATTinyCore #95: Incorrect Power Pinout for x7 Series in README

david476, 2016-11-15:
Based on [the datasheet](http://www.atmel.com/Images/Atmel-8265-8-bit-AVR-Microcontroller-tinyAVR-ATtiny87-ATtiny167_datasheet.pdf) and my testing, the image in the README for ATTiny87/167 has GND and VCC swapped on the left side. I'm happy to fix it if you give me a hint on how to generate them.

SpenceKonde, 2016-11-15:
x61 series has the same issue. 

@MCUdude - can you help me out here?

MCUdude, 2016-11-15:
Sure, here you go! You might need to crop it to fit 😉 
[ATtiny87_167.zip](https://github.com/SpenceKonde/ATTinyCore/files/593141/ATtiny87_167.zip)

david476, 2016-11-16:
It looks like the images currently in the readme are stored on @SpenceKonde 's website. Here are the new ones cropped if you want to add them through that.
[Fixed_x7-x61_pinouts.zip](https://github.com/SpenceKonde/ATTinyCore/files/593416/Fixed_x7-x61_pinouts.zip)

SpenceKonde, 2016-11-16:
Thanks guys <3

commit 1f07ae45 (2016-11-16): fix #95

last part.
closed 2016-11-16 by SpenceKonde