SpenceKonde/ATTinyCore #97: TinySoftwareSerial fail with LTO

sleemanj, 2016-11-27:
With LTO enabled if you compile a serial using sketch for a target that uses TinySoftwareSerial you will get linker errors.

    `uartDelay' referenced in section `.text' of /tmp/ccu7aSIN.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)
    collect2: error: ld returned 1 exit status

I fixed it in my fork like this:
https://github.com/sleemanj/ATTinyCore/commit/fd45ef97f71ad4a478428da9dd2b9bad0d4bdc73

But don't know if that's the right approach or not.

SpenceKonde, 2016-11-27:
If that makes it work, it sounds like that's what we needed there.
Thanks for finding that one. Do you want to PR it, or should I just copy your fix?

sleemanj, 2016-11-27:
It's only one line, just copy it :)

closed 2016-11-27 by SpenceKonde