SpenceKonde/ATTinyCore #98: Interputs from PIN not working

dragonnn, 2016-11-30:
Hi, I have Adafruit Tinker board, use ATTinyCore but I can not get Interput from pin get working, here is my code:
#include <SoftwareSerial.h>
#include "DFPlayer_Mini_Mp3.h"

SoftwareSerial mySerial(0, 2); // RX, TX

void isr()
{
  //if (!digitalRead(3)) {
  //  mp3_play (1);
  //}
  digitalWrite(1, HIGH);
}
// the setup routine runs once when you press reset:
void setup() {
  pinMode(1, OUTPUT);
  digitalWrite(1, LOW);
  pinMode(3,INPUT_PULLUP);                // IR on pin 3
  attachInterrupt(digitalPinToInterrupt(3),isr,FALLING);
  mySerial.begin (9600);
  mp3_set_serial (mySerial);  //set softwareSerial for DFPlayer-mini mp3 module 
  delay(1000);
  mp3_set_volume (15);
  delay(1000);
  mp3_play (1);
}
 
// the loop routine runs over and over again forever:
void loop() {

}

I short the PIN3 to ground and nothing happens :/. Any idea?

SpenceKonde, 2016-11-30:
What chip are we working with here? I can't find any references to an "Adafruit tinker", so I'm not sure what board you're referring to.

dragonnn, 2016-11-30:
Sorry, it is Trinket - an Attiny 85 is on the board.

SpenceKonde, 2016-11-30:
How are you using my core with a trinket? Are you using the usb programming? I didn't realize that worked...

dragonnn, 2016-11-30:
Yes normal usb programing, the rest of the program works fine. I had the same problem with Interputs on the original Trinket board library.

SpenceKonde, 2016-11-30:
I'm uncertain if this is related to the Trinket bootloader - it looks like they do some vector rewriting (judging by the bootloader hex - I couldn't find the part of the bootloader where they actually do that in the code). I haven't worked with a trinket before - but I also haven't tested attachInterrupt on a tiny85 myself either.

dragonnn, 2016-11-30:
Ok... I understand :/. So I will move to Pro Mico for this project. Code above works on it.

closed 2016-11-30 by dragonnn

SpenceKonde, 2016-11-30:
I would definitely expect it to work on a bare ATTiny85 (or a trinket reprogrammed by ISP, removing the USB bootloader) - if it doesn't work on a vanilla '85, it's a bug. 

Try reworking it to use PCINT's - that might get you there. I normally reach for PCINTs when I need interrupts in general.