SpenceKonde/ATTinyCore #99: Can't Compile "Blink" for ATtiny84

batchku, 2016-12-05:
Hello all,

I'm trying to compile the simple Blink.ino program for an ATtiny84, and am getting errors.  I'm using the Arduino IDE (1.6.12); I've added this core using the board manager, and have version 1.1.2 installed, which is the latest according to the Arduino board manager.

I get a lot of warnings and a couple of errors, most notably this error:

```
exec: "avr-g++": executable file not found in $PATH
Error compiling for board ATtiny85 @ 1 MHz  (internal oscillator; BOD disabled).
```

Here is the verbose output and the warnings:

```
Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/ali/Library/Arduino15/packages -hardware /Users/ali/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/ali/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/ali/Documents/Arduino/libraries -fqbn=tiny:avr:attiny85at1 -ide-version=10612 -build-path /var/folders/_v/130dhl9n76d0p3xtjczy7rjw0000gn/T/arduino_build_270347 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/ali/Library/Arduino15/packages -hardware /Users/ali/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/ali/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/ali/Documents/Arduino/libraries -fqbn=tiny:avr:attiny85at1 -ide-version=10612 -build-path /var/folders/_v/130dhl9n76d0p3xtjczy7rjw0000gn/T/arduino_build_270347 -warnings=none -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
Using board 'attiny85at1' from platform in folder: /Users/ali/Documents/Arduino/hardware/tiny/avr
Using core 'tiny' from platform in folder: /Users/ali/Documents/Arduino/hardware/tiny/avr
Warning: Board tiny:avr:attiny2313at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY2313AT8
Warning: Board tiny:avr:attiny2313at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY2313AT1
Warning: Board tiny:avr:attiny25at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY25AT1
Warning: Board tiny:avr:attiny44at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY44AT8
Warning: Board tiny:avr:attiny85at128 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY85AT128
Warning: Board tiny:avr:attiny85at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY85AT1
Warning: Board tiny:avr:attiny85at16p doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY85AT16P
Warning: Board tiny:avr:attiny45at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY45AT1
Warning: Board tiny:avr:attiny24at16 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY24AT16
Warning: Board tiny:avr:attiny84at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY84AT8
Warning: Board tiny:avr:attiny25at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY25AT8
Warning: Board tiny:avr:attiny85at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY85AT8
Warning: Board tiny:avr:attiny84at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY84AT1
Warning: Board tiny:avr:attiny45at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY45AT8
Warning: Board tiny:avr:attiny4313at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY4313AT1
Warning: Board tiny:avr:attiny85at16e doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY85AT16E
Warning: Board tiny:avr:attiny44at1 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY44AT1
Warning: Board tiny:avr:attiny4313at8 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY4313AT8
Warning: Board tiny:avr:attiny84at16 doesn't define a 'build.board' preference. Auto-set to: AVR_ATTINY84AT16
Build options changed, rebuilding all
Warning: platform.txt from core 'Arduino Tiny' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'Arduino Tiny' contains deprecated tools.avrdude.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf, automatically converted to tools.avrdude.config.path={path}/etc/avrdude.conf. Consider upgrading this core.
Warning: platform.txt from core 'Arduino Tiny' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'Arduino Tiny' contains deprecated recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm, automatically converted to recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{archive_file_path}" "-L{build.path}" -lm. Consider upgrading this core.
Detecting libraries used...
"avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=1000000L -DARDUINO=10612 -DARDUINO_AVR_ATTINY85AT1 -DARDUINO_ARCH_AVR  "-I/Users/ali/Documents/Arduino/hardware/tiny/avr/cores/tiny" "/var/folders/_v/130dhl9n76d0p3xtjczy7rjw0000gn/T/arduino_build_270347/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections  -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=1000000L -DARDUINO=10612 -DARDUINO_AVR_ATTINY85AT1 -DARDUINO_ARCH_AVR  "-I/Users/ali/Documents/Arduino/hardware/tiny/avr/cores/tiny" "/var/folders/_v/130dhl9n76d0p3xtjczy7rjw0000gn/T/arduino_build_270347/sketch/Blink.ino.cpp" -o "/var/folders/_v/130dhl9n76d0p3xtjczy7rjw0000gn/T/arduino_build_270347/preproc/ctags_target_for_gcc_minus_e.cpp"
exec: "avr-g++": executable file not found in $PATH
Error compiling for board ATtiny85 @ 1 MHz  (internal oscillator; BOD disabled).
```

SpenceKonde, 2016-12-06:
Something is wrong with your arduino ide install. It appears to not know
where the avr vcc tools are - it should be calling the full path to
avr-g++. Can you compile for an uno? I don't think you'll be able to, due
to same problem. In order to prevent compatibility problems caused by using
its own copy of the compiler tools, my core uses the tools from the
standard avr board package; if they're not working correctly, my core wont
work either.

Let me know if you can compile for the uno. If you can, I will need to talk
to people more versed in the dark magic of arduino-builder.

-Spence

per1234, 2016-12-06:
The problem is that @batchku is using https://code.google.com/archive/p/arduino-tiny, not ATTinyCore. That core is very outdated. I highly recommend using ATTinyCore instead. If you do want to still use arduino-tiny then you can either use an older IDE version that it was written for or search the Arduino forum because I know I posted instructions for how to update the core a while back. If you can't find it then make a post on the forum and I'll try to help you out there but it's off topic to discuss a different core on this repository.

SpenceKonde, 2016-12-06:
Ah, thanks... I missed that

Yeah, if you installed my core as you say, you've got another much older
core installed, and selecfed a board from that instead of one from
ATTinyCore.

closed 2016-12-07 by SpenceKonde

wavewolf, 2018-11-15:
I'm trying to get my "nrf24L01" module working through SPI with Attiny84 but I can not get it to work.
I am using your core attiny and the library nrf24L01 of this link:
http://tmrh20.github.io/RF24/ATTiny.html

This is my pins:

       +-\/-+                                                              
nRF24L01  VCC, pin2 --- VCC  1|o   |14 GND --- nRF24L01  GND, pin1
                        PB0  2|    |13 AREF
                        PB1  3|    |12 PA1
                        PB3  4|    |11 PA2 --- nRF24L01   CE, pin3
                        PB2  5|    |10 PA3 --- nRF24L01  CSN, pin4
                        PA7  6|    |9  PA4 --- nRF24L01  SCK, pin5
nRF24L01 MISO, pin7 --- PA6  7|    |8  PA5 --- nRF24L01 MOSI, pin6
                              +----+

Whay can I do?