SpenceKonde/ATTinyCore #100: Link time optimisation causes compiler error (ATtiny85)

stuartsnow, 2016-12-07:
If I turn on LTO while compiling for the ATtiny85, I get a compiler error if the sketch uses the core's built in Serial (which I understand is a modified SoftwareSerial).

I've isolated it to this basic sketch:

```
uint8_t i;

void setup()
{
  Serial.begin(115200);
  Serial.println("one");
  Serial.println("two");
}

void loop()
{
  i = analogRead(A0);
  Serial.println(i);
  delay(100);
}
```

With LTO off, it compiles and runs as expected. Turn LTO on, and this is the resulting error:

```
`uartDelay' referenced in section `.text' of C:\Users\ssnow\AppData\Local\Temp\cc0IS3eQ.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)

`uartDelay' referenced in section `.text' of C:\Users\ssnow\AppData\Local\Temp\cc0IS3eQ.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)

`uartDelay' referenced in section `.text' of C:\Users\ssnow\AppData\Local\Temp\cc0IS3eQ.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)

`uartDelay' referenced in section `.text' of C:\Users\ssnow\AppData\Local\Temp\cc0IS3eQ.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)

`uartDelay' referenced in section `.text' of C:\Users\ssnow\AppData\Local\Temp\cc0IS3eQ.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)

`uartDelay' referenced in section `.text' of C:\Users\ssnow\AppData\Local\Temp\cc0IS3eQ.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)

`uartDelay' referenced in section `.text' of C:\Users\ssnow\AppData\Local\Temp\cc0IS3eQ.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)

`uartDelay' referenced in section `.text' of C:\Users\ssnow\AppData\Local\Temp\cc0IS3eQ.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)

`uartDelay' referenced in section `.text' of C:\Users\ssnow\AppData\Local\Temp\cc0IS3eQ.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)

`uartDelay' referenced in section `.text' of C:\Users\ssnow\AppData\Local\Temp\cc0IS3eQ.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)

`uartDelay' referenced in section `.text' of C:\Users\ssnow\AppData\Local\Temp\cc0IS3eQ.ltrans0.ltrans.o: defined in discarded section `.text' of TinySoftwareSerial.cpp.o (symbol from plugin)

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board ATtiny25/45/85.

```

If I replace Serial with SoftwareSerial all is well.
I've already tried reinstalling ATTinyCore.

Cheers.

SpenceKonde, 2016-12-07:
This is a duplicate of #97 - it is fixed in the github version, but the fix is not yet in the board manager release. 

The linker was getting confused, and optimizing out that variable, so it needed the 'used' keyword added. 

To use github version, remove any installed versions using board manager, then sync the repo to the hardware folder inside your sketchbook folder. https://github.com/SpenceKonde/ATTinyCore/blob/master/Installation.md

closed 2016-12-07 by SpenceKonde

stuartsnow, 2016-12-07:
My bad, I'm new to GitHub and I didn't check the closed issues. Nice work!

SpenceKonde, 2016-12-08:
No problem. I'd rather close duplicate entries than have people not report
non-duplicates

Thanks!