SpenceKonde/ATTinyCore #102: SoftwareSerial and Interrupts (INT0 &/| PCIntx) at the same time

Flying-Bolt, 2016-12-15:
Hi Master,

it is possible to use the SoftwareSerial library at the same time with  Interrupts (INT0 &/| PCIntx)?

Servus Arno

SpenceKonde, 2016-12-15:
The SoftwareSerial library should work at the same time as INT0, but not PCINTs. 

Since PCINTs are so useful, this core also includes a work-around to get a software serial implementation on chips without a hardware serial, without interfering with PCINTs. If you're using one of the boards that doesn't have any hardware serial, there's a built-in software serial implementation that uses analog comparator and analog comparator interrupt (this seems to be the least useful interrupt that can be made to fire on a pin change); this serial implementation is named "Serial" like the normal hardware serial implemntation (for compatibility). To double-check what serial facilities are available on the chip you're working with, refer to the Serial section of the relevant page here: https://github.com/SpenceKonde/ATTinyCore/tree/master/avr/extras

SpenceKonde, 2017-01-10:
No response, assuming solved.

closed 2017-01-10 by SpenceKonde