SpenceKonde/ATTinyCore #104: ANALOG_PINS_SEPARATE definition causing ADC to return 0

roweboat, 2017-01-10:
Hi, in an older version of your library you did not have this line and its intended functionality.
#define ANALOG_PINS_SEPARATE in pins_arduino.h
This version worked great for us and we could sample the ADC no problem.

After upgrading to the latest version this line is causing all of the ADC channels to read 0. We are requesting the conversion using the arduino names A0, A1 etc. Is this expected?

Cheers
Phil

SpenceKonde, 2017-01-10:
Which micro is this? The A(number) constants are supposed to work. 

Does calling it with the ADC channel number work? (ie, analogRead(0) instead of analogRead(A0) ?)

roweboat, 2017-01-11:
We are using the ATTiny88 micro with 1.8.1 of the IDE having downloaded straight from your link using board manager. 

The problem is with ANALOG_PINS_SEPARATE defined the adc condition is never true which means the channel being used is like 17 to 24 (instead of 0 to 7). 

Cheers
Phil

sleemanj, 2017-01-11:
X8 defines ANALOG_PINS_SEPARATE

https://github.com/SpenceKonde/ATTinyCore/blame/master/avr/variants/tinyX8/pins_arduino.h#L73

but stacks the defines on top of the digital pins, 

https://github.com/SpenceKonde/ATTinyCore/blame/master/avr/variants/tinyX8/pins_arduino.h#L84

I guess nobody uses the X8, because I don't see how analogRead() would ever have worked :-)

SpenceKonde, 2017-01-11:
AFAIK you're the only person using it with the x8, yeah. 

You're certainly the one who finds all the bugs on it.

commit 70431f2d (2017-01-11): Fix #104 

Hopefully this isn't going to break something else.
closed 2017-01-11 by SpenceKonde