SpenceKonde/ATTinyCore #105: EESAVE bit enable option

klotzy550, 2017-01-18:
Not really an issue, just looking if you can add an option under the "BOD" menu for the "attinyX4?"

The option is as follows, High fuse set to 0xC7, BOD disabled with EESAVE bit enabled. This allows me to keep all of my eeprom data between program uploads.

Any help would be greatly appreciated! 

`attinyx4.menu.bod.EESAVE=EESAVE Enabled`
`attinyx4.menu.bod.EESAVE.bootloader.high_fuses=0xC7`

SpenceKonde, 2017-01-18:
I need to see if I can figure out a graceful way to provide an option to configure both BOD and EESAVE options within the menu system. 

Worst case, the BOD options menu will become BOD and EESAVE options, and will have twice as many options in it as it does now.

SpenceKonde, 2017-01-24:
Uh, dude...

The EESAVE bit is already programmed in all the board definitions. Sorry for missing this when you first posted.

Setting the high fuse to D7 (like my core would for BOD disabled) has EESAVE set - EESAVE is the third bit - if it was not enabled, the high fuse would be DF. The difference between C7 and D7 is WDTON (C7 has the WDT permanently on). 

Can you clarify if this is what you meant?

klotzy550, 2017-01-24:
Ahh, sorry for the confusion. I was programming a ATTINY84 a month or so ago and it was clearing the EEPROM every flash. I may have been using another core. I apologize for the inconvenience. I must have looked that the high fuse byte wrongly then. Again, sorry. Also, good to know that EESAVE is active.

MCUdude, 2017-02-10:
> Worst case, the BOD options menu will become BOD and EESAVE options, and will have twice as many options in it as it does now.

FYI you actually can set the induvidal fuse bits using Arduino IDE. All you have to do is to represent the fuses in binary, manipulate them induvidually and concatenate them together. Here's how I was able to [change the ATmega8 BOD option](https://github.com/MCUdude/MiniCore/blob/master/avr/boards.txt#L401) without interfering with the other bits on the low fuse. It's very useful if you want to add even more options to the boards menu 😃

SpenceKonde, 2017-02-10:
Woah! That is *awesome* - that's clearly the Right Way to do this.

closed 2018-02-28 by SpenceKonde